apache/nuttx #15170: [BUG] rv-virt:citest fails with Load Access Fault at ltp_interfaces_pthread_barrierattr_init_2_1 (risc-v-05)

lupyuen, 2024-12-13:
### Description / Steps to reproduce the issue

Inside CI Build risc-v-05: `rv-virt:citest` fails with a Load Access Fault at ltp_interfaces_pthread_barrierattr_init_2_1:

https://gist.github.com/lupyuen/4ec372cea171b99ae5bc5603aa75a6a7
```text
sudo docker run \
  -it \
  --name nuttx \
  ghcr.io/apache/nuttx/apache-nuttx-ci-linux:latest \
  /bin/bash
cd
git clone https://github.com/apache/nuttx
git clone https://github.com/apache/nuttx-apps apps
pushd nuttx ; echo NuttX Source: https://github.com/apache/nuttx/tree/$(git rev-parse HEAD) ; popd
pushd apps  ; echo NuttX Apps: https://github.com/apache/nuttx-apps/tree/$(git rev-parse HEAD) ; popd
cd nuttx/tools/ci
./cibuild.sh -c -A -N -R testlist/risc-v-05.dat 
[Press Ctrl-C a few times to stop it]
cat /root/nuttx/boards/risc-v/qemu-rv/rv-virt/configs/citest/logs/rv-virt/qemu/*

## Output shows:
NuttX Source: https://github.com/apache/nuttx/tree/5cdc6c77cfcecc162b74b1bcc902fde978384352
NuttX Apps: https://github.com/apache/nuttx-apps/tree/fa22f80ee3a0b6b232752e8d796129bd79082da8
...
test_open_posix/test_openposix_.py::test_ltp_interfaces_sem_post_1_2 PASSED               [ 17%]
test_open_posix/test_openposix_.py::test_ltp_interfaces_sigtimedwait_2_1 PASSED           [ 17%]
test_open_posix/test_openposix_.py::test_ltp_interfaces_pthread_barrierattr_init_2_1 FAILED [ 17%]
test_open_posix/test_openposix_.py::test_ltp_interfaces_sigaction_13_9 FAILED             [ 17%]
test_open_posix/test_openposix_.py::test_ltp_interfaces_mq_send_14_1 FAILED               [ 17%]
...
nsh> ltp_interfaces_pthread_barrierattr_init_2_1
riscv_exception: EXCEPTION: Load access fault. MCAUSE: 00000005, EPC: 800074c6, MTVAL: 000002a4
riscv_exception: PANIC!!! Exception = 00000005
dump_assert_info: Current Version: NuttX  12.7.0 5cdc6c77cf Dec 13 2024 05:10:36 risc-v
dump_assert_info: Assertion failed panic: at file: common/riscv_exception.c:131 task: ltp_interfaces_pthread_barriera process: ltp_interfaces_pthread_barriera 0x800664ae
up_dump_register: EPC: 800074c6
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu LTS at GitHub Actions

### NuttX Version

master

### Issue Architecture

[Arch: risc-v]

### Issue Area

[Area: Other]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-12-14:
@lupyuen any idea which commit introduced this issue?

lupyuen, 2024-12-15:
@acassis I think it happens sometime after this commit https://github.com/apache/nuttx/commit/656883fec5561ca91502a26bf018473ca0229aa4, based on the [earlier discussion](https://github.com/apache/nuttx/issues/14808#issuecomment-2539867269)
 
I'll have to run "git bisect" on the past month of commits.

closed 2024-12-16 by xiaoxiang781216