apache/nuttx #15292: [HELP] dafault math.h

geniusdo, 2024-12-19:
### Description

How to replace the default math.h. I tried to use libmcs as my default math lib. I have to directly include "libmcs/math.h" to use libmcs math lib. But I want to directly include math.h in libmcs. I tried the method mentioned in #1503. However when I include math.h it still links to the math lib supplied by my toolchain. Any idea to achieve that?

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-12-19:
you can include math.h directly, build system already setup the search path automatically.

geniusdo, 2024-12-19:
I am using cxx build. It seems cmath still call math lib from my toolchain

xiaoxiang781216, 2024-12-19:
which phase(compe v.s. link) do you reference?

geniusdo, 2024-12-19:
During compile phase. I want cmath to use math from libmcs. Idk how it auto links to toolchain math.
[image]

xiaoxiang781216, 2024-12-19:
it's strange, you need:

1.  Issue `make V=1` to extract the command line
2. Issue the command line with -E and check the preprocessed output file to know why the wrong file get included