apache/nuttx #15314: [BUG] esp32s3 wifi deadlock

yamt, 2024-12-23:
### Description / Steps to reproduce the issue

net_lock() and esp_wifi_lock() are taken in an inconsistent order.
thus it can cause a deadlock.

```
wlan_ifdown net_lock();
  |
  v
esp_wifi_sta_stop esp_wifi_lock(true);
```

```
esp_evt_work_cb esp_wifi_lock(true);
  |
  v
esp32s3_wlan_sta_set_linkstatus(false)
  |
  v
netdev_carrier_off
  |
  v
devif_dev_event net_lock();
```

the most straightforward fix: make esp_evt_work_cb (and other functions taking esp_wifi_lock first if any) acquire net_lock() before esp_wifi_lock(). (note that net_lock() is a recursive lock.)

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS 15.2

### NuttX Version

master

### Issue Architecture

[Arch: xtensa]

### Issue Area

[Area: Drivers]

### Verification

- [X] I have verified before submitting the report.

fdcavalcanti, 2024-12-23:
Hi @yamt. Did you have issues on S3 using WiFi due to this?
What devkit did you use?

fdcavalcanti, 2024-12-23:
Interesting.
Feel free to submit this modification and I can test it on our internal CI.

yamt, 2025-01-07:
> Interesting. Feel free to submit this modification and I can test it on our internal CI.

see https://github.com/apache/nuttx/pull/15445

closed 2025-01-08 by xiaoxiang781216

yamt, 2025-01-14:
> > Interesting. Feel free to submit this modification and I can test it on our internal CI.
> 
> see [#15445](https://github.com/apache/nuttx/pull/15445)

see also https://github.com/apache/nuttx/pull/15532