apache/nuttx #15432: Remove STM32 legacy pinmap

raiden00pl, 2025-01-06:
### Description

It's been quite some time since `CONFIG_STM32xx_USE_LEGACY_PINMAP` was introduced. Some upstream boards have already migrated to the new pinmap, but for many there is no progress. There was enough time for users to migrate their boards.

Steps to fix this issue:

- [ ] migrate `stm32f1` boards to new pinmap 
- [ ] migrate `stm32f2` boards to new pinmap 
- [ ] migrate `stm32f3` boards to new pinmap 
- [ ] migrate `stm32f4` boards to new pinmap 
- [ ] migrate `stm32l1` boards to new pinmap 
- [ ] migrate `stm32l4` boards to new pinmap
- [ ] migrate `stm32l5` board to new pinmap
- [ ] migrate `stm32wb` board to new pinmap
- [ ] remove legacy pinmap from `stm32h7` - boards already migrated to new pinmap
- [ ] remove legacy pinmap from `stm32f7` - boards already migrated to new pinmap
- [ ] remove legacy pinmap from `stm32f0l0g0` - boards already migrated to new pinmap

The biggest problem here is the port of `board/stm32` boards to the new pinmap, the number of stm32 boards is quite large, so practical there is no person who can test all modified board on real HW. In this case, we must accept the risk that part of the configuration will be broken, since compiling the code without errors doesn't provide certainty that the port is still working. Here is an example of me accidentally messing up the stm32f7 configuration while migrating the pinmap: https://github.com/apache/nuttx/pull/13915. We definitely need to watch out for similar cases.

Is anyone willing to help with this issue? 
Maybe someone from px4 team is interested in helping? This issue is basically a result of fixes needed for your project and it was left in "half done" state with absolutely no progress for over a year. So sad...

### Verification

- [X] I have verified before submitting the report.

cederom, 2025-02-02:
* I asked STM several times for NuttX RTOS contracting development support but they were not interested :-(
* We should not move with that subject forward, especially if with a big chance of breaking stuff, until we have an option to runtime validate all / most of the boards.
* Even simple build and runtime scripts can be made just to test changes and push the DRUNX forward. I have some boards and can help here :-) Will list the boards in a free moment.
* STM was always keen to share their devkits even for free so when we have a full list of boards I can requests free samples for me and for you @raiden00pl so we can move forward?

cederom, 2025-02-02:
Okay I have created list of hardware that I have at hand. Some modern devkits seems to have more powerful STLink MCU than the target MCU 🗡 

@raiden00pl please mark the hardware that we need and I will try to ask STM for free samples so we could test changes in NuttX and add them all to DRUNX :-)

raiden00pl, 2025-02-03:
@cederom changing most boars is easy and the risk of breaking something is marginal. Problematic are only boards with 
 peripherals that require complex pin configuration like displays or external memory.
And here comes the problem, because some of these boards are old and probably unavailable. Many boards are also from manufacturers other than ST.