apache/nuttx #15435: [FEATURE] PlantUML support for Documentation

raiden00pl, 2025-01-06:
### Is your feature request related to a problem? Please describe.

PlantUML (https://plantuml.com/) allows to store UML diagrams in text format so we can easy integrated it to our Documentation with sphinx extension: https://github.com/sphinx-contrib/plantuml

We already have sequence diagram on this page: https://nuttx.apache.org/docs/latest/guides/port_bootsequence.html
but it's in image format, which makes it poor for tracing in git history and for viewing in terminal. 
I'm not sure if there are more UML diagrams in the doc, but with an easy to use UML plugin maybe there will be more readable diagrams in the doc. PlantUML also allows to create many other types of diagrams that may be useful.

The cons I see for this feature is that we have to install `plantuml` in docker and the doc compilation time may be a bit longer.

### Describe the solution you'd like

all above

### Verification

- [X] I have verified before submitting the report.

linguini1, 2026-02-10:
I think this is a good idea. I can try to implement it.

cederom, 2026-02-14:
Great idea @raiden00pl !! Looks like @linguini1 just implemented it with https://github.com/apache/nuttx/pull/18379 and https://github.com/apache/nuttx-website/pull/166 please take a look! :-)