apache/nuttx #15436: Simplify PWM driver: remove CONFIG_PWM_MULTICHAN

raiden00pl, 2025-01-06:
### Description

`CONFIG_PWM_MULTICHAN` seems to be redundant and unnecessarily complicates the PWM implementation in `arch/`. We can achieve the same functionality as `CONFIG_PWM_MUTLICHAN=n` by just setting `CONFIG_PWM_NCHANNELS=1` which is the default value in Kconfig.
If a given architecture doesn't support multichanel PWM, then we can just add a simple `#if CONFIG_PWM_NCHANNELS > 1` which returns a compilation error.

### Verification

- [X] I have verified before submitting the report.

acassis, 2025-01-07:
According with git, this symbol was added by @augustofg sometime ago, but I think the original support to PWM multi-channel was created by other people (my memory state that it was created by you or Matias Nitsche @protobits).

raiden00pl, 2025-01-08:
@acassis multichan PWM was added loooong time ago in e8c2466c1cecf22035976a2dd577dfc67f9ea373.
I have no idea why it was done this way, but I think it can be done better by avoiding unnecessary complication of the code. To be clear: this code works fine for many years and no one seems to have complained before. This issue aims to find out the community's opinion, whether anyone has any objections to proposed change :)

michallenc, 2025-01-08:
Hi all, just please note that this might be a breaking API change, since it is safe to assume applications are using `CONFIG_PWM_MULTICHAN` to determine how to access `pwm_info_s` structure. That said, I completely agree with the removal of `CONFIG_PWM_MULTICHAN` option.

I have proposed other breaking change several months ago in https://github.com/apache/nuttx/issues/12381 that simplifies `pwm_info_s` and would possibly also make `CONFIG_PWM_MULTICHAN` obsolete. I have not done much work on it as it is not my priority now, just some initial tests with SAMv7 build, you can see this commit https://github.com/michallenc/incubator-nuttx/commit/8224f0e70e18ee643e08821fd66724a2ff14d616 (though I still used multichan config there, but it shouldn't be problem to remove it). We would basically shrink the entire PWM structure to

```
#ifdef CONFIG_PWM_NCHANNELS
#define PWM_NCHANNELS CONFIG_PWM_NCHANNELS
#else
#define PWM_NCHANNELS 1
#endif

struct pwm_chan_s  channels[PWM_NCHANNELS];
FAR void           *arg; 
``` 

There were also some other proposals in the issue I haven't worked on yet, this is just an initial kick off.

raiden00pl, 2025-01-10:
@michallenc I completely forgot about the issue you created :) so there are at least 2 people who see this problem, so I think it's worth fixing.

I'm closing this issue, we can track the problem in https://github.com/apache/nuttx/issues/12381

closed 2025-01-10 by raiden00pl