apache/nuttx #15446: [FEATURE] mechanism to ensure the consistent locking order

yamt, 2025-01-07:
### Is your feature request related to a problem? Please describe.

it would be nice to have some debug feature which can detect issues like https://github.com/apache/nuttx/issues/15314

### Describe the solution you'd like

precedents:
https://man.freebsd.org/cgi/man.cgi?witness(4)
https://docs.kernel.org/locking/lockdep-design.html

### Describe alternatives you've considered

althogh adhoc local assertions like https://github.com/yamt/incubator-nuttx/commit/ac3dda4f4f85d5774a5e05ec0595fe572110c947
can work, it would be nicer to have a generic mechanism in the upstream.

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2025-01-07:
@yamt can you try https://github.com/apache/nuttx/blob/master/sched/misc/deadlock.c?

yamt, 2025-01-07:
> @yamt can you try https://github.com/apache/nuttx/blob/master/sched/misc/deadlock.c?

glancing at the code, it requires you to actually make a deadlock happen, and then somehow trigger an assertion failure.
is it right?

reproducing a deadlock is not always easy.
well, if you know how to reproduce a deadlock, i guess you likely already know the problem well. just fix it. :-)

what i want to see is something more aggressive.
eg. detect locking attempts with a wrong order. (in most cases it might not actually cause a deadlock. but it _can_ cause a deadlock.)

xiaoxiang781216, 2025-01-07:
> > @yamt can you try https://github.com/apache/nuttx/blob/master/sched/misc/deadlock.c?
> 
> glancing at the code, it requires you to actually make a deadlock happen, and then somehow trigger an assertion failure. is it right?
> 

Yes, the code assume that the deadlock will trigger panic finally. but it's easy to expose a nsh command to trigger the check periodly.

> reproducing a deadlock is not always easy. well, if you know how to reproduce a deadlock, i guess you likely already know the problem well. just fix it. :-)
> 

in a complex system, it's hard to know which mutexs and threads participant the deadlock chain, this tool help people identify the problem mutex and thread after the deadlock happen.

> what i want to see is something more aggressive. eg. detect locking attempts with a wrong order. (in most cases it might not actually cause a deadlock. but it _can_ cause a deadlock.)

this feature requires to record the locking order.

yamt, 2025-01-07:
> > what i want to see is something more aggressive. eg. detect locking attempts with a wrong order. (in most cases it might not actually cause a deadlock. but it _can_ cause a deadlock.)
> 
> this feature requires to record the locking order.

yes.