apache/nuttx #15456: [BUG] ESP32C6 DEV Kit M/C Gpio issue

FelipeMdeO, 2025-01-07:
### Description / Steps to reproduce the issue

Hello All,

I cannot control GPIO 18 and 21 output level. I was not test all pins, but the other one that I tested is ok.
Also, I am able to control GPIO 18 and 21 using esp IDF.

[image]

@tmedicci  can you try reproduce this behavior, please ?

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24

### NuttX Version

master'

### Issue Architecture

[Arch: risc-v]

### Issue Area

[Area: Drivers]

### Verification

- [X] I have verified before submitting the report.

acassis, 2025-01-07:
@tmedicci @fdcavalcanti @eren-terzioglu any idea about this issue?

eren-terzioglu, 2025-01-08:
I will have a look, thanks.

FelipeMdeO, 2025-01-08:
Hello @eren-terzioglu, 

The issue happen in gpio 4 too.
[image]

eren-terzioglu, 2025-01-09:
Hi,
Could you use https://github.com/apache/nuttx/pull/15480 PR to test it. It should work

closed 2025-01-09 by xiaoxiang781216