apache/nuttx #15464: [BUG] many of _POSIX_xxx constants seem inappropriate

yamt, 2025-01-08:
### Description / Steps to reproduce the issue

limits.h has the "Configurable limits required by POSIX" section
with _POSIX_xxx and _POSIX2_xxx constants.

for example,
```
#define _POSIX_NAME_MAX       CONFIG_NAME_MAX
```

it seems inappropriate to me as these constants are not meant to be configurable.

My understanding is that these values are the minimum values allowed for POSIX-conforming platforms.
For example, _POSIX_PATH_MAX is the minimum acceptable value for PATH_MAX and so on.
Thus these values should not depend on any NuttX configs at all.

reference:
search "Minimum Acceptable Value" in https://pubs.opengroup.org/onlinepubs/9799919799/basedefs/limits.h.html

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS 15.2

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Api]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2025-01-22:
why POSIX constant shouldn't be configurable from Kconfig? it perfect to allow the different OS or even the same OS with the different configuration choice a different value.

yamt, 2025-01-27:
> why POSIX constant shouldn't be configurable from Kconfig? it perfect to allow the different OS or even the same OS with the different configuration choice a different value, otherwise, spec should give a fix number.

_POSIX_NAME_MAX is the Minimum Acceptable Value of NAME_MAX.
actually posix gives a fixed number. (14)

otoh, it's ok to make NAME_MAX depend on Kconfig.

closed 2026-10-18 by acassis