apache/nuttx #15491: [BUG] Mirror of NuttX:  macOS Jobs are in a bad state

simbit18, 2025-01-10:
### Description / Steps to reproduce the issue

All Jobs are broken !!! https://github.com/NuttX/nuttx/actions/runs/12685758124

example
```
====================================================================================
Configuration/Tool: stm32f4discovery/nsh,CONFIG_ARM_TOOLCHAIN_GNU_EABI
2025-01-09 08:26:57
------------------------------------------------------------------------------------
  Cleaning...
  Configuring...
make: *** [olddefconfig] Error 1
  Enabling CONFIG_ARM_TOOLCHAIN_GNU_EABI
make: *** [olddefconfig] Error 1
  Building NuttX...
arm-none-eabi-gcc: error: missing argument to '-Wstack-usage='
arm-none-eabi-gcc: error: missing argument to '-Wstack-usage='
ERROR: arm-none-eabi-gcc failed: 1
ERROR: arm-none-eabi-gcc failed: 1
```

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

MacOS at GitHub Actions

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Build System]

### Verification

- [X] I have verified before submitting the report.

larsks, 2025-01-12:
I'm seeing the same errors attempting to build under Linux with:

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (crosstool-NG 1.26.0.106_ed12fa6 - qmk/qmk_toolchains @ a063b96) 14.2.0
```

xiaoxiang781216, 2025-01-22:
@simbit18 does macOS restore now?

simbit18, 2025-01-22:
Hi @xiaoxiang781216  macOS (sim-02) and macOS (sim-03) still need fixing

https://github.com/NuttX/nuttx/actions/runs/12905015657/job/35983441494#logs

https://github.com/NuttX/nuttx/actions/runs/12905015657/job/35983441951#logs

cuiziweizw, 2025-02-05:
@xiaoxiang781216 @lupyuen The current problem is caused by https://github.com/apache/nuttx/issues/14774, maybe directly selecting libcxxabi in Kconfig is not a better choice, but changing the configuration in defconfig is better

closed 2025-02-12 by acassis