apache/nuttx #15502: [HELP] STM32H7 timer driver usage

geniusdo, 2025-01-12:
### Description

I tried to use timer driver based on this page [timer](https://nuttx.apache.org/docs/12.7.0/components/drivers/character/timers/timer.html#enabling-the-timer-support-and-example-in-menuconfing). But when I toggled the option TIMER ARCH IMPLEMENTATION, my nsh just stucked. Any more detailed do I need to know to get this through? Any helps will be appreciated!

### Verification

- [X] I have verified before submitting the report.

raiden00pl, 2025-01-15:
there is no support for `CONFIG_TIMER_ARCH` in `stm32h7` and there is no support for `CONFIG_TIMER` in any of upstream `stm32h7` boards. You can see on `stm32f4discovery` how this can be done on board side:

https://github.com/apache/nuttx/blob/master/boards/arm/stm32/stm32f4discovery/src/stm32_timer.c and https://github.com/apache/nuttx/blob/46c3f354b38291ee400bdbece6945ae6ab883b29/boards/arm/stm32/stm32f4discovery/src/stm32_bringup.c#L364C15-L372

FYI: timer drivers in stm32 don't implement `getstatus()` operation required by `TCIOC_GETSTATUS` used in `examples/timer`, so this example won't work properly with stm32 chips.

geniusdo, 2025-01-31:
I can see timer drivers are usually coupled with hardware interrupt. I am a bit confused with interrupt implementation in nuttx. Are there any materials for me to understand how to implement an hardware interrupt?