apache/nuttx #15526: [BUG] Static Char Arrays are empty for NuttX Apps compiled for rv-virt:knsh and knsh64

lupyuen, 2025-01-14:
### Description / Steps to reproduce the issue

When we print a Static Char Array in a NuttX App: [hello_main.c](https://github.com/lupyuen2/wip-nuttx-apps/blob/uname/examples/hello/hello_main.c#L31-L65)

```c
static char test_static[] = "Testing Static Var";
int main(int argc, FAR char *argv[]) {
  printf("test_static=%s\n", test_static);
  printf("Address of test_static=%p\n", test_static);
  return 0;
}
```

The Static Char Array appears as Empty for `rv-virt:knsh` and `rv-virt:knsh64`:

```text
$ qemu-system-riscv32 -semihosting -M virt,aclint=on -cpu rv32 -kernel nuttx -bios opensbi-1.5-rv-bin/share/opensbi/ilp32/generic/firmware/fw_dynamic.bin -nographic
QEMU emulator version 9.2.0
OpenSBI v1.5

NuttShell (NSH) NuttX-12.8.0
nsh> uname -a
NuttX 12.8.0  risc-v rv-virt

nsh> hello
test_static=
Address of test_static=0xc0100200
```

[rv-virt:knsh Log](https://gist.github.com/lupyuen/df3558ca3d676a2cd0f4da9c26827cfa#file-special-qemu-riscv-knsh-log-L1406-L1437)

[rv-virt:knsh64 Log]https://gist.github.com/lupyuen/3212d8c2cee7e6055db9b1939a02ac20#file-special-qemu-riscv-knsh64-log-L1391-L1422)

See the discussion here:
- https://github.com/apache/nuttx/pull/15444#issuecomment-2586160111

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS 15.2 24C101 arm64 

### NuttX Version

master

### Issue Architecture

[Arch: risc-v]

### Issue Area

[Area: Applications]

### Verification

- [x] I have verified before submitting the report.

xiaoxiang781216, 2025-01-14:
@anjiahao1 please look at this problem.

anjiahao1, 2025-01-14:
https://github.com/apache/nuttx/pull/15527 @lupyuen @yf13  hello, take look this pr, this is rootcuase