apache/nuttx #15582: [BUG] sim_vpnkit is breaking watchdog?

casaroli, 2025-01-16:
### Description / Steps to reproduce the issue

Hello,

I don't see anything wrong with the commit 61f0c97193e3adbb1eee07bb854799d35843f12e, however, it seems to have broken sim_vpnkit.

I come up with a minimal example, that we can reproduce even without running vpnkit.

When NuttX starts, it will try to reach vpnkit socket, and after some timeout it will proceed to nsh:

## Build and run:

```sh
./tools/configure.sh -l sim:vpnkit
make
./nuttx
[... no output even after 10 seconds ...]
```

Running current master (118f8271038afe24732dfc4376fb63b72e039dba) gives no output, and running on the debugger we can see the wd has expired:

```
Breakpoint 1, _assert (filename=0x400789f8 <host_gettime+36> "H\213D$\020Hi\300", linenum=32767, 
    msg=0x7ffff7d8853d <__GI___clock_gettime+29> "\205\300u\0371\300]\303\017\037", regs=0x7fffffffd9c0) at misc/assert.c:826
826	{
(gdb) bt
#0  _assert (filename=0x400789f8 <host_gettime+36> "H\213D$\020Hi\300", linenum=32767, 
    msg=0x7ffff7d8853d <__GI___clock_gettime+29> "\205\300u\0371\300]\303\017\037", regs=0x7fffffffd9c0) at misc/assert.c:826
#1  0x000000004000c60b in __assert (filename=0x4007e7ff "wqueue/kwork_queue.c", linenum=71, msg=0x0) at assert/lib_assert.c:38
#2  0x0000000040044bb2 in work_timer_expiry (arg=1074361008) at wqueue/kwork_queue.c:71
#3  0x0000000040044ae6 in wd_expiration (ticks=4) at wdog/wd_start.c:159
#4  wd_timer (ticks=4) at wdog/wd_start.c:468
#5  0x0000000040042bec in nxsched_process_timer () at sched/sched_processtimer.c:197
#6  0x000000004000b940 in oneshot_callback (lower=0x7ffff3c00398, arg=0x0) at timers/arch_alarm.c:131
#7  0x00000000400139bc in sim_process_tick (entry=0x7ffff3c003a0) at sim/sim_oneshot.c:254
#8  0x00000000400138a6 in sim_timer_update_internal () at sim/sim_oneshot.c:200
#9  0x0000000040013d6a in sim_timer_update () at sim/sim_oneshot.c:530
#10 0x0000000040012a97 in up_idle () at sim/sim_idle.c:86
#11 0x0000000040001a6b in nx_start () at init/nx_start.c:782
#12 0x00000000400014df in main (argc=1, argv=0x7fffffffdd48, envp=0x7fffffffdd58) at sim/sim_head.c:187
```

## Expected result:

If we revert that commit, it works:

```sh
git revert --no-commit 61f0c97193e3adbb1eee07bb854799d35843f12e
make
./nuttx
[... after about 10 seconds ...]

NuttShell (NSH) NuttX-12.8.0
nsh> 
```

I don't know how 61f0c97193e3adbb1eee07bb854799d35843f12e could affect `sim_netdriver.c`, kthreads or kernel workqueues.

Can anyone please give us a clue?

Thank you!

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu-24

### NuttX Version

master

### Issue Architecture

[Arch: simulator]

### Issue Area

[Area: OS Components], [Area: Networking], [Area: Kernel]

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-01-16:
@hujun260 please take a look ^

hujun260, 2025-01-17:
https://github.com/apache/nuttx/pull/15586

closed 2025-01-17 by xiaoxiang781216