apache/nuttx #15592: [BUG] nuttx/ioexpander/mcp23x08.h is missing

paolovolpi, 2025-01-17:
### Description / Steps to reproduce the issue

Enable DeviceDrivers/IO Expander Gpio Support/MCP23008
make -j
 got the following compilation error:
CC:  environ/env_getenv.c In file included from ioexpander/mcp23x08.c:38:
ioexpander/mcp23x08.h:37:10: fatal error: nuttx/ioexpander/mcp23x08.h: No such file or directory
   37 | #include <nuttx/ioexpander/mcp23x08.h>

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.04

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Specific Peripheral]

### Verification

- [x] I have verified before submitting the report.

xiaoxiang781216, 2025-01-17:
but here is it: https://github.com/apache/nuttx/blob/master/drivers/ioexpander/mcp23x08.h

paolovolpi, 2025-01-17:
But it's not the required header, as an example take  a look to mcp23x17, there is an header under drivers/ioexpander and another under include/nuttx/ioexpander.

acassis, 2025-01-17:
@paolovolpi good finding, actually it was missing in my PR: https://github.com/apache/nuttx/commit/4b67e23049f2f4904a79adafdcc0d3d06d29555a

Could you please use mcp23x17.h as base and create the mcp23x08.h? If you need help, please ping me or we can discuss it at Discord: https://discord.gg/YaqhJjHF95

closed 2025-01-20 by xiaoxiang781216