apache/nuttx #15595: [BUG] adb is not working on STM32F4Discovery board

acassis, 2025-01-18:
### Description / Steps to reproduce the issue

$ ./tools/configure.sh stm32f4discovery:adb

$ make -j

$ sudo openocd -f interface/stlink.cfg -f target/stm32f4x.cfg -c init -c "reset halt" -c "flash write_image erase nuttx.bin 0x08000000"

$ adb devices
List of devices attached
1234	offline

The adb0 device files are present:

nsh> ls /dev/adb0/
/dev/adb0:
 ep0
 ep1
 ep2

I don't know if adbd is running, but I think it is because there are many there are 3 Kthread and 1 Task:

nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK            STACK    USED FILLED
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 0002032 0000460  22.6k
    1     0 224 FIFO     Kthread   - Waiting  Semaphore 0000000000000000 0001984 0000368  18.5c
    2     0 100 FIFO     Kthread   - Waiting  Semaphore 0000000000000000 0001984 0000264  13.30
    3     3 100 FIFO     Task      - Running            0000000000000000 0003032 0001536  50.6n
nsh>

Note: there is not Documentation/ to adb on STM32F4Discovery

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.04

### NuttX Version

mainline: 2779989add104b8243b6d06bd187b1a9927983cd

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Applications]

### Verification

- [x] I have verified before submitting the report.

xiaoxiang781216, 2025-01-18:
do you luanch adbd in nsh?

JianyuWang0623, 2025-01-18:
It looks like the enumeration was successful. Maybe you need check wether adbd is launched, as @xiaoxiang781216 suggested. And here is an example enabling ADB for Espressif board: https://github.com/apache/nuttx/pull/14906

acassis, 2025-01-20:
@xiaoxiang781216 @JianyuWang0623 running adbd changed the status from "offline" to "device"

```
alan@dev:~$ adb devices
List of devices attached
1234	device
```

However, when I ran "adb shell" the system crashed: 

```
alan@dev:~$ adb shell

...

NuttShell (NSH) NuttX-12.8.0
nsh> ?
help usage:  help [-v] [<cmd>]

    .           cmp         false       mkfifo      rmdir       unset       
    [           dirname     fdinfo      mkrd        set         uptime      
    ?           dd          free        mount       sleep       usleep      
    alias       df          help        mv          source      watch       
    unalias     dmesg       hexdump     pidof       test        xd          
    basename    echo        kill        printf      time        wait        
    break       env         pkill       ps          true        
    cat         exec        ls          pwd         truncate    
    cd          exit        mkdir       reboot      uname       
    cp          expr        mkfatfs     rm          umount      
                                                                                
Builtin Apps:                                                                   
    adbd    nsh     sh                                                          
nsh> adbd                                                                       
adb_register_service (338): id=1, peer=8                                        
dump_assert_info: Current Version: NuttX  12.8.0 2779989add-dirty Jan 17 2025 2m
dump_assert_info: Assertion failed panic: at file: :0 task: adbd process: adbd 5
up_dump_register: R0: 0000000a R1: 080055fd R2: 20000bd8  R3: 0801febf          
up_dump_register: R4: 00000080 R5: 08000603 R6: 0000000a  FP: 0800562f          
up_dump_register: R8: 0000000a SB: 080055fd SL: 20000bc0 R11: 0801febf          
up_dump_register: IP: 00000000 SP: 0000000a LR: 08017feb  PC: 20000c34          
up_dump_register: xPSR: 0a00000a BASEPRI: 0020000a CONTROL: 00000000            
up_dump_register: EXC_RETURN: 0000000a                                          
dump_stacks: ERROR: Stack pointer is not within the stack                       
dump_stackinfo: User Stack:                                                     
dump_stackinfo:   base: 0x20005770                                              
dump_stackinfo:   size: 00002016                                                
stack_dump: 0x20005d38: ffffffff 20007070 ffffffff 20007488 20002210 0800607f 08
stack_dump: 0x20005d58: ffffffff 20006fd0 20006fd0 0800607f 00000000 20006fd0 27
stack_dump: 0x20005d78: 00000000 080058fb 80000000 20005dec 00000001 20007040 20
stack_dump: 0x20005d98: 00000401 0800601b ffffffff 00000000 200001f0 0800607f 00
stack_dump: 0x20005db8: 200001f0 200001f0 00000402 0800601b ffffffff 00000000 2f
stack_dump: 0x20005dd8: 20005fb8 00000000 20004944 080206f5 00000000 20005e4c 03
stack_dump: 0x20005df8: 20005e70 20005ead 20005fb8 20005f88 20005318 00000402 20
stack_dump: 0x20005e18: 2000754c 2000754c 20005e50 00000000 200060cc 00000001 09
stack_dump: 0x20005e38: 20007530 080195a3 20007af0 0801cdff 20007694 0801bbf5 20
stack_dump: 0x20005e58: 00000004 2000761c 200075d8 00000004 00000000 0801bdc3 20
stack_dump: 0x20005e78: 0000000c 20001998 200060a8 00000000 00000000 000000ff 28
stack_dump: 0x20005e98: ffffffff 00000000 000000ff 200060cc 00000001 00000000 07
stack_dump: 0x20005eb8: 00000001 00000000 20001998 0007a120 00000000 00000000 03
stack_dump: 0x20005ed8: 20001998 00000000 200060a8 08019491 00000000 0000000d 00
stack_dump: 0x20005ef8: 00000000 00000000 00000000 00000000 00000000 00000000 00
stack_dump: 0x20005f18: 00000000 00000000 00000000 00000000 08019455 00000001 21
stack_dump: 0x20005f38: 08019455 08002c41 00000000 00000000 00000000 00000000 00
dump_tasks:    PID GROUP PRI POLICY   TYPE    NPX STATE   EVENT      SIGMASK   D
dump_task:       0     0   0 FIFO     Kthread -   Ready              0000000000k
dump_task:       1     0 224 FIFO     Kthread -   Waiting Semaphore  0000000000c
dump_task:       2     0 100 FIFO     Kthread -   Waiting Semaphore  00000000000
dump_task:       3     3 100 FIFO     Task    -   Waiting Semaphore  0000000000n
dump_task:       4     4 100 FIFO     Task    -   Running            0000000000d
dump_task:       5     5 100 FIFO     Task    -   Ready              0000000000h
[... 41 lines cut ...]
 CONFIG_NSH_ARCHINIT=y
 CONFIG_NSH_BUILTIN_APPS=y
 CONFIG_NSH_FILEIOSIZE=512
 CONFIG_NSH_READLINE=y
+CONFIG_POSIX_SPAWN_DEFAULT_STACKSIZE=2048
 CONFIG_PREALLOC_TIMERS=4
 CONFIG_PSEUDOTERM=y
 CONFIG_RAM_SIZE=114688
 CONFIG_RAM_START=0x20000000
 CONFIG_RAW_BINARY=y
+CONFIG_SCHED_HPWORKSTACKSIZE=2048
 CONFIG_SCHED_LPWORK=y
+CONFIG_SCHED_LPWORKSTACKSIZE=2048
 CONFIG_SCHED_WAITPID=y
 CONFIG_SENSORS=y
 CONFIG_STACK_COLORATION=y
@@ -66,6 +76,7 @@
 CONFIG_STM32_USART2=y
 CONFIG_SYSTEM_ADBD=y
 CONFIG_SYSTEM_NSH=y
+CONFIG_SYSTEM_NSH_STACKSIZE=2048
 CONFIG_TLS_TASK_NELEM=4
 CONFIG_USART2_SERIAL_CONSOLE=y
 CONFIG_USBADB=y
```

acassis, 2025-01-20:
Sometimes when I run adbd I got this error:

```
nsh> adbd &
adbd [4:100]
nsadb_uv_on_data_availableh >( 204): bad data: terminated (data)
db_hal_run (77): uv_loop exit 0
```

acassis, 2025-01-20:
Sometimes "adbd" crashes even before running "adb shell", just running it:

```
NuttShell (NSH) NuttX-12.8.0
nsh> adbd
dump_assert_info: Current Version: NuttX  12.8.0 2779989add-dirty Jan 17 2025 21:28:59 arm
dump_assert_info: Assertion failed panic: at file: :0 task: adbd process: adbd 0x8019455
up_dump_register: R0: 00000053 R1: 00000000 R2: 00000002  R3: 0000007e
up_dump_register: R4: 00000053 R5: 00000001 R6: 0000000b  FP: 20000f40
up_dump_register: R8: 20006408 SB: 00000001 SL: 00000000 R11: 20000f50
up_dump_register: IP: f4240000 SP: 20000e58 LR: 08000ba1  PC: 0000007e
up_dump_register: xPSR: 80000053 BASEPRI: 00000000 CONTROL: 00000000
up_dump_register: EXC_RETURN: fffffff1                                                                                                                           
dump_stacks: ERROR: Stack pointer is not within the stack                                                                                                        
dump_stackinfo: User Stack:                                                                                                                                      
dump_stackinfo:   base: 0x20005830                                                                                                                               
dump_stackinfo:   size: 00002016                                                                                                                                 
stack_dump: 0x20005df8: ffffffff 20007070 ffffffff 20007488 20002210 0800607f 00000000 20007488                                                                  
stack_dump: 0x20005e18: ffffffff 20006fd0 20006fd0 0800607f 00000000 20006fd0 20006fd0 080060e7                                                                  
stack_dump: 0x20005e38: 00000000 080058fb 80000000 20005eac 00000001 20007040 200001f0 200001f0                                                                  
stack_dump: 0x20005e58: 00000401 0800601b ffffffff 00000000 200001f0 0800607f 00000000 00000000                                                                  
stack_dump: 0x20005e78: 200001f0 200001f0 00000402 0800601b ffffffff 00000000 20004948 0800607f                                                                  
stack_dump: 0x20005e98: 200052f0 00000000 20004944 080206f5 00000000 20005f0c 00000402 0800f8b3                                                                  
stack_dump: 0x20005eb8: 20005f30 20005f6d 200052f0 200052c0 200053d8 00000402 20002210 20002210                                                                  
stack_dump: 0x20005ed8: 2000754c 2000754c 20005f10 00000000 200060cc 00000001 00000000 0801bf49                                                                  
stack_dump: 0x20005ef8: 20007530 080195a3 08019589 20007694 20007694 0801bbf5 20005f10 20005f10                                                                  
stack_dump: 0x20005f18: 00000004 2000761c 200075d8 00000004 00000000 0801bdc3 2000754c 00000000                                                                  
stack_dump: 0x20005f38: 0000000c 20001998 200060a8 00000000 00000000 000000ff 200060cc 200060a8                                                                  
stack_dump: 0x20005f58: ffffffff 00000000 000000ff 200060cc 00000001 00000000 00000000 08019b57                                                                  
stack_dump: 0x20005f78: 00000001 00000000 20001998 0007a120 00000000 00000000 00000000 08019503                                                                  
stack_dump: 0x20005f98: 20001998 00000000 200060a8 08019491 00000000 0000000d 000021b6 00000000                                                                  
stack_dump: 0x20005fb8: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000                                                                  
stack_dump: 0x20005fd8: 00000000 00000000 00000000 00000000 08019455 00000001 20005820 08006171                                                                  
stack_dump: 0x20005ff8: 08019455 08002c41 00000000 00000000 00000000 00000000 00000000 00000000                                                                  
dump_tasks:    PID GROUP PRI POLICY   TYPE    NPX STATE   EVENT      SIGMASK          STACKBASE  STACKSIZE      USED   FILLED    COMMAND                         
dump_task:       0     0   0 FIFO     Kthread -   Ready              0000000000000000 0x20001a1c      2032       696    34.2%    Idle_Task                       
dump_task:       1     0 224 FIFO     Kthread -   Waiting Semaphore  0000000000000000 0x20002680      1984       368    18.5%    hpwork 0x2000010c 0x2000013c    
dump_task:       2     0 100 FIFO     Kthread -   Waiting Semaphore  0000000000000000 0x20002f50      1984       264    13.3%    lpwork 0x200000c0 0x200000f0    
dump_task:       3     3 100 FIFO     Task    -   Waiting Semaphore  0000000000000000 0x20003b78      3032      1472    48.5%    nsh_main                        
dump_task:       4     4 100 FIFO     Task    -   Running            0000000000000000 0x20005830      2016       536    26.5%    adbd                            

```

acassis, 2025-01-20:
Because PC is equal 0x0000007e, it means some allocation failed and it is trying to access the field of a structure that was not allocated (point to 0/NULL).

The LR was here:
```
alan@dev:~/nuttxspace/nuttx$ arm-none-eabi-addr2line -e nuttx 08000ba1
/home/alan/nuttxspace/nuttx/arch/arm/src/armv7-m/arm_doirq.c:54
```

This line:

```
  if (g_running_tasks[this_cpu()] != this_task())
```

JianyuWang0623, 2025-01-20:
@acassis 
It seems that the value of `CONFIG_ADBD_STACKSIZE` is still 2048, could you increase it (e.g. to 4KB or 8KB) and try again?
Currently `CONFIG_ARCH_INTERRUPTSTACK` is zero for "stm32f4discovery:adb", the user task stacks will be used during interrupt handling.

acassis, 2025-01-20:
@JianyuWang0623 increasing ADBD stack to 8KB and INT STACK to 2KB avoid the crash, but when I run "adb shell" I get:

```
NuttShell (NSH) NuttX-12.8.0                                                                                                                                     
nsh> adbd                                                                                                                                                        
adb_check_frame_magic (27): invalid frame magic                                                                                                                  
adb_uv_on_data_available (188): bad header: terminated (data)                                                                                                    
adb_hal_run (77): uv_loop exit 0
```

JianyuWang0623, 2025-01-20:
@acassis The UDC(USB Device Controller) driver may have set the ready flag when the host has not sent the complete package.
The root cause may need to be analyzed in conjunction with the URB(USB Request Block) and the driver source code.