apache/nuttx #15632: [HELP] NuttX Make issues inside of Docker container

stbenn, 2025-01-22:
### Description

I am creating a process to have easily reproducible development environments for my various NuttX projects. The solution I have come up with uses a very simple Docker container to manage dependencies and Makefile to automate the container setup process.

The Dockerfile is below, I build the image with: `docker build -t nuttx_container .`
I was originally starting the container with: `docker run --name=nuttx_experiment -rm -it --privileged -v $(pwd):/usr/src/project nuttx_container`
Then I switched the command to this: `docker run --name=nuttx_experiment -rm -it --privileged -v $(pwd):$(pwd) nuttx_container`

### Problem
After switching the directory the volume is mounted to, I now receive this error whenever I do a make command in either the container or the host directory that is mounted:
```
tools/Unix.mk:32: /home/tbennett/NuttxDockerSpace/nuttx/Make.defs: No such file or directory
make: *** No rule to make target `/home/tbennett/NuttxDockerSpace/nuttx/Make.defs'. Stop.
```

Is there a way this issue can be avoided or easily remedied (without deleting and re-cloning the nuttx directory)?

### Ideal Result
It would be fantastic if I could call `make clean` or `make distclean` from either the host or the container. I would prefer to not have to spin up the container just to remove all the temporary files. 

Note: Eventually I plan on switching to run with a specific developer user in the container instead of using root.

## Project Setup
Using Ubuntu:24.04 (WSL2) on Windows 11, so the code lives there and then is mounted into the container. 

Project Directory Structure
```Markdown
project_nuttxspace
├── apps
│   ├── clone of nuttx-apps
├── nuttx
│   ├── clone of apache/nuttx
│   ├── Makefile
│   └── Make.defs
├── Dockerfile
└── Makefile
```

Dockerfile
```Dockerfile
FROM ubuntu:24.04
LABEL Description="Nuttx Development Environment"

RUN apt-get update && apt-get -y install \
build-essential git gcc gcc-multilib \
bison flex gettext texinfo libncurses5-dev libncursesw5-dev xxd \
gperf automake libtool pkg-config genromfs \
libgmp-dev libmpc-dev libmpfr-dev libisl-dev binutils-dev libelf-dev \
libexpat1-dev gcc-multilib g++-multilib picocom u-boot-tools util-linux \
kconfig-frontends gcc-arm-none-eabi binutils-arm-none-eabi
```

### Verification

- [x] I have verified before submitting the report.

fdcavalcanti, 2025-01-22:
You can find some Dockerfile examples online and even [here](https://github.com/apache/nuttx/blob/master/tools/ci/docker/linux/Dockerfile). Use those as examples.

To apply modifications with your local user and from inside Docker, you need to have proper permissions, otherwise it won't work. Usually we pass to the running image the UID and GID. There's plenty of examples online.

Regarding the issue you had, try `git clean -xfd` on `nuttx` and `apps` repositories and then try again. This will bring everything back to initial state (by deleting every change, so be aware of that) and you don't have to clone again.

acassis, 2025-01-22:
@stbenn try to just remove .config and run "./tools/configure.sh" again! Normally it is enough to fix directory modifications

closed 2025-01-27 by stbenn