apache/nuttx #15689: [HELP] Questions about NuttX on the ESP32

artdeell, 2025-01-24:
### Description

I have no proper RTOS and ESP32 experience (but I have worked with an ESP8266 on a basic level before), and I wanted to try NuttX. So i wanted to ask some newbie questions that i didn't find answers in the documentation:
1. How to properly use `make menuconfig`? Yes, the menu entries have clear names, but it seems like it's very easily possible to get yourself into an un-buildable configuration.
2. Is it possible to have Wi-Fi and SMP enabled at the same time on the ESP32?
3. What is the best version of NuttX to use on the ESP32 with SMP enabled? Reading other issues, there seem to be some problems with Espressif devices right now.

### Verification

- [x] I have verified before submitting the report.

tmedicci, 2025-01-24:
Hi @artdeell !

Glad to hear that you are considering using the ESP32 on NuttX. Regarding your questions:

> * How to properly use `make menuconfig`? Yes, the menu entries have clear names, but it seems like it's very easily possible to get yourself into an un-buildable configuration.

The easiest way to navigate through `make menuconfig` is by loading an existing `defconfig` and enabling/disabling new configs.

So, basically:

1. Install the required packages to build NuttX as described [here](https://nuttx.apache.org/docs/latest/quickstart/install.html).
2. Install the related toolchain. For ESP32, you can check it [here](https://nuttx.apache.org/docs/latest/platforms/xtensa/esp32/index.html#esp32-toolchain).
3. Set an existing `defconfig` as described [here](https://nuttx.apache.org/docs/latest/quickstart/compiling_make.html#initialize-configuration).
4. Build it and validate that everything is working as expected, as described [here](https://nuttx.apache.org/docs/latest/platforms/xtensa/esp32/index.html#building-and-flashing).
5. Then, run `make menuconfig` and try different settings.

Yes, you can get yourself into an un-buildable configuration. This is not a problem specific for NuttX. NuttX's developers try to avoid it by managing the requirements related to each configuration. Don't forget to report it if you find something that doesn't build properly.

> * What is the best version of NuttX to use on the ESP32 with SMP enabled? Reading other issues, there seem to be some problems with Espressif devices right now.

We have no issues using SMP on ESP32 and ESP32-S3 (dual-core SoCs supported by NuttX). You can check the `sta_softap` example that enables both station and soft-AP modes for the Wi-Fi and the SMP.

Please don't forget to close this issue if you find the information you need. If the documentation is not clear enough, please report it (or, even better, submit a PR to make it clearer and help others ;) ). Thanks!

artdeell, 2025-01-24:
Alright, thank you! I'll keep exploring and see what I find.

closed 2025-01-24 by artdeell