apache/nuttx #15692: [BUG] Cannot enable separate IRAM heap and SMP at the same time

artdeell, 2025-01-25:
### Description / Steps to reproduce the issue

When enabling ESP32_IRAM_HEAP on the default esp32-devkitc:smp configuration, the resulting build cannot start.

Serial output:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffb40e0,len:1420
ho 0 tail 12 room 4
load:0x40080000,len:22760
entry 0x40082884
*** Booting NuttX ***
dram: lma 0x00001020 vma 0x3ffb40e0 len 0x58c    (1420)
iram: lma 0x000015b4 vma 0x40080000 len 0x58e8   (22760)
padd: lma 0x00006ea8 vma 0x00000000 len 0x9150   (37200)
imap: lma 0x00010000 vma 0x400e0000 len 0x1abf8  (109560)
padd: lma 0x0002ac00 vma 0x00000000 len 0x5418   (21528)
dmap: lma 0x00030020 vma 0x3f400020 len 0x96c4   (38596)
total segments stored 6
[CPU0] xtensa_user_panic: User Exception: EXCCAUSE=0003 task: CPU0 IDLE
[CPU0] dump_assert_info: Current Version: NuttX  12.7.0 5d8cdeae-dirty Jan 25 2025 13:46:53 xtensa
[CPU0] dump_assert_info: Assertion failed user panic: at file: :0 task(CPU0): CPU0 IDLE process: Kernel 0x400e1d64
[CPU0] up_dump_register:    PC: 400fa1a7    PS: 00060930
[CPU0] up_dump_register:    A0: 800e44d4    A1: 3ffb0b20    A2: 400858e8    A3: 00000000
[CPU0] up_dump_register:    A4: 00000001    A5: fffffffc    A6: 00000031    A7: 00000000
[CPU0] up_dump_register:    A8: 00000000    A9: 3ffb0b00   A10: 00000000   A11: 00000000
[CPU0] up_dump_register:   A12: 00060720   A13: 00007fff   A14: 00a0a000   A15: b0000000
[CPU0] up_dump_register:   SAR: 00000020 CAUSE: 00000003 VADDR: 400858e8
[CPU0] up_dump_register:  LBEG: 400e5934  LEND: 400e5942  LCNT: 00000000
[CPU0] dump_stacks: ERROR: Stack pointer is not within the stack
[CPU0] dump_stack: IRQ Stack:
[CPU0] dump_stack:   base: 0x3ffb1cec
[CPU0] dump_stack:   size: 00002048
[CPU0] stack_dump: 0x3ffb1cfc: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1d1c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1d3c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1d5c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1d7c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1d9c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1dbc: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1ddc: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1dfc: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1e1c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1e3c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1e5c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1e7c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1e9c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1ebc: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1edc: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1efc: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1f1c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1f3c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1f5c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1f7c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1f9c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1fbc: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1fdc: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb1ffc: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb201c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb203c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb205c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb207c: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb209c: 00000109 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb20bc: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[CPU0] stack_dump: 0x3ffb20dc: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[... 41 lines cut ...]
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Debian Sid (unstable)

### NuttX Version

12.7.0 (dirty due to bypassed esptool check)

### Issue Architecture

[Arch: xtensa]

### Issue Area

[Area: Kernel]

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-01-26:
@eren-terzioglu @tmedicci @fdcavalcanti could you please take a look?

eren-terzioglu, 2025-01-27:
Hi,
Which version of esptool you are using? Seems you bypassed version check due to https://github.com/apache/nuttx/issues/15690 issue. esptool v4.7 might cause the issue. Please create a venv and try to flash it with required version of esptool.

artdeell, 2025-01-27:
I use esptool 4.8.x Odont remember the exact version, not home at the moment), bypassed the check to use the version installed through pipx

artdeell, 2025-01-27:
Got home, checked, esptool version used by the build process is 4.8.1
Full build log:
```
Create version.h
LN: platform/board to /home/maks/esp32/apps/platform/dummy
Register: hello
Register: nsh
Register: sh
Register: taskset
Register: ostest
Register: smp
CPP:  /home/maks/esp32/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld-> /home/maks/esp32/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/esp_rom/esp32/ld/esp32.rom.libgcc.ldCPP:  /home/maks/esp32/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld-> /home/maks/esp32/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/esp_rom/esp32/ld/esp32.rom.newlCPP:  /home/maks/esp32/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld-> /home/maks/esp32/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/esp_rom/esp32/ld/esp32.rom.newCPP:  /home/maks/esp32/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld-> /home/maks/esp32/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/esp_rom/esp32/ld/esp32.rom.syscallLD: nuttx
CP: nuttx.hex
MKIMAGE: ESP32 binary
esptool.py -c esp32 elf2image --ram-only-header -fs 4MB -fm dio -ff 40m -o nuttx.bin nuttx
esptool.py v4.8.1
Creating esp32 image...
Image has only RAM segments visible. ROM segments are hidden and SHA256 digest is not appended.
Merged 1 ELF section
Successfully created esp32 image.
Generated: nuttx.bin
esptool.py -c esp32 -p /dev/ttyUSB0 -b 921600  write_flash -fs detect -fm dio -ff 40m 0x1000 nuttx.bin
esptool.py v4.8.1
Serial port /dev/ttyUSB0
Connecting......
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: d8:bc:38:fc:f2:1c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00039fff...
Compressed 231140 bytes to 110125...
Wrote 231140 bytes (110125 compressed) at 0x00001000 in 2.2 seconds (effective 837.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

acassis, 2026-02-09:
@fdcavalcanti @eren-terzioglu @tmedicci for visibility

casaroli, 2026-11-08:
Fixed by #18695.

closed 2026-11-08 by casaroli