apache/nuttx #15693: Unable to use LED example on STM32 Nucleo-F446RE board

maitiSoutrik, 2025-01-25:
### Description

I have the following led configuration : 

```
# CONFIG_ARCH_LEDS is not set
# LED Support
CONFIG_USERLED=y
CONFIG_USERLED_LOWER=y
# CONFIG_USERLED_LOWER_READSTATE is not set
# CONFIG_USERLED_EFFECTS is not set
# CONFIG_LEDS_APA102 is not set
# CONFIG_LEDS_MAX7219 is not set
# CONFIG_RGBLED is not set
# CONFIG_DRIVERS_POWERLED is not set
# CONFIG_ARCH_PHY_POLLED is not set
CONFIG_EXAMPLES_LEDS=y
CONFIG_EXAMPLES_LEDS_PROGNAME="leds"
CONFIG_EXAMPLES_LEDS_PRIORITY=100
CONFIG_EXAMPLES_LEDS_STACKSIZE=2048
CONFIG_EXAMPLES_LEDS_DEVPATH="/dev/userleds"
CONFIG_EXAMPLES_LEDS_LEDSET=0x01
# CONFIG_EXAMPLES_LEDS_RUST is not set
# CONFIG_EXAMPLES_LEDS_ZIG is not set
# CONFIG_EXAMPLES_RGBLED is not set
```

When I build and flash the firmware on the board and access the nsh shell, I don't see the /dev/userleds 
furthermore the leds example doesn't work.

I tried looking through some PRs for other controllers 
[11169](https://github.com/apache/nuttx/pull/11169)
[15422](https://github.com/apache/nuttx/pull/15422)

But this is also not helping.
Any leads on using LED on Nucleo-F446RE board

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-01-26:
@yashvi-05 I think you faced similar issue recently with this board, could you please help here?

acassis, 2025-01-26:
@maitiSoutrik you need to do: ls /dev to confirm that /dev/userleds was created.

Probably it is not created during the board initialization

maitiSoutrik, 2025-01-26:
Hi @acassis , you are right. It's not being created.

ldube, 2025-01-27:
Does it get to this line in your stm32_bringup?
ret = userled_lower_initialize("/dev/userleds");

yashvi-05, 2025-01-27:
In that solution, I edited the stm32.bringup file, which does not have the
userled *lower *initialize function.

yashvi-05, 2025-01-27:
Also, type 'ls /dev' in Minicom and tell me the options that are visible.

yashvi-05, 2025-01-27:
In that solution, I edited some configuration in make menuconfig

maitiSoutrik, 2025-01-27:
Hi @yashvi-05 ,

I followed the steps : 

1. `make distclean`

2. `./tools/configure.sh nucleo-f446re:nsh`

3. `make menuconfig`

I selected the following from menuconfig : 

- Device Drivers -> LED Support
 [ * ] LED Driver
 [ * ] Generic Lower Half LED Driver

- Application Config -> Examples
 [ * ] LED driver example

4. `make -j8 && st-flash --connect-under-reset write nuttx.bin 0x08000000`

I am using `picocom` to view the terminal 

```
nsh>
nsh> ls /dev
/dev:
 console
 null
 ttyS0
 zero
nsh>
```
As you can see the `/dev/userleds` is not there.

raiden00pl, 2025-01-27:
most likely you are missing board initialization. Try with `CONFIG_NSH_ARCHINIT=y`.
This is probably the most common issue with NuttX and new users: https://github.com/apache/nuttx/issues/11321

maitiSoutrik, 2025-01-27:
> most likely you are missing board initialization. Try with `CONFIG_NSH_ARCHINIT=y`. This is probably the most common issue with NuttX and new users: [#11321](https://github.com/apache/nuttx/issues/11321)

Hey thanks a lot @raiden00pl ! Worked like a charm 👍🏾

closed 2025-01-27 by maitiSoutrik

acassis, 2025-01-27:
@raiden00pl I think after the board initialization "simplification" we could keep the board bring-up initialization called by default to avoid issues like this

ldube, 2025-01-27:
> [@raiden00pl](https://github.com/raiden00pl) I think after the board initialization "simplification" we could keep the board bring-up initialization called by default to avoid issues like this

Who is working on this "simplification"? Can you provide a link?

maitiSoutrik, 2025-01-27:
> > [@raiden00pl](https://github.com/raiden00pl) I think after the board initialization "simplification" we could keep the board bring-up initialization called by default to avoid issues like this
> 
> Who is working on this "simplification"? Can you provide a link?
> 

Can I also contribute?

raiden00pl, 2025-01-27:
@ldube Probably no one is working on it yet. We are still in the discussion stage: https://github.com/apache/nuttx/issues/11321 

This is a big change and it would be nice to have more pros and cons from the community, but it seems like there aren't many people interested in this issue.

acassis, 2025-01-27:
> > > [@raiden00pl](https://github.com/raiden00pl) I think after the board initialization "simplification" we could keep the board bring-up initialization called by default to avoid issues like this
> > 
> > 
> > Who is working on this "simplification"? Can you provide a link?
> 
> Can I also contribute?

Sure, all ideas, suggestions and, of course, code is welcome!