apache/nuttx #15712: [BUG] Make: sim:quickjs fails to build

simbit18, 2025-01-29:
### Description / Steps to reproduce the issue

sim:quickjs Cmake is ok while Make fails to build with this Error:

LN: platform/board to /github/workspace/sources/apps/platform/dummy
Downloading quickjs-2020-11-08.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  6  734k    6 49152    0     0  37885      0  0:00:19  0:00:01  0:00:18 37867
[100](https://github.com/simbit18/nuttx_test_pr/actions/runs/13029942215/job/36346890425#step:7:101)  734k  100  734k    0     0   412k      0  0:00:01  0:00:01 --:--:--  412k
Unpacking quickjs-2020-11-08.tar.xz to quickjs
patching file quickjs-libc.c
Hunk #1 succeeded at 3558 (offset -137 lines).
Hunk #2 succeeded at 3573 with fuzz 2 (offset -137 lines).
Hunk #3 FAILED at 3743.
1 out of 3 hunks FAILED -- saving rejects to file quickjs-libc.c.rej
patching file CMakeLists.txt
patching file qjsc.cmake
make[3]: *** [Makefile:74: quickjs] Error 1
make[2]: *** [Makefile:57: /github/workspace/sources/apps/interpreters/quickjs_context] Error 2
make[1]: *** [Makefile:187: context] Error 2
make: *** [tools/Unix.mk:458: /github/workspace/sources/apps/.context] Error 2

Testing with nuttx docker and jobs Cmake and Make

https://github.com/simbit18/nuttx_test_pr/actions/runs/13029942215/job/36346890425#logs

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu at GitHub Actions

### NuttX Version

master

### Issue Architecture

[Arch: simulator]

### Issue Area

[Area: Build System]

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-01-29:
@simbit18 did you try to find which commit broke it? It was working some time ago!

simbit18, 2025-01-29:
Hi @acassis  currently on Linux is built with Cmake.

```
====================================================================================
Cmake in present: sim/quickjs
Configuration/Tool: sim/quickjs
2025-01-28 03:52:56
------------------------------------------------------------------------------------
  Cleaning...
  Configuring...
  Select HOST_LINUX=y
  Select HOST_X86_64=y
  Building NuttX...
====================================================================================
```

Has been moved to cmake with this PR https://github.com/apache/nuttx/pull/14693

@lupyuen can you build with make ?

Even macOS in the NuttX mirror fails because it is built with make
```
====================================================================================
Configuration/Tool: sim/quickjs
2025-01-29 08:05:54
------------------------------------------------------------------------------------
  Cleaning...
  Configuring...
  Building NuttX...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
 30  734k   30  224k    0     0   186k      0  0:00:03  0:00:01  0:00:02  186k
100  734k  100  734k    0     0   540k      0  0:00:01  0:00:01 --:--:--  540k
make[3]: *** [quickjs] Error 1
make[3]: Target `context' not remade because of errors.
make[2]: *** [/Users/runner/work/nuttx/nuttx/sources/apps/interpreters/quickjs_context] Error 2
make[2]: Target `context_all' not remade because of errors.
make[1]: *** [context] Error 2
make: *** [/Users/runner/work/nuttx/nuttx/sources/apps/.context] Error 2
make: Target `all' not remade because of errors.
/Users/runner/work/nuttx/nuttx/sources/nuttx/tools/testbuild.sh: line 385: /Users/runner/work/nuttx/nuttx/sources/nuttx/../nuttx/nuttx.manifest: No such file or directory
  [1/1] Normalize sim/quickjs
====================================================================================
```

https://github.com/NuttX/nuttx/actions/runs/13026374853/job/36336155739#step:7:1491

acassis, 2025-01-29:
Understood! I think moving some configs to be tested only with a building system "maker" will end up with issues like this

lupyuen, 2025-01-30:
@simbit18 You're right, `sim:quickjs` won't build correctly with `make` on Ubuntu with Docker:
https://gist.github.com/lupyuen/e18229d879a99a5a0b582a35f09004e6
```text
sudo docker run -it \
  ghcr.io/apache/nuttx/apache-nuttx-ci-linux:latest \
  /bin/bash
cd
git clone https://github.com/apache/nuttx
git clone https://github.com/apache/nuttx-apps apps
cd nuttx
tools/configure.sh sim:quickjs
make -j

patching file CMakeLists.txt
patching file qjsc.cmake
make[3]: *** [Makefile:74: quickjs] Error 1
make[2]: *** [Makefile:57: /root/apps/interpreters/quickjs_context] Error 2
make[1]: *** [Makefile:187: context] Error 2
make: *** [tools/Unix.mk:458: /root/apps/.context] Error 2
```

simbit18, 2025-02-13:
@acassis @lupyuen 
fix https://github.com/apache/nuttx-apps/pull/2993

closed 2025-02-21 by acassis