apache/nuttx #15731: [DRUNX] Dahsboard for Distributed Runtime and bUild Test Farm for NuttX.

cederom, 2025-01-31:
### Is your feature request related to a problem? Please describe.

Dashboard will be a common place to gather, store, and review distributed build and runtime logs from DRUNX (Distributed Runtime and bUild Test Farm for NuttX).

### Describe the solution you'd like

* Dashboard will be a common place to gather, store, and review distributed build and runtime logs.
* Verify if Dashboard can be attached to PR automation as external runner.

### Verification

- [x] I have verified before submitting the report.

lupyuen, 2025-01-31:
I have an article about this. Stay tuned next weekend :-)

[image]

cederom, 2025-01-31:
Thank you @lupyuen :-) We are playing with GitHub Projects [1] to see if it can be used to create a roadmap etc, you are Admin too so play around and lets see how it works :-)

[1] https://github.com/apache/nuttx/projects?query=nuttx

lupyuen, 2025-02-01:
> * Verify if Dashboard can be attached to GH PR automation as external runner so we could have have a build and runtime tests results before merging a PR.

Reminder to myself: Be careful when running Unmerged Code on our Home Computers. In case the PR contains any Scripts or Apps that may cause problems on our Home Computer or Home Network.

cederom, 2025-02-01:
Whaaat did something bad happen??

Basically we only operate on NuttX repos, so there should be no malware... but anyone can send PR with anything true!!!

How to prevent problems? Network separation? One time use containers / jails / vm with restricted network access?

I will mark this important point in architecture requiremens! Thanks @lupyuen !

cederom, 2025-02-01:
I found a REST API to Pull Requests and Issues reference, added to top description. Its only to see if this is possible @lupyuen not really need to be active right now and you can test on a local private repo.

lupyuen, 2025-02-01:
> I found a REST API to Pull Requests and Issues reference

Yep thanks I'm already using the GitHub API for our PR Review Bot :-)

I might try a scaled-down simpler implementation that has less security risk. For example, when I post a PR Comment `@nuttxpr please test`, then our Test Bot will download the PR and run Build + Test on QEMU RISC-V 🤔

lupyuen, 2025-02-01:
> Whaaat did something bad happen??

5 Years Ago: We had Security Issues with a PineTime Smartwatch that we opened up for Remote Testing :-)

https://github.com/lupyuen/remote-pinetime-bot?tab=readme-ov-file#security-issues