apache/nuttx #15737: Releases

cederom, 2025-01-31:


cederom, 2025-02-02:
What should we do with potential changes ready for 13.0.0 and/or stuff that we want in 13 but would break master and short term upcoming 12.x updates? A dedicated branch? Merge to master when ready then going 13? :-) @jerpelea

jerpelea, 2025-02-06:
branch out is scheduled for 1 Match followed by a 4 week stabilization period

cederom, 2025-02-06:
> @jerpelea: branch out is scheduled for 1 Match followed by a 4 week stabilization period

Sorry Alin, I have updated the description, its just a basket of ideas for possibly breaking changes that could land on yet not planned 13 release, thanks for the remark :-)

linguini1, 2025-09-02:
I have added the empty documentation/old README.txt files issues to this tracker. I think that these *must* be solved before the new release. It's not really acceptable to have empty documentation for users. Some effort should be spent requesting contributors who wrote applications to write some docs, or experts on certain sub-systems to fill in the blank pages for those applications.

linguini1, 2026-02-23:
> One question [@linguini1](https://github.com/linguini1) do you know how to create a MILESTONE in these dashboards / github projects? :-P I thought we could use them to keep all in one dashboard / github project.. but if creating separate projects works better then lets stick to it :D

I actually am not sure, I haven't used milestones before. Let me take a look online

linguini1, 2026-02-23:
Maybe something else we could consider for 13.0.0 is removing the STM32 legacy pinmaps that have deprecation warnings? Thought of this because I'm encountering it in a lot of my builds right now:
```
   44 | #  pragma message "CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py"
```