apache/nuttx #15755: [BUG] arch/xtensa: wrong definition for `_int32_t` and `_uint32_t`

tmedicci, 2025-02-04:
### Description / Steps to reproduce the issue

## Summary

One of the commits introduced by https://github.com/apache/nuttx/pull/2222 (https://github.com/apache/nuttx/commit/c18f0748303a236cc2b55969e20082c03452b0c5) changes the definition of `_int32_t` and `_uint32_t` with the following commit message:

> xtensa: Change _int32_t from long to int to match the compiler

This change, however, doesn't seem to be true, generating problems to compile `newlib`, for instance, due to incompatible compiler and NuttX definitions.

## How to Reproduce?

Just try to build an xtensa device with `newlib`. We can use ESP32-S3, for instance:

```
$ make -j distclean && ./tools/configure.sh esp32s3-devkit:nsh && kconfig-tweak -d LIBM_TOOLCHAIN && kconfig-tweak -e LIBM_NEWLIB && make olddefconfig -s -j && make -j bootloader && make flash ESPTOOL_PORT=/dev/ttyUSB0
...
CC:  newlib/newlib/newlib/libm/complex/cproj.c In file included from /home/tiago/Documents/work/espressif/projects/nuttx/nuttxspace_alfa/nuttx/include/sys/types.h:34,
                 from /home/tiago/Documents/work/espressif/projects/nuttx/nuttxspace_alfa/nuttx/include/stddef.h:30,
                 from /home/tiago/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/xtensa-esp32s3-elf/sys-include/sys/cdefs.h:47,
                 from newlib/newlib/newlib/libm/complex/cproj.c:71:
/home/tiago/Documents/work/espressif/projects/nuttx/nuttxspace_alfa/nuttx/include/stdint.h:178:29: error: conflicting types for 'int32_t'; have '_int32_t' {aka 'int'}
  178 | typedef _int32_t            int32_t;
      |                             ^~~~~~~
<command-line>: note: previous declaration of 'int32_t' with type 'int32_t' {aka 'long int'}
/home/tiago/Documents/work/espressif/projects/nuttx/nuttxspace_alfa/nuttx/include/stdint.h:179:29: error: conflicting types for 'uint32_t'; have '_uint32_t' {aka 'unsigned int'}
  179 | typedef _uint32_t           uint32_t;
      |                             ^~~~~~~~
<command-line>: note: previous declaration of 'uint32_t' with type 'uint32_t' {aka 'long unsigned int'}
make[1]: *** [Makefile:53: bin//cproj.o] Error 1
make: *** [tools/LibTargets.mk:206: libs/libm/libm.a] Error 2
```

## The Error

Check the error for `uint32_t`:
```
/home/tiago/Documents/work/espressif/projects/nuttx/nuttxspace_alfa/nuttx/include/stdint.h:179:29: error: conflicting types for 'uint32_t'; have '_uint32_t' {aka 'unsigned int'}
  179 | typedef _uint32_t           uint32_t;
      |                             ^~~~~~~~
<command-line>: note: previous declaration of 'uint32_t' with type 'uint32_t' {aka 'long unsigned int'}
```

We are defining `uint32_t` at [`include/stdint.h`](https://github.com/apache/incubator-nuttx/blob/cd2fcf52520314bf375b4ffbf2ee527f53a0d23a/include/stdint.h#L179) as `_uint32_t`. `_uint32_t` is defined as `unsigned int` at [`arch/xtensa/include/types.h`](https://github.com/apache/incubator-nuttx/blob/8d50457dfc6f79f58487fb33c42407f115e94de6/arch/xtensa/include/types.h#L60) (after the changes introduced by https://github.com/apache/nuttx/pull/2222), but `uint32_t` was previously defined as `long unsigned int` by the compiler:
```
<command-line>: note: previous declaration of 'uint32_t' with type 'uint32_t' {aka 'long unsigned int'}
```

In fact, this compiler's definition seems to be true by checking it directly:
```
xtensa-esp32-elf-gcc -dM -E - < /dev/null | grep __UINT32_TYPE__
#define __UINT32_TYPE__ long unsigned int
```

## Proposed Solution

Reverting https://github.com/apache/nuttx/commit/c18f0748303a236cc2b55969e20082c03452b0c5 enables `newlib` to be built. On the other hand, a lot of warnings due to the expected format:

```
misc/ramdisk.c:241:9: warning: format '%u' expects argument of type 'unsigned int', but argument 3 has type 'blkcnt_t' {aka 'long unsigned int'} [-Wformat=]
  241 |   finfo("sector: %" PRIuOFF " nsectors: %u sectorsize: %d\n",
      |         ^~~~~~~~~~~
  242 |         start_sector, nsectors, dev->rd_sectsize);
      |         ~~~~~~~~~~~~
      |         |
      |         blkcnt_t {aka long unsigned int}
```

### Suppositions

[... 426 lines cut ...]
HAL version:
  sync/release_v5.1.c-nuttx-20230814-1576-g87ccbf88b6

CHIP ID:
  ESP32-S3 has no Chip ID. Reading MAC instead.

Flash ID:
  Manufacturer: 20
  Device: 4017

Security information:
  Flags: 0x00000000 (0b0)
  Key Purposes: (0 0 0 0 0 0 12)
  Chip ID: 9
  API Version: 0
  Secure Boot: Disabled
  Flash Encryption: Disabled
  SPI Boot Crypt Count (SPI_BOOT_CRYPT_CNT): 0x0

Flash status:
  0x0200

### Verification

- [x] I have verified before submitting the report.

yamt, 2025-02-05:
> @yamt , do you recall how did you test the compiler's definition (as stated by https://github.com/apache/nuttx/commit/c18f0748303a236cc2b55969e20082c03452b0c5)?

i don't remember what exactly i used. i guess it was `xtensa-esp32-elf-gcc -dM -E - < /dev/null | grep INT32_TYPE` or something like that.

it seems that the esp-provided compiler has changed its configuration at some point.
i guess you or your colleagues can have a better idea about it than i have.

```
spacetanuki% .espressif.old/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

spacetanuki% .espressif.old/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -dM -E - < /dev/null|grep INT32_TYPE
#define __INT32_TYPE__ int
#define __UINT32_TYPE__ unsigned int
spacetanuki% 
```

```
(venv) spacetanuki% xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG esp-12.2.0_20230208) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

(venv) spacetanuki% xtensa-esp32-elf-gcc -dM -E - < /dev/null|grep INT32_TYPE
#define __INT32_TYPE__ long int
#define __UINT32_TYPE__ long unsigned int
(venv) spacetanuki% 
```

> What should we do? Try to fix them all?

adapt to the configuration used by the latest esp toolchain?
ie. make it long.

yamt, 2025-02-05:
> > What should we do? Try to fix them all?
> 
> adapt to the configuration used by the latest esp toolchain? ie. make it long.

or, restore the toolchain configuration if the change was not intentional.

tmedicci, 2025-02-05:
> > [@yamt](https://github.com/yamt) , do you recall how did you test the compiler's definition (as stated by [c18f074](https://github.com/apache/nuttx/commit/c18f0748303a236cc2b55969e20082c03452b0c5))?
> 
> i don't remember what exactly i used. i guess it was `xtensa-esp32-elf-gcc -dM -E - < /dev/null | grep INT32_TYPE` or something like that.
> 
> it seems that the esp-provided compiler has changed its configuration at some point. i guess you or your colleagues can have a better idea about it than i have.
> 
> ```
> spacetanuki% .espressif.old/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc --version
> xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
> Copyright (C) 2018 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> 
> spacetanuki% .espressif.old/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -dM -E - < /dev/null|grep INT32_TYPE
> #define __INT32_TYPE__ int
> #define __UINT32_TYPE__ unsigned int
> spacetanuki% 
> ```
> 
> ```
> (venv) spacetanuki% xtensa-esp32-elf-gcc --version
> xtensa-esp32-elf-gcc (crosstool-NG esp-12.2.0_20230208) 12.2.0
> Copyright (C) 2022 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> 
> (venv) spacetanuki% xtensa-esp32-elf-gcc -dM -E - < /dev/null|grep INT32_TYPE
> #define __INT32_TYPE__ long int
> #define __UINT32_TYPE__ long unsigned int
> (venv) spacetanuki% 
> ```
> 
> > What should we do? Try to fix them all?
> 
> adapt to the configuration used by the latest esp toolchain? ie. make it long.

Thanks for the information!

> or, restore the toolchain configuration if the change was not intentional.

I don't think it's an option. Check https://docs.espressif.com/projects/esp-idf/en/stable/esp32/migration-guides/release-5.x/5.0/gcc.html#espressif-toolchain-changes for the associated information:

> This change now matches upstream GCC which long integers for int32_t and uint32_t on Xtensa, RISC-V, and other architectures.

So, I think we should change it on NuttX and provide a compiler's version check.

yamt, 2025-02-06:
> So, I think we should change it on NuttX and provide a compiler's version check.

yes, it makes sense.

PR #15918 (2025-03-01): drivers/audio/es8388: set proper format specifier macro for logging
## Summary
Use `PRIu32` instead of `%u` for `uint32_t` to avoid build warnings on different architectures.

This is required as part of an effort to fix #15755.

## Impact

- Impact on user: No.
- Impact on build: Yes, removes build warnings due to wrong data type depending on architecture.
- Impact on hardware: No.
- Impact on documentation: No.
- Impact on security: No.
- Impact on compatibility: No.

## Testing

### Building

- ./tools/configure.sh esp32s3-korvo-2:audio
- make

### Results

Before this change:

```
In file  [... cut]
closed 2025-07-23 by fdcavalcanti