apache/nuttx #15760: [BUG] RP2040 SMP configurations broken

linguini1, 2025-02-05:
### Description / Steps to reproduce the issue

I am unable to boot into NSH using the SMP configuration on RP2040 devices. Testing was primarily done on a Raspberry Pi Pico W.

Steps to reproduce:

1. Checkout to almost latest master branch (abcbb1eebd).
2. `./tools/configure.sh -l raspberrypi-pico-w:smp`
3. `make -j`
4. Upload resulting UF2 to Pico W.
5. No shell appears on UART0.

I have reverted back to 9d1e7a0510 and that version of NuttX has a working SMP configuration. I plan to continue bisecting.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Linux 6.12.10-arch1-1 #1 SMP PREEMPT_DYNAMIC x86_64 GNU/Linux

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Kernel]

### Verification

- [x] I have verified before submitting the report.

xiaoxiang781216, 2025-02-05:
@hujun260 look at this problem.

linguini1, 2025-02-05:
The bad commit can be found here: b69111d16a2a330fa272af8175c832e08881844b
Which is from PR #14578

xiaoxiang781216, 2025-02-05:
do you try with this patch: https://github.com/apache/nuttx/pull/15728? @linguini1

linguini1, 2025-02-05:
> do you try with this patch: https://github.com/apache/nuttx/pull/15728? @linguini1 

No I haven't, the master branch I was testing with is 8 days old and this is from 5 days ago. Thank you for pointing this out, I will try with this patch included tomorrow morning (EST timezone)

xiaoxiang781216, 2025-02-05:
Thanks, let's wait your verification.

hujun260, 2025-02-05:
I tested with the latest code and found no issues.

linguini1, 2025-02-05:
Latest master branch resolves this issue.

shtirlic, 2025-04-12:
Still broken for rp23xx (pico2), reverting to at least to 7774271ad613308816eb71c8f371b0a7ba866935, fixed the SMP build

shtirlic, 2025-05-19:
> > Still broken for rp23xx (pico2) on master, reverting to at least to [b49f428](https://github.com/apache/nuttx/commit/b49f4286fb60a2b1ceaefc0ce2b712d1992bf2e6) fixes the SMP build For me the first bad commit is [914ae53](https://github.com/apache/nuttx/commit/914ae532e68bf4ca75c758d6f541a1d2fcdce8c3) [@linguini1](https://github.com/linguini1)
> 
> Is this still broken?

yep