apache/nuttx #15779: [Feedback] PR Test Bot

lupyuen, 2025-02-06:
### Is your feature request related to a problem? Please describe.

Please provide your feedback for the PR Test Bot. Thanks!

TODO: More details coming

### Describe the solution you'd like

TODO: More details coming

### Describe alternatives you've considered

TODO: More details coming

### Verification

- [x] I have verified before submitting the report.

tmedicci, 2025-02-24:
@lupyuen , I have a question:

If I understood correctly, anyone could trigger (by tagging @nuttxpr) the testbot but I'd only run if you "like" the message that tagged it? Is that correct?

lupyuen, 2025-02-25:
@tmedicci Right now I have to review manually every Test Bot request, for security issues. Please tag me `@lupyuen` after sending a `@nuttxpr` request :-)

lupyuen, 2025-05-13:
> how can I add my milkv_duos to the CI? It should be nice if we could run this test automatically without calling the nuttxpr bot manually

Hmmm are we thinking of triggering the Hardware Test directly from CI? There's a problem: The PR might contain malicious scripts or code, that could mess up our Test PC and Network. We should review the PR Code before doing the Hardware Test.

We could do this instead (and it already works today)
1. Anyone may post a PR comment: `@nuttxpr test milkv_duos:nsh` or `@nuttxpr test avaota-a1:nsh`
2. Right now, this will trigger an Email Alert to me
3. I will review the PR for malicious code
4. Then I will start the [PR Test Bot](https://github.com/lupyuen/nuttx-test-bot) manually
5. Please gimme some time to review the PR, in case I'm sick (or out for marathon training)

Or if we prefer to run the Hardware Test Manually, without calling any GitHub API:

```bash
git clone https://github.com/lupyuen/nuttx-build-farm
cd nuttx-build-farm

## For Milk-V Duo S and Oz64 SG2000:
## Build and Test github.com/Fix-Point/nuttx, branch timer-improve
./build-test.sh oz64 \
  /tmp/build-test.log \
  HEAD HEAD \
  https://github.com/Fix-Point/nuttx timer-improve \
  https://github.com/apache/nuttx-apps master

## For Avaota-A1: Change oz64 to avaota
## Needs a MicroSD Multiplexer
```

This requires:
1. A TFTP Server to be installed for hosting the NuttX Image
2. On Milk-V Duo S: U-Boot should be configured to [boot NuttX over TFTP](https://lupyuen.org/articles/sg2000.html#boot-nuttx-over-tftp)

closed 2025-05-13 by lupyuen