apache/nuttx #15788: [BUG] <nuttx:nrf52840_dk:sdc_nimble issue. >

zhihui2015, 2025-02-08:
### Description / Steps to reproduce the issue

  1. The project version used is NuttX-12.6.0, and the compiler version is arm-none-eabi-gcc (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7)) 13.2.1 20231009  
  2. Execute ./tools/configure.sh -l nrf52840-dk:sdc_nimble  
  3. Execute make, sdk-nrfxlib will be downloaded in the process, and the nuttx.bin will be generated  
  4. nuttx.bin the burning is performed, it is found that it cannot be burned, and after analysis, it is found that the size of the nuttx.bin is 536871168, which is the abnormal size  
  5. Open the nuttx.bin with jflash, and find that most of the back is 0, which should be an exception to the construction, and the back should be useless. After truncate -s 512K nuttx.bin truncate and then burn again, i can burn normally.  
  6.  Restart after burning, and print as follows  
  nx_mount: ERROR: target /proc exists and is a special node  
  ERROR: Failed to mount the PROC filesystem: -20  
  ERROR: nrf52_sdc_initialize() failed: -1  

  NuttShell (NSH) NuttX-12.6.0     
  

  Why is that?  Do I need to download a separate softdevice controller, such as the S140? Or does nuttx.bin already include a softdevice controller? I'm not sure.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 20.04 LTS

### NuttX Version

12.6.0

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Applications]

### Verification

- [x] I have verified before submitting the report.

raiden00pl, 2025-02-08:
I haven't seen this before because I always use `nuttx` (elf file) or `nuttx.hex`.

`nuttx.bin` it is huuuuge which doesn't look good:

```
-rwxr-xr-x  1 raiden00 raiden00 513M Feb  8 08:08 nuttx.bin
```
> Why is that? Do I need to download a separate softdevice controller, such as the S140? Or does nuttx.bin already include a softdevice controller? I'm not sure.

all is done automatically. `nuttx.bin` seems broken, try with `nuttx` or `nuttx.hex`.

`nrf52840-dk:nsh` configuration is normal size:

```
-rwxr-xr-x  1 raiden00 raiden00  80K Feb  8 08:18 nuttx.bin*
```

so the soft device support breaks something in `nuttx.bin` generation. The same issue is with cmake build

zhihui2015, 2025-02-08:
@raiden00pl Thanks for your reply!  

With NuttX, I can successfully flash using J-Link. After flashing, the log output is as follows:  

```
ABCDE▒nx_mount: found procfs  
bt_initialize: btdev 0x200007e4  
SoftDevice Controller initialized  
NRF52 Bringup complete  
btnet_ifup: Bringing up: 00:00:00:00:00:00  
NSH Library Initialized  
NuttShell (NSH) NuttX-12.6.0  
nsh> hci_tx_kthread: started  
hci_tx_kthread: HCI TX thread started  
```

Has NimBLE started running normally? However, I cannot find any BLE advertisements using **nRF Connect**.  

I tried entering `nimble` in NSH, and the system executed `nimble_main`. I also added print logs in `nimble_main`, and they were successfully printed, which indicates that the system has loaded `nimble_main` correctly. However, I still cannot find the BLE advertisement in **nRF Connect**, and I am not sure why.  

Could you share your test steps with me?

raiden00pl, 2025-02-08:
I confirm that it doesn't work with release 12.6, but it works with 12.7 and 12.8. Master branch is also broken

raiden00pl, 2025-02-08:
master is broken by this commit 620f4bc32f7641c85ab86dffe6e0b9f52d20581a

hujun260, 2025-02-09:
OK

hujun260, 2025-02-10:
Please help me verify it.
https://github.com/apache/nuttx/pull/15797

zhihui2015, 2025-02-10:
Thanks for the reply! @raiden00pl  I switched to 12.6-RC1, or 12.8, and the tests are all fine. I can now detect BLE advertisements and establish connections using nrf-connect. Both 12.6 and master versions have issues.

@hujun260 I cherry-picked this PR-15797, and it seems mostly fine. BLE advertisements work during testing, but I see the following log during startup:

```
nsh> nimble
GAP procedure initiated: stop advertising.
Failed to restore IRKs from store; status=8
advertise
```

Not sure if there are any other issues?

hujun260, 2025-02-10:
> Thanks for the reply! [@raiden00pl](https://github.com/raiden00pl) I switched to 12.6-RC1, or 12.8, and the tests are all fine. I can now detect BLE advertisements and establish connections using [...]

If you revert https://github.com/apache/nuttx/issues/15788#issuecomment-2646822492, will there still be problems?

raiden00pl, 2025-02-10:
> Failed to restore IRKs from store; status=8

This message is expected.