apache/nuttx #15807: [BUG] <Configuring for Raspberry Pi Pico W failing due to missing 'olddefconfig'>

Justin-Morrison-github, 2025-02-10:
### Description / Steps to reproduce the issue

Hi, I am a Carleton InSpace Rocket Team member and have issues configuring NuttX for my Raspberry Pi Pico W. Carleton's master branch should be up to date with NuttX master.

I am able to run make distclean with no issue, but when I try to configure usbnsh for NuttX but get the error logs below. I am a new user of NuttX so any advice would be appreciated.

$ ./tools/configure.sh -l raspberrypi-pico-w:usbnsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
CP: arch/dummy/Kconfig to /home/morri/InSpace/InSpace-Nuttx/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/morri/InSpace/InSpace-Nuttx/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /home/morri/InSpace/InSpace-Nuttx/apps/platform/dummy
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/morri/InSpace/InSpace-Nuttx/nuttx/boards/arm/rp2040/raspberrypi-pico-w/include
LN: drivers/platform to /home/morri/InSpace/InSpace-Nuttx/nuttx/drivers/dummy
LN: include/arch/chip to /home/morri/InSpace/InSpace-Nuttx/nuttx/arch/arm/include/rp2040
LN: arch/arm/src/chip to /home/morri/InSpace/InSpace-Nuttx/nuttx/arch/arm/src/rp2040
LN: arch/arm/src/board to /home/morri/InSpace/InSpace-Nuttx/nuttx/boards/arm/rp2040/raspberrypi-pico-w/../common
LN: arch/arm/src/board/board to /home/morri/InSpace/InSpace-Nuttx/nuttx/boards/arm/rp2040/raspberrypi-pico-w/src
/usr/bin/bash: line 1: olddefconfig: command not found
make: *** [tools/Unix.mk:726: olddefconfig] Error 1
ERROR: failed to refresh

and when I run make menuconfig I get these error logs:

$ make menuconfig
CP: arch/dummy/Kconfig to /home/morri/InSpace/InSpace-Nuttx/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/morri/InSpace/InSpace-Nuttx/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /home/morri/InSpace/InSpace-Nuttx/apps/platform/dummy
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/morri/InSpace/InSpace-Nuttx/nuttx/boards/arm/rp2040/raspberrypi-pico-w/include
LN: drivers/platform to /home/morri/InSpace/InSpace-Nuttx/nuttx/drivers/dummy
LN: include/arch/chip to /home/morri/InSpace/InSpace-Nuttx/nuttx/arch/arm/include/rp2040
LN: arch/arm/src/chip to /home/morri/InSpace/InSpace-Nuttx/nuttx/arch/arm/src/rp2040
LN: arch/arm/src/board to /home/morri/InSpace/InSpace-Nuttx/nuttx/boards/arm/rp2040/raspberrypi-pico-w/../common
LN: arch/arm/src/board/board to /home/morri/InSpace/InSpace-Nuttx/nuttx/boards/arm/rp2040/raspberrypi-pico-w/src
/home/morri/.local/bin/menuconfig: /home/morri/InSpace/InSpace-Nuttx/apps/examples/Kconfig:67: '/home/morri/InSpace/InSpace-Nuttx/apps/examples/hello_rust/Kconfig' not found (in 'source "/home/morri/InSpace/InSpace-Nuttx/apps/examples/hello_rust/Kconfig"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
make: *** [tools/Unix.mk:731: menuconfig] Error 1

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

5.15.167.4-microsoft-standard-WSL2 #1 SMP Tue Nov 5 00:21:55 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

### NuttX Version

master (abcbb1eebd2c7186e2f0e5683eec490b905b49c0)

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Build System]

### Verification

- [x] I have verified before submitting the report.

simbit18, 2025-02-11:
hi @Justin-Morrison-github it seems related to wrong path

```
apps/examples/hello_rust/Kconfig' not found (in 'source "/home/morri/InSpace/InSpace-Nuttx/apps/examples/hello_rust/Kconfig"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
make: *** [tools/Unix.mk:731: menuconfig] Error 1
```
After this PR https://github.com/apache/nuttx-apps/pull/2944  the path has changed

Have you tried updating the [apps](https://github.com/apache/nuttx-apps) repository ?

fdcavalcanti, 2025-02-12:
Run `git clean -xfdn` on your apps folder after you `make distclean` on `nuttx`.
See if there are pending files there.

Justin-Morrison-github, 2025-03-01:
> hi [@Justin-Morrison-github](https://github.com/Justin-Morrison-github) it seems related to wrong path
> 
> ```
> apps/examples/hello_rust/Kconfig' not found (in 'source "/home/morri/InSpace/InSpace-Nuttx/apps/examples/hello_rust/Kconfig"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
> make: *** [tools/Unix.mk:731: menuconfig] Error 1
> ```
> 
> After this PR [apache/nuttx-apps#2944](https://github.com/apache/nuttx-apps/pull/2944) the path has changed
> 
> Have you tried updating the NuttX [apps](https://github.com/apache/nuttx-apps) repository ?

> Run `git clean -xfdn` on your apps folder after you `make distclean` on `nuttx`. See if there are pending files there.

> Run `git clean -xfdn` on your apps folder after you `make distclean` on `nuttx`. See if there are pending files there.

Sorry I have been away for a while. I was up to date with the apps repo at the time. That was one of the first suggestions from my team lead.

I have run into another issue where  `make distclean` on 'nuttx' fails with the following:
'''
$ make distclean
make[3]: *** No rule to make target 'clean'.  Stop.
make[2]: *** [/home/morri/InSpace/InSpace-Nuttx/apps/Directory.mk:43: getprime/_clean] Error 2
make[1]: *** [Makefile:60: /home/morri/InSpace/InSpace-Nuttx/apps/testing/_clean] Error 2
make: *** [tools/Unix.mk:794: /home/morri/InSpace/InSpace-Nuttx/apps_clean] Error 2
'''

Justin-Morrison-github, 2025-03-01:
Even though `make distclean` on `nuttx` fails, I still ran `git clean -xfdn` on `apps` and a lot of files came up. I'm sure how to interpret the following output:
```
$ git clean -xfdn
Would remove .context
Would remove .depend
Would remove .dirlinks
Would remove CustomApps/
Would remove Kconfig
Would remove audioutils/.kconfig
Would remove audioutils/Kconfig
Would remove benchmarks/.kconfig
Would remove benchmarks/Kconfig
Would remove boot/.kconfig
Would remove boot/Kconfig
Would remove builtin/.depend
Would remove builtin/Make.dep
Would remove builtin/registry/.updated
Would remove builtin/registry/ostest.bdat
Would remove builtin/registry/ostest.pdat
Would remove builtin/registry/sh.bdat
Would remove builtin/registry/sh.pdat
Would remove canutils/.kconfig
Would remove canutils/Kconfig
Would remove crypto/.kconfig
Would remove crypto/Kconfig
Would remove database/.kconfig
Would remove database/Kconfig
Would remove examples/.kconfig
Would remove examples/Kconfig
Would remove examples/hello/.depend
Would remove examples/hello/Make.dep
Would remove examples/mcuboot/.kconfig
Would remove examples/mcuboot/Kconfig
Would remove examples/module/.kconfig
Would remove examples/module/Kconfig
Would remove examples/sotest/.kconfig
Would remove examples/sotest/Kconfig
Would remove fsutils/.kconfig
Would remove fsutils/Kconfig
Would remove games/.kconfig
Would remove games/Kconfig
Would remove graphics/.kconfig
Would remove graphics/Kconfig
Would remove industry/.kconfig
Would remove industry/Kconfig
Would remove inertial/.kconfig
Would remove inertial/Kconfig
Would remove inspace-nxtelem
Would remove interpreters/.kconfig
Would remove interpreters/Kconfig
Would remove interpreters/luamodules/.kconfig
Would remove interpreters/luamodules/Kconfig
Would remove logging/.kconfig
Would remove logging/Kconfig
Would remove lte/.kconfig
Would remove lte/Kconfig
Would remove math/.kconfig
Would remove math/Kconfig
Would remove mlearning/.kconfig
Would remove mlearning/Kconfig
Would remove netutils/.kconfig
Would remove netutils/Kconfig
Would remove nshlib/.depend
Would remove nshlib/Make.dep
Would remove platform/.depend
Would remove platform/Make.dep
Would remove platform/board
Would remove sdr/.kconfig
Would remove sdr/Kconfig
Would remove staging/
[... 62 lines cut ...]
Would remove testing/ostest/timedmutex.c.home.morri.InSpace.InSpace-Nuttx.apps.testing.ostest.o
Would remove testing/ostest/timedmutex.c.home.morri.InSpace.InSpace-Nuttx.apps.testing.ostest_1.o
Would remove testing/ostest/timedwait.c.home.morri.InSpace.InSpace-Nuttx.apps.testing.ostest.o
Would remove testing/ostest/timedwait.c.home.morri.InSpace.InSpace-Nuttx.apps.testing.ostest_1.o
Would remove testing/ostest/vfork.c.home.morri.InSpace.InSpace-Nuttx.apps.testing.ostest.o
Would remove testing/ostest/vfork.c.home.morri.InSpace.InSpace-Nuttx.apps.testing.ostest_1.o
Would remove testing/ostest/waitpid.c.home.morri.InSpace.InSpace-Nuttx.apps.testing.ostest.o
Would remove testing/ostest/waitpid.c.home.morri.InSpace.InSpace-Nuttx.apps.testing.ostest_1.o
Would remove testing/ostest/wdog.c.home.morri.InSpace.InSpace-Nuttx.apps.testing.ostest.o
Would remove testing/ostest/wdog.c.home.morri.InSpace.InSpace-Nuttx.apps.testing.ostest_1.o
Would remove testing/sched/getprime/.built
Would remove testing/sched/getprime/.depend
Would remove testing/sched/getprime/Make.dep
Would remove testing/sched/getprime/getprime_main.c.home.morri.InSpace.InSpace-Nuttx.apps.testing.sched.getprime.o
Would remove testing/sched/getprime/getprime_main.c.home.morri.InSpace.InSpace-Nuttx.apps.testing.sched.getprime_1.o
Would remove videoutils/.kconfig
Would remove videoutils/Kconfig
Would remove wireless/.kconfig
Would remove wireless/Kconfig
Would remove wireless/bluetooth/.kconfig
Would remove wireless/bluetooth/Kconfig
Would remove wireless/ieee802154/.kconfig
Would remove wireless/ieee802154/Kconfig
```
Do you have any insight into what this means?

fdcavalcanti, 2025-03-05:
Usually this happens when you change critical stuff without `distcleaning` first. Changing KConfig files in this condition could cause this kind of behavior. If you run `git clean -xfd` without `n` option it will delete all untracked files. Should work normally after that.

Justin-Morrison-github, 2025-03-08:
> Usually this happens when you change critical stuff without `distcleaning` first. Changing KConfig files in this condition could cause this kind of behavior. If you run `git clean -xfd` without `n` option it will delete all untracked files. Should work normally after that.

Hi @fdcavalcanti, I ran the `git clean -xfd` on the `apps` directory. 

I then tried to configure fora Raspberry Pi Pico W with usbnsh and it failed with mention of an `olddefconfig` command

$ ./tools/configure.sh -l raspberrypi-pico-w:usbnsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
CP: arch/dummy/Kconfig to /home/morri/InSpace/InSpace-Nuttx/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/morri/InSpace/InSpace-Nuttx/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /home/morri/InSpace/InSpace-Nuttx/apps/platform/dummy
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/morri/InSpace/InSpace-Nuttx/nuttx/boards/arm/rp2040/raspberrypi-pico-w/include
LN: drivers/platform to /home/morri/InSpace/InSpace-Nuttx/nuttx/drivers/dummy
LN: include/arch/chip to /home/morri/InSpace/InSpace-Nuttx/nuttx/arch/arm/include/rp2040
LN: arch/arm/src/chip to /home/morri/InSpace/InSpace-Nuttx/nuttx/arch/arm/src/rp2040
LN: arch/arm/src/board to /home/morri/InSpace/InSpace-Nuttx/nuttx/boards/arm/rp2040/raspberrypi-pico-w/../common
LN: arch/arm/src/board/board to /home/morri/InSpace/InSpace-Nuttx/nuttx/boards/arm/rp2040/raspberrypi-pico-w/src
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/audioutils
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/benchmarks
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/boot
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/canutils
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/crypto
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/database
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/examples/mcuboot
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/examples/module
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/examples/rust
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/examples/sotest
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/examples
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/fsutils
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/games
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/graphics
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/industry
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/inertial
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/interpreters/luamodules
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/interpreters
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/logging
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/lte
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/math
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/mlearning
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/netutils
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/sdr
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/system
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/testing/arch
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/testing/cxx
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/testing/drivers
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/testing/fs
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/testing/libc
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/testing/mm
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/testing/sched
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/testing
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/videoutils
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/wireless/bluetooth
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/wireless/ieee802154
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps/wireless
mkkconfig in /home/morri/InSpace/InSpace-Nuttx/apps
/usr/bin/bash: line 1: olddefconfig: command not found
make: *** [tools/Unix.mk:726: olddefconfig] Error 1
ERROR: failed to refresh

Let me know if you've ever seen something like this before.

fdcavalcanti, 2025-03-10:
I've never had this: ` /usr/bin/bash: line 1: olddefconfig: command not found`.

Are your apps and nuttx on master branch? Do the `git clean` trick on both and checkout to master.

If that fails, can you try a fresh clone of `nuttx` and `nuttx-apps`? It really seems like something is out-of-sync. If we figure this out, it would be possible to improve the build system to avoid this problem in the future.