apache/nuttx #15856: [FEATURE] Common interface for LORA chips

raiden00pl, 2025-02-18:
### Is your feature request related to a problem? Please describe.

Currently NuttX supports only `sx127x` but more drivers for LORA are coming. All these drivers should use the same common API, so that it will be possible to create portable user applications.

This most likely will involve breaking changes for `sx127x` (easy to detect and fix compiler errors).

The problem is partly due to a decision that I made a long time ago when adding `sx127x` support with Greg approval, that the LORA chips in NuttX would be character drivers, not network devices. The reason for this is that edge devices supporting LORA are low-power devices with limited resources. For such devices, supporting the network stack is a waste of resources.
The result of this decision is that we cannot use the portable interface provided by the network stack and we have to design a portable interface as a character device.

We want to avoid the fate of legacy sensors where there was no common interface, making it impossible to create portable applications with them.

Here are two PRs that add another LORA chips:

- https://github.com/apache/nuttx/pull/15828
- https://github.com/apache/nuttx/pull/15795

### Describe the solution you'd like

Common API for LORA chips.

### Verification

- [x] I have verified before submitting the report.

keever50, 2025-02-18:
my idea was to use the current WLIOC, but extend it with the modulation. ex: WLIOC_LORA, WLIOC_GFSK, WLIOC_OOK, WLIOC_QPSK etc. LoRa devices often can use multiple modulation techniques like that.
It would be way easier to spot mistakes of misplaced wlioc commands. (gfsk fdiv on a lora modulation mode. Not possible).

This allows you to have
WLIOC_LORA_BW
WLIOC_LORA_SF
these are common across all lora devices so far i know.

Then there are some more specific commands such as CRC, fixed header etc. I guess these can be left in familly specifics. Obviously with default values.

The result would be an API that all apps can work with. Including ones that allow adaptive data rate, like LoRaWAN.

raiden00pl, 2025-02-18:
@keever50 it's a good idea to introduce 3 levels of ioctl: 1. common RF commands, 2. modulation specific commands, 3. chips specific commands. However, the last type of commands should be avoided if possible, but I understand that it may be necessary in some cases.

The ultimate goal should be an API enabling support for LoRaWAN. I haven't been following the LoRaWAN topic for a while, but the last time I saw 'loramac-node' implementation from Semtech it was awful, so I gave up on porting it to NuttX.

linguini1, 2025-02-18:
I suppose this common interface would also still involve having `read` and `write` as rx/tx?

linguini1, 2025-02-19:
> yes, `read` and `write` should be used. But in addition to the data itself, metadata such as RSSI or SNR should be returned if available from  the driver. 

Maybe as an ioctl?

keever50, 2025-02-19:
I used the SX127x way in my SX126x already, but i guess we also need to standardize this too.

https://github.com/apache/nuttx/blob/ff4d461fdaed36db9e6acfe3aabb7b418b0608c2/include/nuttx/wireless/lpwan/sx127x.h#L113-L124

The signal information is quite nice to have in the packet.
I also provided a crc_error variable in my own variant.
Its ofcourse 1 when there is a CRC check error. Otherwise always 0.
Useful for when the user wants to repair the packet.

Maybe nice to have in the common interface.

I am not sure what the reserved bytes are for?

keever50, 2025-02-19:
I'm not sure what is the best way to figure out together what the best common interface is going to be and how to make one. What about the mailing list?
But maybe this is a good starting point.
How about the following

**— IOCTL —**
Common:
- WLIOC_COM_FREQ // Hz
- WLIOC_COM_PWR // Power ( in dBm or mW? )
- WLIOC_COM_PREAMBLES // Bytes
- WLIOC_COM_MOD // Modulation technique ( Maybe? )

LoRa:
- WLIOC_LORA_SF // Spreading factor
- WLIOC_LORA_BW // Bandwidth
- WLIOC_LORA_CR // Code rate
- WLIOC_LORA_CRC // CRC enable (enables CRC bytes)
- WLIOC_LORA_FIXED_HDR // Fixed header length (enables length byte)
- WLIOC_LORA_SYNCWORD // Syncword bytes (Include length?)

Specific implementations
- WLIOC_model_function

**— Read header —**
- Payload bytes
- Payload length
- RSSI // What format?
- SNR // What format?
- Error // Comes from CRC or checksums

Now, because IOCTL is a limited resource,
is it a good idea to compact some parameters into a struct?
It is done already in some implementations (including in mine).
I configure the entire lora modulation using one struct.

raiden00pl, 2025-02-23:
@keever50 it looks good.

> is it a good idea to compact some parameters into a struct?

yes, it's ok. It makes sense to limit the ioctl calls needed to configure the radio.
I haven't used LORA radio for a long time, so it's hard for me to say what approach would be best from the user's point of view. One call to configure many radio parameters will definitely save some FLASH memory and CPU cycles, which is good for low-power applications like LORA nodes.

keever50, 2025-02-23:
True!
In case of struct:

**— IOCTL using struct —**
```
Common:
WLIOC_COM_PARAMS
takes
{
 uint32_t freq_hz;
 int8_t power_dbm;
 uint16_t preamble_bytes;
}

WLIOC_COM_MOD
takes
Modulation technique (enum like lora, gfsk, ook, etc)

```
**— LoRa structs —**
```
WLIOC_LORA_MODULATION
takes
{
 uint8_t spreading_factor;
 uint16_t bw_khz;
}

WLIOC_LORA_PACKET
takes
{
 enum coding_rate;
 bool include_crc;
 bool include_length; // Also called implicit header or fixed length header. It actually just includes a length byte.
}

WLIOC_LORA_SET_SYNCWORD
takes
{
 uint8_t *sync_bytes;
 size_t sync_length;
}

```
**— Read Header —**
```
{
 uint8_t *payload;
 size_t payload_length;
 int32_t RSSI_fixed_float_dBm; // 2 points precision 0.00dBm
 int32_t SNR_fixed_float_ratio; // 2 points precision 0.00dB
 bool error;
}
```

I hope these data types make sense.

Technically it would also be possible to add headers for writing, which can set frequency, power, preambles and length.
That would allow for quicker response.
What do you think about that, @raiden00pl ?

keever50, 2025-02-27:
The initial sx126x is merged, now i think the strategy is ill create the common API and leave the old SX127x definitions in. Atleast for one PR.
The following PRs can be porting the SX127x to common(Removing the old defs) and further integrating the SX126x and the RN ones.

I think these steps are the most logical without combining too many commits into one?

keever50, 2025-02-27:
Also i think the structs idea for common settings might not be a really good idea after all, since it requires more work porting. Also the driver must unpack the entire struct after each change. Might result into awkward implementations. For small structs like lora settings its fine.

keever50, 2025-03-01:
https://github.com/keever50/nuttx/blob/9caff9604715a4eae1235c0fef03547ebfc730cc/include/nuttx/wireless/ioctl.h

I made progress to the start of the common API. Let me know what you think.

linguini1, 2025-03-05:
> https://github.com/keever50/nuttx/blob/9caff9604715a4eae1235c0fef03547ebfc730cc/include/nuttx/wireless/ioctl.h
> 
> I made progress to the start of the common API. Let me know what you think.

I like it so far, but I'm still not sure about using `fixedmath`. Why not just take up the extra integer space for RSSI and dBm to have more precision?

Just so I understand correctly, the process for the user will be to do something like the following?

```c
uint8_t payload[SIZE];
struct wlioc_rx_hdr_s hdr = {
    .payload_length = sizeof(payload),
    .payload = &hdr,
};
read(radio, &hdr, sizeof(hdr));
```

keever50, 2025-03-05:
> > https://github.com/keever50/nuttx/blob/9caff9604715a4eae1235c0fef03547ebfc730cc/include/nuttx/wireless/ioctl.h
> > I made progress to the start of the common API. Let me know what you think.
> 
> I like it so far, but I'm still not sure about using `fixedmath`. Why not just take up the extra integer space for RSSI and dBm to have more precision?
> 
> Just so I understand correctly, the process for the user will be to do something like the following?
> 
> uint8_t payload[SIZE];
> struct wlioc_rx_hdr_s hdr = {
>     .payload_length = sizeof(payload),
>     .payload = &hdr,
> };
> read(radio, &hdr, sizeof(hdr));

Yeah honestly you are right, it is possible to use 32bit - 16 fractional bits using fixedmath.h.
The 16bit - 8 fractional bits did fit nicely in common RSSI and SNR values.
But saving up on these very few bytes is of course insignificant, so lets just make it 32 (16 fractional bits)?

About the rx_hdr, thats interesting! i was more thinking of .payload pointing to an internal driver buffer.
But this is also an option, the buffer being the user's responsibility. Might actually allow more user flexibility.

keever50, 2025-03-05:
It is important to note the actual precision. It is however a bit comical.
They should be 
precision steps = 2^-b
where b are the fractional bits.

32bit (16 bits for fraction) is 0.0000152587890625 steps. Where the remaining 16bit gives a range of 65536.
In case for some reason you wonder what happens if you do 64bit. 0.000000000232830643654 steps, 4294967296 range.

raiden00pl, 2025-03-06:
> About the rx_hdr, thats interesting! i was more thinking of .payload pointing to an internal driver buffer.
But this is also an option, the buffer being the user's responsibility. Might actually allow more user flexibility.

the buffer **must be owned by user-space**, you can't return a pointer owned by kernel because it won't work in build mode other than `CONFIG_BUILD_FLAT`

keever50, 2025-03-06:
> the buffer **must be owned by user-space**, you can't return a pointer owned by kernel because it won't work in build mode other than `CONFIG_BUILD_FLAT`

ohhhh right! That makes completely sense. In that case, yes @linguini1 you are right.

keever50, 2025-03-06:
@linguini1 
Merged the new things. I'm looking which added IOCTLs of you can stay or move.

- WLIOC_SETBITRATE, is this for GFSK? Then lets move that under GFSK tag
- WLIOC_CRCEN -> WLIOC_LORA_ENCRC
- WLIOC_SETCODERATE -> WLIOC_LORA_SETCR (with an enum)
- WLIOC_IQEN can stay. Good idea for common.
- WLIOC_SETSYNC -> WLIOC_LORA_SYNCWORD
- WLIOC_SETMOD can stay for common
- WLIOC_SETSPREAD -> WLIOC_LORA_SF
- WLIOC_SETBANDWIDTH -> WLIOC_LORA_BW
- WLIOC_GETSNR -> in the rx_header, since its common packet specific, right?

WLIOC_SETTXPOWERF is the same as WLIOC_SETTXPOWER? I think so far this is the only breaking change.
I'm not entirely sure if we really need that fine precision afterall. Nearly all chips only do steps of 1dB.
If we do really need it, we should probably make this fixedmath.h b16_t.

Do we really need to GET the values too? Everything is usually reset after closing the driver anyway.

What do you think of that?

_Also Your driver has a sneaky bug in it WLIOC_GETTXPOWERF is 0x0020. You accidently incremented from 0x19 to 0x20. Should be 0x1a lol._

linguini1, 2025-03-06:
> @linguini1 
> Merged the new things. I'm looking which added IOCTLs of you can stay or move.
> 
> - WLIOC_SETBITRATE, is this for GFSK? Then lets move that under GFSK tag

Yeah, it's for FSK modulation.

> - WLIOC_CRCEN -> WLIOC_LORA_ENCRC
> - WLIOC_SETCODERATE -> WLIOC_LORA_SETCR (with an enum)
> - WLIOC_IQEN can stay. Good idea for common.
> - WLIOC_SETSYNC -> WLIOC_LORA_SYNCWORD
> - WLIOC_SETMOD can stay for common
> - WLIOC_SETSPREAD -> WLIOC_LORA_SF
> - WLIOC_SETBANDWIDTH -> WLIOC_LORA_BW

Looks good to me!

> - WLIOC_GETSNR -> in the rx_header, since its common packet specific, right?

Yes, but I'm still thinking about that approach. I personally almost feel it's better to keep `read()`  implemented such that it just stores the recieved packet in the user buffer. Anything else strays too far from the POSIX idea of read imo. For instance, a really great testing use case for me is to be able to swap the fd being passed to read. Instead of the radio, I can mock this using packet data sent over USB or UART or something like that. If I have to structure the read buffer, this removes that easy feature. I think users should separately query for the SNR. I find this a flaw with other NuttX drivers, like the ADC standard driver. If the read buffer has to be structured, I think a different interface is required.

> WLIOC_SETTXPOWERF is the same as WLIOC_SETTXPOWER? I think so far this is the only breaking change.

The former is for float arguments to keep backward compatibility with raiden's driver, but we can toss that in favour of the latter which uses integers of 0.01dBm steps.

> I'm not entirely sure if we really need that fine precision afterall. Nearly all chips only do steps of 1dB.

I recommend at least a step of 0.1dBm. The RN2XX3 at least has that much precision. Idk about other chips.

> If we do really need it, we should probably make this fixedmath.h b16_t.

I'm really against fixedmath because I find it unintuitive and it requires the user to use nonstandard representations for numbers. C standard numbers are my preference by far.

> Do we really need to GET the values too? Everything is usually reset after closing the driver anyway.

I think some of them yes. For instance, setting TX power may round to the next highest level the chip supports. It would be good for the user to be able to check. I guess we can store the real value back in the user pointer though.

It may also be useful to check some things like modulation, etc. What if someone's application involves slowly increasing TX power over time (maybe distance is increasing). Having the GET commands let's them check TX power and increment it instead of maintaining their own state. Same thing could be said for other GETs. I'm not against just leaving some out for now until there's a use case for it, but whatever we standardize is going to be hard to add to after the fact for all drivers.

> What do you think of that?
> 
> _Also Your driver has a sneaky bug in it WLIOC_GETTXPOWERF is 0x0020. You accidently incremented from 0x19 to 0x20. Should be 0x1a lol._
> 

Oops!

raiden00pl, 2025-03-07:
RSSI and SNR are properties of RX packet, not a property of the radio, so they MUST BE returned as part of read data. Adding a separate interface to read RSSI/SNR not only introduces unnecessary system calls, but can also lead to reading parameters of the wrong RX package.

keever50, 2025-03-07:
@linguini1 
Thanks

I really love the idea of simplifying read(), so good, however @raiden00pl got a really strong point. reading rx_hdr struct might make handling easier with less errors.

About the discussion of the floating, integer or fixedpoint.h. I agree. For some reason i thought we were avoiding floats all together.
I would love to call it 1/100dBm, centi-dBm or 10x milli-dBm. First one is probably the best.

The GET/SET. Good points, but the radios i worked with do not have hardware instructions to GET back values. So requiring the driver to keep an internal state. However, I completely understand "setting TX power may round to the next highest level the chip supports. ", which is a nice use for GET to check.
It is indeed nicer to "snap" to nearest than returning errors.
However that leaves room for hidden errors if you dont check GET and only error.
But yeah maybe leave it in for consistency.

Ill make these changes

keever50, 2025-03-08:
I implemented the changes.

**Changes:**

- Changed SETRADIOFREQ from MHz to Hz 32bit. Because steps of MHz is not precise enough. For example in EU regulations, 869.250MHz would be one of the frequencies that allow 500mW transmission. 868 or 869 have different rules. Allowing Hz instead kHz allows fine tuning. In cases of drift or temperature.
- Added modulation schemes. G/FSK and ASK/OOK.
- Added SET/GET for everything.
- Added void *private to rx_hdr for driver+application specific stuff.
- Changed b8_t fixedmath.h to uint32_t 1/100 steps in rx_hdr.
**Questions**
Why do we have WLIOC_SETADDR under common while it clearly says its driver specific? This sounds like specific packet engine stuff.

linguini1, 2025-03-08:
> I implemented the changes. https://github.com/keever50/nuttx/blob/common_wireless_character_driver_API/include/nuttx/wireless/ioctl.h
> 
> **Changes:**
> 
> - Changed SETRADIOFREQ from MHz to Hz 32bit. Because steps of MHz is not precise enough. For example in EU regulations, 869.250MHz would be one of the frequencies that allow 500mW transmission. 868 or 869 have different rules. Allowing Hz instead kHz allows fine tuning. In cases of drift or temperature.
> - Added modulation schemes. G/FSK and ASK/OOK.
> - Added SET/GET for everything.
> - Added void *private to rx_hdr for driver+application specific stuff.
> - Changed b8_t fixedmath.h to uint32_t 1/100 steps in rx_hdr.
> - Changes lora bandwidth from kHz to Hz, because its possible to have a kHz and a half.
> 
> **Questions**
> Why do we have WLIOC_SETADDR under common while it clearly says its driver specific? This sounds like specific packet engine stuff.
> 
> What do we return in rx_hdr when SNR or RSSI is not supported? UINT32_MAX?
> 

These sound good! I can't comment on your second last question, but for the last one I think UINT32_MAX seems reasonable.

Why is there a void *private in the rx header? It's a buffer coming from the user, the driver shouldn't need to store anything between calls there.

keever50, 2025-03-08:
> Why is there a void *private in the rx header? It's a buffer coming from the user, the driver shouldn't need to store anything between calls there.

Ah yeah like the previous discussion about user not being able to use pointers pointing to the kernel.
This seems pointless now that i think about it.
The idea was this could be an extension for driver specifics, but this cant be used.
Will be removed.

I think i will remove WLIOC_SETADDR. It is not used anywhere, at all.
The NRF chip only aliases it but for nothing. Funny.

**Correction on dB**
Also oops, i did it again. The dB and RSSI must be INT not UINT. That makes UINT32_MAX obviously INT32_MAX.

After these changes, its probably a good idea to share this with the mailing list. More people are likely to respond there.

keever50, 2025-03-15:
@linguini1 Hey again

I wonder if we really need the 1/100 step dB power setting. What if we do not change it, keep 1/1 steps? Would that affect projects?
Practically 1/100 or 1/10 makes no sense and is really for special precision equipment only. Or measurements.
If this is actually required, we can later add an additional "precision set" command for dB power.
This would remove the breaking aspect of this update. Then its purely an additive update.

Also i plan on PR'ing this update next week. I've posted a follow up on the mailing list that's on going.

raiden00pl, 2025-03-15:
Can't we just choose the closest possible value in dB supported by a given chip? For example, if we choose TX power = 18db but chip supports only 17.6db and 18.1db then driver will select 18.1db. This way we can stay with the db unit, but the question is whether the accuracy of the power setting has any value here.

keever50, 2025-03-15:
> Can't we just choose the closest possible value in dB supported by a given chip? For example, if we choose TX power = 18db but chip supports only 17.6db and 18.1db then driver will select 18.1db. [...]

I think this is the solution at least for now.

The accuracy has practically no significance if it is not lab controlled equipment.
In the real world, the field, radio signals are always unstable. Due many factors, the dB can swing multiple dBs.
If for some reason you exactly precisely want like 5.23dB, these radios are probably not the right choice.
But i'm talking about 99% of projects.

For high power radio equipment it might be a different story, but then we are looking at amateur HAM radio, which is a different category of radios. We talk about hundreds of watts.

linguini1, 2025-03-15:
No, because the RN2XX3 has transmit power settings where it's not possible to round to the nearest dBm. There are a few levels where two fractional dBm transmit powers round to the same whole number.

If you want to avoid making breaking changes I would support a precision dBm command. However, I think that it has issues of its own. As a user I would never choose less precision. You could keep it just for legacy reasons I guess and have the precision version be the new de facto choice to implement.

raiden00pl, 2025-03-15:
> No, because the RN2XX3 has transmit power settings where it's not possible to round to the nearest dBm. There are a few levels where two fractional dBm transmit powers round to the same whole number.

You can always arbitrarily chose the level to which you round. This way you'll lose some supported power levels by the chip, but the question is if ALL supported levels are useful in practice.

keever50, 2025-03-15:
Keeping the non precision one 1/1 dB might not be a bad idea. It is less confusing, because the user does not need to remember the 1/100 step rule.
If the user only cares about normal precision, it creates a cleaner code.
and yes it is possible to add the extra precision one. However i dont see the use? Can be done at a later stage if there is demand.

I understand your issue where two fractionals round to the same number. But is it possible to select either one anyway? Flooring or c eiling all numbers, not round.
In case that even that cant work, you might have to pick ones yourself.

linguini1, 2025-03-15:
> If the user only cares about normal precision, it creates a cleaner code.
> and yes it is possible to add the extra precision one. However i dont see the use? Can be done at a later stage if there is demand.

I have demand for it. I understand the argument that fractional dBm may be unimportant because of path loss fluctuations, but there is an alternative reason why it would matter: current draw. My application is a low power one and I would like full control over the current draw from transmitting.

> I understand your issue where two fractionals round to the same number. But is it possible to select either one anyway? Flooring or c eiling all numbers, not round.

Flooring 2.1 and 2.4 will both yield 2. Ceil will yield 3. Round will yield 2.

> In case that even that cant work, you might have to pick ones yourself.

Maybe, but I think all drivers should behave consistently. Is it always picking the closest level? Always flooring? Always ceiling? I think that the transmit power that is set should be greater or equal to the number the user requests to avoid surprises.

I understand wanting to make things appear cleaner or avoiding breaking legacy commands but I think this would be an unnecessary limitation. I have this use case now and I'm sure others will too. Adding it later will just require more modifications.

keever50, 2025-03-15:
@linguini1 just out of curiosity. What kinds of currents are we talking about between 2.1 and 2.4 dBm?

linguini1, 2025-03-15:
> @linguini1 just out of curiosity. What kinds of currents are we talking about between 2.1 and 2.4 dBm?
> 

~2mA, which sounds insignificant but for battery powered applications that run for long periods of time, that makes a large difference in required battery capacity.

My point is, if we're designing a finalized framework for these chips, the framework shouldn't be excluding features that are so easy to include in the name of whole numbers being cleaner. If the chip provides that granularity, the user should be able to access it.

keever50, 2025-03-16:
> ~2mA, which sounds insignificant but for battery powered applications that run for long periods of time, that makes a large difference in required battery capacity.
>
> My point is, if we're designing [...]

Fair, if you would be transmitting continuously that can have an effect. 

I've decided on adding the extra ioctl command. Something along the lines WLIOC_SET_POWER_FINE which takes 1/100.

If people prefer to use this over the other, power to them. However, it might be possible for drivers to return an error if this precision is not implemented.

This makes implementing this API easier without having to break existing applications. Which removes the breaking aspect and no need to immediately port everything.

linguini1, 2025-03-16:
> Fair, if you would be transmitting continuously that can have an effect.
>
> I've decided on adding the extra ioctl command. Something along the lines WLIOC_SET_POWER_FINE which takes 1/100.
>
> If [...]

This solution is totally fine with me!

closed 2025-03-23 by xiaoxiang781216