apache/nuttx #15880: [HELP] <title> Minimum configuration for CDC ECM to work?

tao-at-pricer, 2025-02-20:
### Description

Dear community, I've been having trouble to get USB CDC ECM to work, tried on both Raspberry Pi Zero W and STM32F4Discovery. There doesn't seem to be examples of how to configure this profile. I tried to get it show up in Debian as a standalone device, and as a composite together with a CDC ACM, the ACM worked fine, but couldn't see ECM popping up. My latest configure for STM32F4Discovery looks like this (composite and NOT working):

```Makefile
CONFIG_ARCH="arm"
CONFIG_ARCH_BOARD="stm32f4discovery"
CONFIG_ARCH_BOARD_STM32F4_DISCOVERY=y
CONFIG_ARCH_BUTTONS=y
CONFIG_ARCH_CHIP="stm32"
CONFIG_ARCH_CHIP_STM32=y
CONFIG_ARCH_CHIP_STM32F407VG=y
CONFIG_ARCH_LEDS_CPU_ACTIVITY=y
CONFIG_ARCH_STACKDUMP=y
CONFIG_BOARDCTL_USBDEVCTRL=y
CONFIG_BOARD_LOOPSPERMSEC=16717
CONFIG_BUILTIN=y
CONFIG_CDCACM=y
CONFIG_CDCACM_COMPOSITE=y
CONFIG_CDCACM_CONSOLE=y
CONFIG_CDCACM_RXBUFSIZE=256
CONFIG_CDCACM_TXBUFSIZE=256
CONFIG_CDCECM_COMPOSITE=y
CONFIG_COMPOSITE_IAD=y
CONFIG_DEBUG_FULLOPT=y
CONFIG_DEBUG_SYMBOLS=y
CONFIG_EXAMPLES_HELLO=y
CONFIG_FS_PROCFS=y
CONFIG_HAVE_CXX=y
CONFIG_HAVE_CXXINITIALIZE=y
CONFIG_IDLETHREAD_STACKSIZE=2048
CONFIG_INIT_ENTRYPOINT="nsh_main"
CONFIG_INTELHEX_BINARY=y
CONFIG_IOB_NBUFFERS=8
CONFIG_IOB_NCHAINS=0
CONFIG_IOB_THROTTLE=0
CONFIG_LINE_MAX=64
CONFIG_MM_REGIONS=2
CONFIG_NETINIT_NOMAC=y
CONFIG_NETINIT_THREAD=y
CONFIG_NETUTILS_PING=y
CONFIG_NET_CDCECM=y
CONFIG_NET_LOOPBACK=y
CONFIG_NET_LOOPBACK_PKTSIZE=64
CONFIG_NET_MBIM=y
CONFIG_NET_TCP=y
CONFIG_NET_TCP_WRITE_BUFFERS=y
CONFIG_NSH_ARCHINIT=y
CONFIG_NSH_BUILTIN_APPS=y
CONFIG_NSH_FILEIOSIZE=512
CONFIG_NSH_READLINE=y
CONFIG_PREALLOC_TIMERS=4
CONFIG_RAMLOG=y
CONFIG_RAMLOG_BUFSIZE=4096
CONFIG_RAMLOG_SYSLOG=y
CONFIG_RAM_SIZE=114688
CONFIG_RAM_START=0x20000000
CONFIG_RAW_BINARY=y
CONFIG_RR_INTERVAL=200
CONFIG_SCHED_HPWORK=y
CONFIG_SCHED_LPWORK=y
CONFIG_SCHED_WAITPID=y
CONFIG_STACK_COLORATION=y
CONFIG_START_DAY=27
CONFIG_START_YEAR=2013
CONFIG_STM32_JTAG_SW_ENABLE=y
CONFIG_STM32_OTGFS=y
CONFIG_STM32_PWR=y
CONFIG_STM32_SPI1=y
CONFIG_STM32_USART2=y
CONFIG_SYSTEM_NSH=y
CONFIG_USBDEV=y
CONFIG_USBDEV_COMPOSITE=y
CONFIG_WQUEUE_NOTIFIER=y
```

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-02-20:
@tao-at-pricer seems like there are examples only for lpc17xx and sama5 boards. You need to look at these boards to understand how to configure the options in menuconfig and how to create the composite USB to use with ECM.

@adamfeuer since you configured it to your board, could you please give some hints here?

tao-at-pricer, 2025-02-20:
> [@tao-at-pricer](https://github.com/tao-at-pricer) seems like there are examples only for lpc17xx and sama5 boards. You need to look at these boards to understand how to configure the options in menuconfig and how to create the composite USB to use with ECM.

Thank you @acassis , yes I indeed found these two configurations and tried to copy the settings, but I was not able to reproduce. Since I don't have dev boards with these two MCUs I can't verify if the stock configures still work with them. I assume it worked at the time it was created, but with development evolving, I want to get a reassurance if someone can verify that. I'll post my settings based on these two moment later.

adamfeuer, 2025-02-22:
@tao-at-pricer @acassis Hey there. I am still on NuttX 11.x for my SAMA5 app, TCP/IP over CDC/ECM works there. I will install 12.x and see if anything has changed.

I've never used CDC/ECM on a non-SAMA5 device, but I do have an STMF4-Discovery, so will try that out and see if I can get it to work, and report back.

Some questions. How are you looking to see if the CDCECM device is being recognized on the Linux side? The NuttX system will appear as a new ethernet interface. You should be able to see it using `ifconfig`.

In the meantime. I know this config works on the SAMA5 on 11.x:

https://github.com/apache/nuttx/blob/master/boards/arm/sama5/jupiter-nano/configs/sdmmc-nsh-net-resolvconf/defconfig

And you may have looked here, but this is the Kconfig file (used by the menuconfig system that configures NuttX) for CDCECM:

https://github.com/apache/nuttx/blob/fd6d80462303fc363d885150e49f9c5f175639c5/drivers/usbdev/Kconfig#L1251

tao-at-pricer, 2025-02-22:
Thanks @adamfeuer , please let me know what you'll find out on the STM32F4 discovery board, I have exactly this and I read in multiple places that it's the recommended board for newbies to start with.

I did try the Mongoose example and it worked out of the box, link (https://mongoose.ws/documentation/tutorials/rp2040/pico-rndis-dashboard/), the title says RNDIS but it also contains example implementing CDC ECM, and that's the profile I used. When plugged in to computer it shows up as a network device and indeed I could find it with ip addr show, and I could use Wireshark and ScaPy to send and monitor ethernet packets. So I was expecting the same behavior when running NuttX examples.

tao-at-pricer, 2025-02-22:
Ok I base my config on `stm32f4discovery:nsh` only enabled CDC-ECM and work queue, compile with error:
```bash
nuttx$ make
Create version.h
LN: platform/board to nuttxspace/apps/platform/dummy
Register: hello
Register: nsh
Register: sh
CC:  chip/stm32_gpio.c chip/stm32_gpio.c:44:11: note: '#pragma message: CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py'
   44 | #  pragma message "CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py"
      |           ^~~~~~~
CC:  chip/stm32_flash.c In file included from chip/stm32_flash.c:40:
chip/stm32f20xxf40xx_flash.c:53:4: warning: #warning "Default Flash Configuration Used - See Override Flash Size Designator" [-Wcpp]
   53 | #  warning "Default Flash Configuration Used - See Override Flash Size Designator"
      |    ^~~~~~~
CPP:  nuttxspace/nuttx/boards/arm/stm32/stm32f4discovery/scripts/ld.script-> /hoLD: nuttx
arm-none-eabi-ld: nuttxspace/nuttx/staging/libarch.a(arm_initialize.o): in function `up_initialize':
arm_initialize.c:(.text.up_initialize+0xe): undefined reference to `arm_usbinitialize'
Memory region         Used Size  Region Size  %age Used
           flash:       93216 B         1 MB      8.89%
            sram:        7988 B       112 KB      6.96%
make[1]: *** [Makefile:214: nuttx] Error 1
make: *** [tools/Unix.mk:552: nuttx] Error 2
```

The saved defconfig is like this:
```Makefile
CONFIG_ARCH="arm"
CONFIG_ARCH_BOARD="stm32f4discovery"
CONFIG_ARCH_BOARD_STM32F4_DISCOVERY=y
CONFIG_ARCH_BUTTONS=y
CONFIG_ARCH_CHIP="stm32"
CONFIG_ARCH_CHIP_STM32=y
CONFIG_ARCH_CHIP_STM32F407VG=y
CONFIG_ARCH_STACKDUMP=y
CONFIG_BOARD_LATE_INITIALIZE=y
CONFIG_BOARD_LOOPSPERMSEC=16717
CONFIG_BUILTIN=y
CONFIG_EXAMPLES_HELLO=y
CONFIG_FS_PROCFS=y
CONFIG_HAVE_CXX=y
CONFIG_HAVE_CXXINITIALIZE=y
CONFIG_INIT_ENTRYPOINT="nsh_main"
CONFIG_INTELHEX_BINARY=y
CONFIG_LINE_MAX=64
CONFIG_MM_REGIONS=2
CONFIG_NETDEV_WORK_THREAD=y
CONFIG_NET_CDCECM=y
CONFIG_NSH_BUILTIN_APPS=y
CONFIG_NSH_FILEIOSIZE=512
CONFIG_NSH_READLINE=y
CONFIG_PREALLOC_TIMERS=4
CONFIG_RAM_SIZE=114688
CONFIG_RAM_START=0x20000000
CONFIG_RAW_BINARY=y
CONFIG_RR_INTERVAL=200
CONFIG_SCHED_LPWORK=y
CONFIG_SCHED_WAITPID=y
CONFIG_START_DAY=6
CONFIG_START_MONTH=12
CONFIG_START_YEAR=2011
CONFIG_STM32_JTAG_SW_ENABLE=y
CONFIG_STM32_PWR=y
CONFIG_STM32_SPI1=y
CONFIG_STM32_USART2=y
CONFIG_SYSTEM_NSH=y
CONFIG_USART2_RXBUFSIZE=128
CONFIG_USART2_SERIAL_CONSOLE=y
CONFIG_USART2_TXBUFSIZE=128
CONFIG_USBDEV=y
```

adamfeuer, 2025-02-22:
@tao-at-pricer Does it compile with the unmodified defconfig?

adamfeuer, 2025-02-23:
@tao-at-pricer @acassis I take it back about working fine. I was using `12.0.0-rc0` - which was the main branch from 11.x, I think.

I installed `12.8.0` and compiled for the SAMA5, and can't start NuttX using a pure nsh defconfig (no CDC-ECM configured). So there is something wrong with the build, 12.x may have broken SAMA5.

I'll look at it more tomorrow night.

adamfeuer, 2025-02-23:
Ok, I did a little bit of compiling different versions of NuttX. CDC-ECM networking works for the SAMA5 on the`12.0.0.0` release.` SAMA5 with just the basic nsh defconfig (no CDC-ECM) does not work on `12.1.0` releases and later. So it looks like `12.1.0` broke the SAMA5 boards.

I don't know about the STM32F4-Discovery... I can't spend any more time on this until maybe next weekend. But one thing you can try is check out the `12.0.0.` tag which I know works for the SAMA5. Then:

1. Get NuttX to build and run using just the nsh defconfig. If this works, 
2. Use menuconfig to enable CDC-ECM ethernet and compile. Then report back.

TimJTi, 2025-02-23:
I am using the master branch on my SAMA5 board and it works, but when I rebase every few weeks I do sometimes have to run make menuconfig because of random build or K config changes. Can’t remember anything specific or that was a complete PITA to sort though.

tao-at-pricer, 2025-02-23:
> [@tao-at-pricer](https://github.com/tao-at-pricer) Does it compile with the unmodified defconfig?

Unmodified nsh config works, got error only when I added CDC-ECM and work queue. Compiling for nsh gives a few warnings but the process succeeded never the less:
```bash
nuttx$ make -j4
LN: platform/board to nuttxspace/apps/platform/dummy
Create version.h
Register: hello
Register: nsh
Register: sh
CC:  nsh_builtin.c chip/stm32_gpio.c:44:11: note: '#pragma message: CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py'
   44 | #  pragma message "CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py"
      |           ^~~~~~~
CC:  pthread/pthread_mutexconsistent.c In file included from chip/stm32_flash.c:40:
chip/stm32f20xxf40xx_flash.c:53:4: warning: #warning "Default Flash Configuration Used - See Override Flash Size Designator" [-Wcpp]
   53 | #  warning "Default Flash Configuration Used - See Override Flash Size Designator"
      |    ^~~~~~~
CPP:  nuttxspace/nuttx/boards/arm/stm32/stm32f4discovery/scripts/ld.script-> nuttxspace/nuttx/boards/arm/stm32/stm32f4discovery/scripts/LD: nuttx
Memory region         Used Size  Region Size  %age Used
           flash:       82056 B         1 MB      7.83%
            sram:        5432 B       112 KB      4.74%
CP: nuttx.hex
CP: nuttx.bin
```

tao-at-pricer, 2025-02-23:
> Ok, I did a little bit of compiling different versions of NuttX. CDC-ECM networking works for the SAMA5 on the`12.0.0.0` release. SAMA5 with just the basic nsh defconfig (no CDC-ECM) does not work on `12.1.0` releases and later. So it looks like `12.1.0` broke the SAMA5 boards.
> 
> I don't know about the STM32F4-Discovery... I can't spend any more time on this until maybe next weekend. But one thing you can try is check out the `12.0.0.` tag which I know works for the SAMA5. Then:
> 
> 1. Get NuttX to build and run using just the nsh defconfig. Report the results here. If this works,
> 2. Use menuconfig to enable CDC-ECM ethernet and compile. Then report back.

I switched both nuttx and apps to 12.0, and ran vanilla nsh config, it compiled without error. However, adding CDC-ECM and work queue/low priority got me this error:

```bash
netlib_setifstatus.c: In function 'netlib_ifup':
netlib_setifstatus.c:66:44: error: 'NET_SOCK_TYPE' undeclared (first use in this function)
   66 |       int sockfd = socket(NET_SOCK_FAMILY, NET_SOCK_TYPE, NET_SOCK_PROTOCOL);
      |                                            ^~~~~~~~~~~~~
netlib_setifstatus.c:66:44: note: each undeclared identifier is reported only once for each function it appears in
netlib_setifstatus.c: In function 'netlib_ifdown':
netlib_setifstatus.c:109:44: error: 'NET_SOCK_TYPE' undeclared (first use in this function)
  109 |       int sockfd = socket(NET_SOCK_FAMILY, NET_SOCK_TYPE, NET_SOCK_PROTOCOL);
      |                                            ^~~~~~~~~~~~~
make[3]: *** [nuttxspace/apps/Application.mk:147: netlib_setifstatus.c.home.tao.repos.nuttxspace.apps.netutils.netlib.o] Error 1
make[3]: Leaving directory 'nuttxspace/apps/netutils/netlib'
make[2]: *** [Makefile:52: nuttxspace/apps/netutils/netlib_all] Error 2

In file included from arp/arp_input.c:54:
arp/arp_input.c: In function 'arp_in':
nuttxspace/nuttx/net/arp/arp.h:87:30: error: invalid use of undefined type 'struct iob_s'
   87 |                   &dev->d_iob->io_data[CONFIG_NET_LL_GUARDSIZE])
      |                              ^~
arp/arp_input.c:89:31: note: in expansion of macro 'ARPBUF'
   89 |   FAR struct arp_hdr_s *arp = ARPBUF;
      |                               ^~~~~~
In file included from nuttxspace/nuttx/net/arp/arp.h:48:
nuttxspace/nuttx/net/devif/devif.h:241:28: error: invalid use of undefined type 'struct iob_s'
  241 |                 &dev->d_iob->io_data[CONFIG_NET_LL_GUARDSIZE - \
      |                            ^~
arp/arp_input.c:111:41: note: in expansion of macro 'ETHBUF'
  111 |             FAR struct eth_hdr_s *eth = ETHBUF;
      |                                         ^~~~~~
arp/arp_input.c: In function 'arp_input':
arp/arp_input.c:199:31: error: invalid use of undefined type 'struct iob_s'
  199 |       dev->d_buf = &dev->d_iob->io_data[CONFIG_NET_LL_GUARDSIZE -
      |                               ^~
make[1]: *** [Makefile:77: arp_input.o] Error 1

```

casaroli, 2026-11-08:
Possibly fixed by #19371.