apache/nuttx #15893: [FEATURE] We need a better designed web page

acassis, 2025-02-22:
### Is your feature request related to a problem? Please describe.

Unfortunately we are lagging behind in many aspects, compared to more "professional" project like https://mongoose.ws our web page is missing a lot of information, testimonials, etc

### Describe the solution you'd like

We need to implement a new page with better design and including relevant information about NuttX

### Verification

- [x] I have verified before submitting the report.

cederom, 2025-02-23:
Thanks @acassis I am also pro replacing current website with something also minimalistic in use but simpler to build. Lots of dependencies are required to build out simple website and I always have problems with that on my FreeBSD box. Also lots of components are obsolete because of that (see https://github.com/apache/nuttx-website/issues).

If there is agreement on website update I would search for some simpler mechanism. We are curently using Jekyll, alternatively Apache recommends Docusaurus or Pelican, see https://github.com/apache/apache-website-template.

I keep top level website and documentation issue here with other subissues attached for easier tracking: https://github.com/apache/nuttx-website/issues/129.

What additional content would you like to see on our website?

acassis, 2025-02-23:
Thank @cederom! I agree that we need to use a simpler website, that is also important, but the main point is the design and organization of NuttX site, it is really bad.

I like the idea of using static site generator. I think Jekyll is a good option, maybe a better/fast option could be Hugo (it uses Golang instead of Ruby).

I think before we change the site generator we need to find some professional designer to help us. Maybe Sony and Xiaomi can help here (ping @jerpelea and @xiaoxiang781216 )

ChaiWithJai, 2025-03-04:
@acassis hey! I just started getting involved in the Nuttx ecosystem because I'm a design engineer looking to build a hearing aid and my goal is to accelerate my learning about the "bare metal" layer. I'm loving this project right now.

@maitiSoutrik and I wanted to perform seva (service) to contribute back to the community to help on this issue. We'd love to learn how we can effectively contribute and be good stewards of the community as new members.

Demo of New Site: https://justshipitai-nuttx-docs-demo.vercel.app/

[image]

Please let us know how we can be helpful.

Life is so rich,
Jai and @maitiSoutrik

acassis, 2025-03-04:
Hi @ChaiWithJai and @maitiSoutrik the website that you created is really nice!!! Kudos!!!
Please include the source the the information for the data you included there, otherwise it could be misleading.

For instance @raiden00pl did some recent tests reducing the firmware size, that 8KB Flash and 2KB was what I got many years ago, but at the time I remove many of option like VFS support that currently we cannot remove (until we go back with these options again)

cederom, 2025-03-04:
Thanks @ChaiWithJai :-) Is this website compliant with Apache requirements? https://github.com/apache/apache-website-template

ChaiWithJai, 2025-03-04:
> Hi [@ChaiWithJai](https://github.com/ChaiWithJai) and [@maitiSoutrik](https://github.com/maitiSoutrik) the website that you created is really nice!!! Kudos!!! Please include the source the the [...]

We were just some newbies spiking on this. We would absolutely need to connect and confirm what belongs here for the needs of the core team and community. Speaking of the core team, I wasn't sure whom we should be looping in.

We're happy to contribute, we'd probably be best off contributing as design partners who can advise on how to make it look great, but for following rules and managing CMS, we're too new here to know what to do to drive this to completion.

Any suggestions on next steps?

acassis, 2026-02-25:
Amazing work @Abhishekmishra2808, kudo!!!

NuttX deserved a better website! Now we have a good site!