apache/nuttx #15916: [HELP] Porting NuttX to BeagleY-AI (TI AM67) – Booting R5F via Linux RemoteProc

emomaxd, 2025-02-27:
### Description

I've spent the past week porting NuttX to the BeagleY-AI board. I've gone through all available forums, issue trackers, and documentation, carefully following the official guides, yet I still haven't been able to get it working.

The BeagleY-AI is a Texas Instruments board featuring the TI AM67 SoC, a heterogeneous system-on-chip that includes an ARM Cortex-A53 processor and an ARM Cortex-R5F co-processor.

On the A53 cores, we're running a Debian-based Linux distribution and using remoteproc to boot firmware on the MAIN R5F core. The system expects firmware in ELF format, but when attempting to boot the ELF output of the NuttX kernel, I encounter the following error:

```shell
Invalid Argument Error.
Boot failed: -22
```
I've tried various configuration options, particularly tweaking the defconfig and linker scripts, but so far, I haven't made any progress.

One of my biggest challenges is understanding which configurations need to be modified for a successful boot. I've experimented with various values for kernel load address, RAM start address, and linker load address, yet none of them worked. What should these values be?

Additionally, I’m unsure whether the addresses configured in remoteproc matter or if Linux handles this dynamically. Right now, the A53 core boots Linux via U-Boot, but I don't know where exactly remoteproc is loading the firmware for the R5F core. Any guidance on these memory addresses and configurations would be greatly appreciated.

From what I understand, the error seems to be caused by missing sections in the ELF file, preventing Linux's remoteproc mechanism from properly parsing and booting it. This is just my assumption, but when I used readelf to inspect other working ELF firmware files, I noticed that they contained certain boot-related sections that were missing in NuttX’s ELF output—particularly at the beginning of the file. These could be the vector table, boot section, resource table, or even the device tree.

So, as a last resort, if I can't resolve this issue through configuration changes, I’d like to explore the possibility of passing device tree data to remoteproc and NuttX. If there is a way to do this, a guide on how to implement it would be greatly appreciated.

Starting tomorrow, I plan to add images for further debugging. Additionally, I'm considering creating a PR that links to this issue—would that be preferable, or should I include my code directly in this issue? Looking for guidance on the best approach.

Also, I expect **extremely detailed insights** on the **device tree**, **remoteproc configuration**, and **kernel configuration** since these seem to be critical areas that might be causing the issue and could be could be valuable for future reference.

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-02-28:
@emomaxd nice know you are porting NuttX to a new board. @pkarashchenko since you ported NuttX to the original BeagleBoneBoard, could you please take a look?

The error -22 means -EINVAL (Invalid Argument), seems like it is something coming from U-Boot itself, as you said it is expecting the ELF header at the right position. @lupyuen @raiden00pl @xiaoxiang781216 any idea?

lupyuen, 2025-03-01:
> On the A53 cores, we're running a Debian-based Linux distribution and using remoteproc to boot firmware on the MAIN R5F core. The system expects firmware in ELF format, but when attempting to boot the ELF output of the NuttX kernel, I encounter the following error

Sorry are we using U-Boot Bootloader to boot NuttX? If so then our NuttX Image should have a Linux Kernel Header, instead of an ELF Header: https://lupyuen.github.io/articles/uboot#linux-kernel-header

Wonder if "remoteproc" is meant for booting NuttX over the network? On RISC-V Devices we use TFTP on U-Boot to boot NuttX, maybe this will help: https://lupyuen.github.io/articles/sg2000#boot-nuttx-over-tftp

acassis, 2025-03-01:
U-Boot boots NuttX on BBB, but I'm not sure if the create image is ELF file with a Linux header, I don't think so.

lupyuen, 2025-03-01:
On PinePhone (Arm A53) it's actually the Binary NuttX Image (`nuttx.bin`) that has a Linux Header in front, without any ELF Sections. Will this work on BeagleY-AI?
https://lupyuen.github.io/articles/uboot#nuttx-header
```text
  /* Kernel startup entry point.
   * ---------------------------
   *
   * This must be the very first address in the loaded image.
   * It should be loaded at any 4K-aligned address.
   * __start will be set to 0x4008 0000 in the Linker Script
   */
  .globl __start;
__start:

  /* DO NOT MODIFY. Image header expected by Linux boot-loaders.
   *
   * This add instruction has no meaningful effect except that
   * its opcode forms the magic "MZ" signature of a PE/COFF file
   * that is required for UEFI applications.
   */
  add     x13, x18, #0x16  /* the magic "MZ" signature */
  b       real_start       /* branch to kernel start */

  /* PinePhone Image load offset from start of RAM */
  .quad   0x0000  

  /* Effective size of kernel image, little-endian */
  .quad   _e_initstack - __start

  /* Informative flags, little-endian */
  .quad   __HEAD_FLAGS

  .quad   0          /* reserved */
  .quad   0          /* reserved */
  .quad   0          /* reserved */
  .ascii  "ARM\x64"  /* Magic number, "ARM\x64" */
  .long   0          /* reserved */

/* NuttX OS Code begins here, after the header */
real_start: ... 
```

emomaxd, 2025-03-02:
Linux runs on the A53 core, trying to boot NuttX on the R5F core using remoteproc, but it doesn't accept the "start" command of the remoteproc and throws Invalid Argument error when trying to boot NuttX.

Beagle provides the customized Debian OS and boots that Debian with U-boot to run on A53 core, this has nothing to do with the NuttX and the R5F, that is a completely different procedure.

Wonder is there any supported SoC that starts with remoteproc?

emomaxd, 2025-03-11:
I've successfully resolved the issue by adding a resource table into the ELF file, incorporating a new section in the linker script called ".resource_table," which eliminated the "Invalid Argument" error.

Now, I’m facing a challenge with verifying whether NuttX has booted successfully. To address this, I decided to test the system using GPIO pins. I created a simple GPIO driver that calculates the correct address and bitmask to toggle the appropriate bit on the GPIO bank. Then, I invoked the led_blink_demo function within the arm_boot() function. However, things didn’t go as expected.

The LED in question is the onboard LED connected to MAIN_GPIO0 PIN11. To control it, I modified the R5 core's working domain to the MAIN domain. This was necessary because the MCU domain does not have direct access to MAIN domain peripherals. I verified this limitation by testing with one of the examples from the TI AM67A SDK, which confirmed that the MCU domain cannot access MAIN domain peripherals. I am contemplating whether it’s possible to configure this access and plan to inquire further on the TI forum.

Any thoughts on this?

emomaxd, 2025-06-01:
I've already set up the JTAG and GDB combo. I had completely forgotten this issue was still open — feel free to close it.