apache/nuttx #15932: [HELP] Cannot Build Esp32C6 when efuse and mcuboot are enable during same time

FelipeMdeO, 2025-03-03:
### Description

Hello All.

We have a solution with mcuboot enabled and, recently we add to our solution efuse driver. In this situation I cannot build the project because efuse files only is available to compilation in simple boot way. See it below:

```
ifeq ($(CONFIG_ESPRESSIF_SIMPLE_BOOT),y)
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)nuttx$(DELIM)src$(DELIM)bootloader_banner_wrap.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)bootloader_console.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)bootloader_console_loader.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)${CHIP_SERIES}$(DELIM)bootloader_${CHIP_SERIES}.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)bootloader_common.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)bootloader_common_loader.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)bootloader_flash$(DELIM)src$(DELIM)bootloader_flash.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)bootloader_flash$(DELIM)src$(DELIM)bootloader_flash_config_${CHIP_SERIES}.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)bootloader_clock_init.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)bootloader_clock_loader.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)bootloader_efuse.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)bootloader_mem.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)bootloader_random.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)bootloader_random_${CHIP_SERIES}.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)esp_image_format.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)${CHIP_SERIES}$(DELIM)bootloader_soc.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)${CHIP_SERIES}$(DELIM)bootloader_sha.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)bootloader_support$(DELIM)src$(DELIM)flash_encrypt.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)soc$(DELIM)${CHIP_SERIES}$(DELIM)uart_periph.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)esp_rom$(DELIM)patches$(DELIM)esp_rom_uart.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)esp_rom$(DELIM)patches$(DELIM)esp_rom_sys.c
  CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)esp_rom$(DELIM)patches$(DELIM)esp_rom_spiflash.c
  **CHIP_CSRCS += chip$(DELIM)$(ESP_HAL_3RDPARTY_REPO)$(DELIM)components$(DELIM)efuse$(DELIM)src$(DELIM)esp_efuse_fields.c**

  LDFLAGS += --wrap=bootloader_print_banner
endif
```

@eren-terzioglu Why these files only should be compiled when we are using CONFIG_ESPRESSIF_SIMPLE_BOOT?

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-03-03:
@FelipeMdeO what is the compilation issue?

FelipeMdeO, 2025-03-03:
Guys, I edited the post because I presented a corrected version where I moved file to out of `ifeq`. Please consider the file esp_efuse_api_key.c

@acassis The error is:
```
LD: nuttx
riscv-none-elf-ld: warning: /home/felipe-moura/Downloads/msav2-connectivity/nuttx/nuttx has a LOAD segment with RWX permissions
riscv-none-elf-ld: /home/felipe-moura/Downloads/msav2-connectivity/nuttx/staging/libarch.a(esp_efuse_utility.o): in function `esp_efuse_utility_burn_chip':
/home/felipe-moura/Downloads/msav2-connectivity/nuttx/arch/risc-v/src/chip/esp-hal-3rdparty/components/efuse/esp32c6/esp_efuse_utility.c:108:(.text.esp_efuse_utility_burn_chip+0x6c): undefined reference to `esp_efuse_get_coding_scheme'
riscv-none-elf-ld: /home/felipe-moura/Downloads/msav2-connectivity/nuttx/staging/libarch.a(esp_efuse_utility.o): in function `esp_efuse_utility_apply_new_coding_scheme':
/home/felipe-moura/Downloads/msav2-connectivity/nuttx/arch/risc-v/src/chip/esp-hal-3rdparty/components/efuse/esp32c6/esp_efuse_utility.c:159:(.text.esp_efuse_utility_apply_new_coding_scheme+0xe): undefined reference to `esp_efuse_get_coding_scheme'
```
This build isse happens because I need `esp_efuse_get_coding_scheme` compiled to use efuse driver. But in the `hal_esp32c6.mk` file, the compilation for `esp_efuse_get_coding_scheme` only happen in simple_boot

eren-terzioglu, 2025-03-04:
Hi,
Having a look to it. Thanks for notifying.

eren-terzioglu, 2025-03-14:
Hi,
Created 15932

closed 2025-03-14 by acassis