apache/nuttx #15940: [BUG] Compilation error [ESP32-S3-devkit] [config- BLE]

abisop, 2025-03-04:
### Description / Steps to reproduce the issue

Error while compiling for config- esp32s3-devkit:ble
I found that IFF_DOWN was removed earlier

[image]

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 22.04

### NuttX Version

master

### Issue Architecture

[Arch: xtensa]

### Issue Area

[Area: Build System], [Area: Networking]

### Verification

- [x] I have verified before submitting the report.

eren-terzioglu, 2025-03-04:
Hi,
Seems it is building fine on our internal systems. Did you change some settings? If not please don't forget to make `make distclean` and sync your branch to current master

abisop, 2025-03-04:
> If not please don't forget to run `make distclean` and sync your branch to current master

hi eren, 
I am already aware about these things and I am using ble config without changing anything 

Also when using this config Nuttx is forced to compile netlib (apps/netutils/netlib) and hence this error mentioned in the screenshot.

abisop, 2025-03-04:
[image]
I work on master branch only.

[image]
Here is the netlib which we cannot unselect... . If i try unselect it manually other errors occurs.

The thing is I am unable to understand the bluetooth stack (nuttx).

fdcavalcanti, 2025-03-05:
Can you check if your `nuttx-apps` is up-to-date?
Both on master?

abisop, 2025-03-05:
Thanks a lot @fdcavalcanti, it worked.