apache/nuttx #15966: Build error - raspberrypi-pico-w:nsh

phasianus, 2025-03-10:
### Description / Steps to reproduce the issue

Following steps: https://nuttx.apache.org/docs/latest/platforms/arm/rp2040/boards/raspberrypi-pico-w/index.html
Ends with error:
```
/home/pb/nuttxspace/nuttx/tools/mkdeps --obj-path bin --obj-suffix .o --dep-path assert --dep-path builtin --dep-path ctype --dep-path dirent --dep-path dlfcn --dep-path errno --dep-path fixedmath --dep-path grp --dep-path inttypes --dep-path libgen --dep-path machine/arm/armv6-m --dep-path machine/arm/gnu --dep-path machine/arm --dep-path machine --dep-path misc --dep-path net --dep-path obstack --dep-path pthread --dep-path pwd --dep-path queue --dep-path sched --dep-path search --dep-path semaphore --dep-path signal --dep-path spawn --dep-path stdio --dep-path stdlib --dep-path stream --dep-path string --dep-path symtab --dep-path syslog --dep-path termios --dep-path time --dep-path tls --dep-path uio --dep-path unistd --dep-path uuid --dep-path wchar --dep-path wctype "arm-none-eabi-gcc" -- -Wstrict-prototypes -Wno-attributes -Wno-unknown-pragmas -Wno-psabi -Os -fno-strict-aliasing -fomit-frame-pointer --param=min-pagesize=0 -fno-common -Wall -Wshadow -Wundef -ffunction-sections -fdata-sections "-g" -mlittle-endian  -mcpu=cortex-m0 -mthumb -mfloat-abi=soft -mthumb -Wa,-mthumb -Wa,-mimplicit-it=always -isystem /home/pb/nuttxspace/nuttx/include -D__NuttX__ -DNDEBUG -D__KERNEL__  -I /home/pb/nuttxspace/nuttx/libs/libc -- stdlib/lib_strtold.c > lib_strtold.ddc
./stdlib/lib_strtold.c:42:10: fatal error: math.h: No such file or directory
   42 | #include <math.h>
      |          ^~~~~~~~
compilation terminated.
ERROR: arm-none-eabi-gcc failed: 1
       command: arm-none-eabi-gcc -MT bin/lib_strtold.o  -M '-Wstrict-prototypes' '-Wno-attributes' '-Wno-unknown-pragmas' '-Wno-psabi' '-Os' '-fno-strict-aliasing' '-fomit-frame-pointer' '--param=min-pagesize=0' '-fno-common' '-Wall' '-Wshadow' '-Wundef' '-ffunction-sections' '-fdata-sections' '-g' '-mlittle-endian' '-mcpu=cortex-m0' '-mthumb' '-mfloat-abi=soft' '-mthumb' '-Wa,-mthumb' '-Wa,-mimplicit-it=always' '-isystem' '/home/pb/nuttxspace/nuttx/include' '-D__NuttX__' '-DNDEBUG' '-D__KERNEL__' '-I' '/home/pb/nuttxspace/nuttx/libs/libc' ./stdlib/lib_strtold.c
make[2]: *** [/home/pb/nuttxspace/nuttx/tools/Config.mk:232: lib_strtold.ddc] Error 1
make[2]: Leaving directory '/home/pb/nuttxspace/nuttx/libs/libc'
make[1]: *** [Makefile:202: .depend] Error 2
make[1]: Leaving directory '/home/pb/nuttxspace/nuttx/libs/libc'
make: *** [tools/Unix.mk:660: pass2dep] Error 2 
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Fedora 41

### NuttX Version

commit fa5590d5b19f1a224a7863bcc71a6c53ad74af64

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Build System]

### Verification

- [x] I have verified before submitting the report.

ppisa, 2025-03-10:
NuttX allows to select multiple option for math library (LIBM) and availability of thee options depends on the actual target and used toolchain. I expect that you have used `./tools/configure.sh raspberrypi-pico-w:nsh`  for configuration. This configuration does not specify exact LIBM   and Kconfig default choice is `LIBM_TOOLCHAIN` , see [libs/libm/Kconfig](https://github.com/apache/nuttx/blob/master/libs/libm/Kconfig) which is usually the best matching and the most performant library. You do not write which toolchain have you used. On Linux distribution, you usually need to install NewLib or its nano, pico variants separately. I ma not sure how it should be done in the specific case, I have usually my own toolchains setups and their customization.

The problem can be resolved for the most architectures to switch to NuttX provided LIBM, it can be suboptimal, but it is present directly in NuttX sources. Run `make menuconfig` or `make qconfig` and select  

- Math library from NuttX (`LIBM`)

in
```
"Library Routines" -> "Standard C Library Options"
```
The other options are
 
- Math library from toolchain (`LIBM_TOOLCHAIN`)
- Math library from Newlib (`LIBM_NEWLIB`)
- No math library (`LIBM_NONE`)
- Math library from openlibm (`LIBM_OPENLIBM`)

It would worth to add some suggestion about toolchain to use into given platform page in the manual.

May it be it should go into common RaspberryPi rp2040 [install section](https://nuttx.apache.org/docs/latest/platforms/arm/rp2040/index.html#installation) and use of the internal LIBM shoudl be mentioned as workaround. It could be quite slow because RPi1 Cortex-M0+ processor is quite obscure and performance can be used by some integer math accelerator peripherals and if float and double support functions can be written with its existence in mind, may it be even in assembly, than it can be much faster in some SD specific library... 

The corresponding document source (where some clarification can be added) is [Documentation/platforms/arm/rp2040](https://github.com/apache/nuttx/tree/master/Documentation/platforms/arm/rp2040).

pyrossh, 2025-04-22:
Came across this issue while trying to make for rp2350 board. Couldn't figure out the option in `menuconfig`.
Just edited .config file and added. 
`CONFIG_LIBM=y` and it compiled properly.