apache/nuttx #15984: [HELP] menuconfig compiling errors about kconfig

kevinuav, 2025-03-13:
### Description

Hello. I am a UAV developer from px4 community who have encountered compiling errors about nuttx with kconfiglib. I get menuconfig building errors after I reinstalled everything of px4. The very same code can be built successfully before. So I am sure the issue comes from the kconfig’s installation. Could any one tell me what's the problem is it? Thank you!!

here is the errors:
PX4-Autopilot % make px4_fmu-v5_default menuconfig

[0/2] Running NuttX make menuconfig for nsh

CP: arch/dummy/Kconfig to /Users/kevindsp/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/arch/dummy/dummy_kconfig

CP: boards/dummy/Kconfig to /Users/kevindsp/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/../../../../boards/px4/fmu-v5/nuttx-config/Kconfig

LN: platform/board to /Users/kevindsp/PX4-Autopilot/platforms/nuttx/NuttX/apps/platform/dummy

LN: include/arch to arch/arm/include

LN: include/arch/board to /Users/kevindsp/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/../../../../boards/px4/fmu-v5/nuttx-config/include

LN: drivers/platform to /Users/kevindsp/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/drivers/dummy

LN: include/arch/chip to /Users/kevindsp/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/arch/arm/include/stm32f7

LN: arch/arm/src/chip to /Users/kevindsp/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/arch/arm/src/stm32f7

LN: arch/arm/src/board to /Users/kevindsp/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/../../../../boards/px4/fmu-v5/nuttx-config/src

arch/arm/src/imxrt/Kconfig:1806: syntax error

arch/arm/src/imxrt/Kconfig:1805: invalid option

arch/arm/src/rp2040/Kconfig:303: syntax error

arch/arm/src/rp2040/Kconfig:295: missing end statement for this entry

arch/arm/src/rp2040/Kconfig:269: missing end statement for this entry

arch/arm/Kconfig:1250: missing end statement for this entry

arch/arm/Kconfig:7: missing end statement for this entry

Kconfig:1993: missing end statement for this entry

arch/arm/src/rp2040/Kconfig:302: invalid statement

arch/arm/src/rp2040/Kconfig:311: unexpected end statement

arch/arm/src/rp2040/Kconfig:317: unexpected end statement

arch/arm/src/rp2040/Kconfig:336: syntax error

arch/arm/src/rp2040/Kconfig:328: missing end statement for this entry

arch/arm/src/rp2040/Kconfig:335: invalid statement

arch/arm/src/rp2040/Kconfig:344: unexpected end statement

arch/arm/src/rp2040/Kconfig:350: unexpected end statement

arch/arm/src/rp2040/Kconfig:369: syntax error

arch/arm/src/rp2040/Kconfig:361: missing end statement for this entry

arch/arm/src/rp2040/Kconfig:368: invalid statement

arch/arm/src/rp2040/Kconfig:377: unexpected end statement

arch/arm/src/rp2040/Kconfig:383: unexpected end statement

arch/arm/src/rp2040/Kconfig:404: syntax error

arch/arm/src/rp2040/Kconfig:394: missing end statement for this entry

arch/arm/src/rp2040/Kconfig:403: invalid statement

arch/arm/src/rp2040/Kconfig:412: unexpected end statement

arch/arm/src/rp2040/Kconfig:418: unexpected end statement

arch/arm/src/rp2040/Kconfig:437: syntax error

arch/arm/src/rp2040/Kconfig:429: missing end statement for this entry

arch/arm/src/rp2040/Kconfig:436: invalid statement

arch/arm/src/rp2040/Kconfig:445: unexpected end statement

arch/arm/src/rp2040/Kconfig:451: unexpected end statement

arch/arm/src/rp2040/Kconfig:470: syntax error

arch/arm/src/rp2040/Kconfig:462: missing end statement for this entry

arch/arm/src/rp2040/Kconfig:469: invalid statement

arch/arm/src/rp2040/Kconfig:478: unexpected end statement

arch/arm/src/rp2040/Kconfig:484: unexpected end statement

arch/arm/src/rp2040/Kconfig:503: syntax error

arch/arm/src/rp2040/Kconfig:495: missing end statement for this entry

arch/arm/src/rp2040/Kconfig:502: invalid statement

arch/arm/src/rp2040/Kconfig:511: unexpected end statement

arch/arm/src/rp2040/Kconfig:517: unexpected end statement

arch/arm/src/rp2040/Kconfig:536: syntax error

arch/arm/src/rp2040/Kconfig:528: missing end statement for this entry

arch/arm/src/rp2040/Kconfig:535: invalid statement

arch/arm/src/rp2040/Kconfig:546: unexpected end statement

arch/arm/src/rp2040/Kconfig:552: unexpected end statement

arch/arm/src/rp2040/Kconfig:554: unexpected end statement

arch/arm/Kconfig:1251: unexpected end statement

arch/arm/src/stm32l4/Kconfig:5382: syntax error

arch/arm/src/stm32l4/Kconfig:5331: missing end statement for this entry

arch/arm/src/stm32l4/Kconfig:5199: missing end statement for this entry

arch/arm/src/stm32l4/Kconfig:8: missing end statement for this entry

arch/arm/Kconfig:1280: missing end statement for this entry

arch/arm/src/stm32l4/Kconfig:5381: invalid statement

arch/arm/src/stm32l4/Kconfig:5391: unexpected end statement

arch/arm/src/stm32l4/Kconfig:5394: syntax error

arch/arm/src/stm32l4/Kconfig:5393: invalid statement

arch/arm/src/stm32l4/Kconfig:5403: unexpected end statement

arch/arm/src/stm32l4/Kconfig:5406: syntax error

arch/arm/src/stm32l4/Kconfig:5405: invalid statement

arch/arm/src/stm32l4/Kconfig:5415: unexpected end statement

arch/arm/src/stm32l4/Kconfig:5417: unexpected end statement

arch/arm/src/stm32l4/Kconfig:5419: unexpected end statement

arch/arm/src/stm32l4/Kconfig:6294: unexpected end statement

..........

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-03-13:
@kevinuav are you able to compile this board profile using the using "make px4_fmu-v5_default "

I tried here and it is reporting this error:

[  4%] Built target ver_gen
gmake[3]: *** No rule to make target 'src/modules/uxrce_dds_client/dds_topics.h', needed by 'events/px4.json'.  Stop.
gmake[2]: *** [CMakeFiles/Makefile2:18044: src/lib/events/CMakeFiles/events_json.dir/all] Error 2
gmake[2]: *** Waiting for unfinished jobs....
[  4%] Linking CXX static library libyaw_estimator.a
[  4%] Built target yaw_estimator
[  5%] Built target nuttx_arch_build
[  5%] Built target nuttx_c_build
gmake[1]: *** [Makefile:156: all] Error 2
make: *** [Makefile:227: px4_fmu-v5_default] Error 2

I'm using kconfig-frontend and it works fine:

$ make px4_fmu-v5_default menuconfig

kevinuav, 2025-03-14:
Yeah. It can build the target"px4_fmu-v5_default"successfully but "px4_fmu-v5_default menuconfig".

raiden00pl, 2025-03-14:
@kevinuav it looks like you're using some old version of kconfiglib. You need a version that supports the `if` syntax

linguini1, 2026-02-12:
Hello @kevinuav, did raiden's suggestion to use a new kconfiglib version solve this for you?