apache/nuttx #16004: [BUG] <title> PR #15931 breaks ttyACM

jlaitine, 2025-03-17:
### Description / Steps to reproduce the issue

After this PR, I am experiencing USB crashes in the following use case:

1) Have ttyACM serial connection open between the PC and NuttX device (MPFS 64-bit risc-v)
2) Continuously send data from PC to the device, and from device to PC
3) Pull the USB plug

Result:
[CPU0] riscv_exception: EXCEPTION: Load access fault. MCAUSE: 0000000000000005, EPC: 00000000a003490a, MTVAL: 000000014065b788
[CPU0] riscv_exception: PANIC!!! Exception = 0000000000000005
[CPU0] dump_assert_info: Current Version: NuttX  10.4.0 fa125d9e21 Mar 17 2025 14:27:35 risc-v

Reverting commit c497c5feb01c39d20c6a34209679ec719c2e2b43 fixes the issue

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 22.04

### NuttX Version

master

### Issue Architecture

[Arch: risc-v]

### Issue Area

[Area: Drivers]

### Verification

- [x] I have verified before submitting the report.

yangsong8-a1, 2025-03-18:
@jlaitine 
Is the action of unplugging the USB in the process of continuous data transmission? 
What tool is used for continuous data transmission? Can you share it? 
Can you explain the detailed steps to reproduce the problem? I will follow your step to reproduce it. 
thank you！

michallenc, 2025-03-18:
Hi, I also had some issues with CDC ACM after the patch, but only for our custom [Silicon Heaven protocol](https://silicon-heaven.github.io/shv-doc/), wasn't able to reproduce it with a simple application I could share to reproduce. To be short, we use a lot of request-response communication and sometimes the response is not sent from the board to the computer. This happens only after the patch and I don't think it is caused by the protocol as it only uses calls like `fwrite`/`fread`/etc. It seems more like some timing issue with endpoints or requests.

The configuration change of `CONFIG_CDCACM_NWRREQS` and `CONFIG_CDCACM_NRDREQS` from 4 to 8 helped, but this seems more like a hot workaround than a proper solution. I have tested it only on SAMv7 MCU, so it may be related to that MCU as well (the USB there is a mess, it is slow as hell, I am getting 10 times slower communication than I should have, but haven't had time to look at that properly yet).

This is just for info, maybe it is related to this issue, but I don't expect you to invest a time into this as this can't be easily reproduced and might be only SAMv7 related. I plan to look in to it in some days, hopefully.

acassis, 2025-03-18:
@yangsong8-a1 please take a look!

yangsong8-a1, 2025-03-19:
@acassis ok

yangsong8-a1, 2025-03-19:
I tried 3 boards, esp32s3-devkit, raspberrypi-pico-2, sim usbdev, and after exiting the test app, it did not crash. I will try other environments.

jlaitine, 2025-03-19:
Does the FIONREAD ioctl work for you on those platforms?

yangsong8-a1, 2025-03-19:
There is indeed a problem with FIONREAD ioctl. I will solve this problem first.
Could you please help to analyze the crash problem in depth? I haven't found a suitable reproducible environment so far.
Thank you.

jlaitine, 2025-03-19:
> There is indeed a problem with FIONREAD ioctl. I will solve this problem first. Could you please help to analyze the crash problem in depth? I haven't found a suitable reproducible environment so far. Thank you.

I'll come back to this after the FIONREAD ioctl thing is resolved, just in case the issues are related

yangsong8-a1, 2025-03-21:
@jlaitine I uploaded a patch to solve the ioctl problem https://github.com/apache/nuttx/pull/16029/commits/f6b28e91af9fff70ce99fd387dd49f8c2fb93f58.

And a modification was made in reader_writer function in the hello app:
```
          /* NOTE: FIONREAD is broken. remove the following line to
           * experience this
           */

        //bytes_to_read = 1;

          if (bytes_available > 0)
            {
              memset(buffer, 0, sizeof(buffer));
              nread = 0;
            }

          if (retval == 0 && bytes_available >= bytes_to_read)
            {
              int n = 0;
              do
                {
                  n = read(ttyacm_fd, buffer + n, bytes_to_read - nread);
                  nread += n;
                } while (nread < bytes_to_read);

              printf("read %d characters : %s\n", nread, buffer);
            }

          write(ttyacm_fd, hello_world, sizeof(hello_world));
          usleep(100000);
```

jlaitine, 2025-03-24:
Thanks, I'll check further

jlaitine, 2025-03-24:
This doesn't work at all. I tried with a slightly modified test app as well. See the comments in the snippet below. Also, the device still crashes when exiting the test app.

[0001-Replace-helloworld-with-ttyACM-test-app.txt](https://github.com/user-attachments/files/19427462/0001-Replace-helloworld-with-ttyACM-test-app.txt)

```
          int bytes_available = 0;
          int retval = ioctl(ttyacm_fd, FIONREAD, &bytes_available);

          /* Collect 2 characters at a time to the buffer and
           * print them out
           */

          int bytes_to_read = 2;

          if (retval == 0 && bytes_available >= bytes_to_read)
            {
              memset(buffer, 0, sizeof(buffer));

              /* The following should read as many characters as there are, up to sizeof(buffer) bytes.
               * since we only end up here, if FIONREAD above returns >= 2, there should be at least 2 characters
               * /
          
              nread = read(ttyacm_fd, buffer, sizeof(buffer));

              /* But here nread is just 1, and printing out the "buffer" we find out that we received only 1 character! */

              if (nread)
                {
                  printf("got %d bytes: %s\n", nread, buffer);
                }
            }

```

jlaitine, 2025-03-24:
The ttyACM is still completely broken in f3a4011a203a26f07c1ce2e54054c3fd45219fb4

One can only read 1 byte at a time.

If you set bytes_to_read to something a bit bigger (like 5), you don't get anything out any more.

yangsong8-a1, 2025-03-25:
> If you set bytes_to_read to something a bit bigger (like 5), you don't get anything out any more. Also the thing keeps crashing.

In order to reduce the copying of data, the USB reqbuf is directly used as the serial port buf. When CONFIG_CDCACM_NRDREQS is 4, at most 4 packets of USB data are stored without reading. Minicom uses 1 byte as 1 packet of data, so it can only receive 4 bytes. More bytes can be received by expanding CONFIG_CDCACM_NRDREQS.

If bytes_to_read is bigger than 5, it will cause a crash？If so, it should be a driver problem, because when there is no buf, the USB device should return NAK but not crash.

jlaitine, 2025-03-25:
> In order to reduce the copying of data, the USB reqbuf is directly used as the serial port buf. When CONFIG_CDCACM_NRDREQS is 4, at most 4 packets of USB data are stored without reading. Minicom [...]

If I understand you correctly, this is completely unacceptable way of working. It doesn't work at all like other serial drivers!

The data MUST be copied out from an endpoint to a separate buffer as soon as possible. You can't make assumptions on how often the ttyACM client reads out the data. There may be other class drivers running in parallel such as MSC, which need the endpoint buffers as well.

It is a very common optimization technique for serial protocols to wait for n bytes to arrive before processing the serial data. Now this is not working - at all. First of all, the polling of amount of serial data in the buffer is heavy because of big kernel lock, and in addition you can't accumulate data in the buffers.

See for example (close to my heart) PX4: https://github.com/PX4/PX4-Autopilot/blob/735777862d590ef8d428282ab2b26deacfd8c382/src/drivers/cdcacm_autostart/cdcacm_autostart.cpp#L237

I suggest that we revert the patches which cause these issues. Please comment @xiaoxiang781216 , @pussuw , @acassis , others?

xiaoxiang781216, 2025-03-25:
the output side should be OK? let's chagne the input side as an Kconfig option?

jlaitine, 2025-03-25:
> the output side should be OK? let's chagne the input side as an Kconfig option?

Doesn't also the output side unnecessarily block on write? I think the client should be able to push e.g. 120 bytes to the output buffers on a single write, and let the kernel side handle sending out the data. Even if there is not enough room in the CDCDACM buffers for the amount of data. That is why there was the configurable buffer sizes for the CDCACM serial originally!

xiaoxiang781216, 2025-03-25:
if you write 120 bytes, all will be sent in one USB packet.

michallenc, 2025-03-25:
> In order to reduce the copying of data, the USB reqbuf is directly used as the serial port buf. When CONFIG_CDCACM_NRDREQS is 4, at most 4 packets of USB data are stored without reading. Minicom [...]

I think this explains my issues I mentioned earlier. I am not quite sure I like this and I agree with @jlaitine on this. I don't know if revert is necessary, it seems like making it a configuration option should be sufficient, but the default behavior HAS to be as before this patch.

yangsong8-a1, 2025-03-28:
@jlaitine Please try this PR https://github.com/apache/nuttx/pull/16080  to see if it solves this issue. thank you.

closed 2025-03-28 by xiaoxiang781216