apache/nuttx #16044: [HELP] `select()` never returns a positive value when listening to a TCP socket on SoftAP (ESP32-C6)

engdavidiogo, 2025-03-24:
### Description

**Summary**
I'm developing a simple HTTP server that listens for TCP connections on the wlan1 interface (SoftAP mode).
The main loop uses `select()` with a timeout to wait for incoming connections. However, even when a client is connected and sending valid HTTP requests, the `select()` function never returns a positive value, which prevents `accept()` from being called.

**The Problem**
 - The TCP socket is properly configured using `socket()`, `bind()`, and `listen()`, associated with the IP address of the SoftAP interface (`wlan1`).
 - The main loop uses `select()` with a 1-second timeout to wait for activity.
 - Even when a **client is connected** and sends an **HTTP request, `select()` always returns 0** (timeout).
 - As a result, `FD_ISSET(server_sock, &read_fds)` is never true.
 - This prevents `accept()` from being called, making the HTTP server **completely inaccessible**, even with active network traffic.

**Expected behavior**
 - A client connects to the SoftAP (`wlan1`).
 - The client sends a valid HTTP request (e.g., `GET /state`).
 - `server_sock` should be marked as readable.
 - `select()` should return 1.
 - `FD_ISSET(server_sock, &read_fds)` should be true.
 - `accept()` should be called and the client handled.

**Current Behavior**
 - `select()` **always returns** `0` (timeout), even when the client is actively connected and sending requests.
 - `FD_ISSET(server_sock, &read_fds)` is never true.
 - `accept()` is never called.
 - The server remains "active", but is completely unresponsive.

**Socket Configuration Example**

```c
server_sock = socket(AF_INET, SOCK_STREAM, 0);

netlib_get_ipv4addr("wlan1", &softap_ip);

server_addr.sin_family = AF_INET;
server_addr.sin_port = htons(80);
server_addr.sin_addr = softap_ip;

bind(server_sock, (struct sockaddr *)&server_addr, sizeof(server_addr));
listen(server_sock, 5);
```

**Main Loop with `select()`**

```c
while (webserver_running)
{
    fd_set read_fds;
    struct timeval timeout;

    FD_ZERO(&read_fds);
    FD_SET(server_sock, &read_fds);

    timeout.tv_sec = 1;
    timeout.tv_usec = 0;

    int sel_ret = select(server_sock + 1, &read_fds, NULL, NULL, &timeout);

    if (sel_ret < 0)
    {
        break; // select() failed
    }
    else if (sel_ret == 0)
    {
        continue; // Timeout, no activity
    }

    if (FD_ISSET(server_sock, &read_fds))
    {
        int client_sock = accept(server_sock, ...);
        if (client_sock >= 0)
        {
            handle_client(client_sock);
        }
    }
}
```

**Questions**
1. Is this behavior **expected** when using NuttX sockets in SoftAP mode?

2. Is there any known limitation with `select()` or socket monitoring on the `wlan1` interface?

3. Is the way I'm using `select()` appropriate in this context? Am I missing something important?

4. Could it be that the current implementation of `select()` is not correctly monitoring `SOCK_STREAM` sockets on the SoftAP (`wlan1`) interface?

*I would be grateful for any analysis or guidance!*

**Platform**
**Board:** *ESP32-C6 (ESP32C6-DevKitC)*

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-03-25:
@tmedicci @fdcavalcanti @eren-terzioglu please take a look?

@engdavidiogo some questions:

1) Is this code working in normal WiFi mode (not SoftAP) ?
2) Did you test it on POSIX OS (Linux, MacOS) ?

engdavidiogo, 2025-03-26:
Sorry for the delay!

@acassis, thank you for the questions.

To answer them, I used a code sample attached at the end of this comment.

Here’s what I found:

1. When I use the `select()` function on an **ESP32-C6** with the `esp32c6-devkitm:wifi` configuration, I observe the same behavior where `select()` never returns.
Evidence below:

[image]

2. When I use the `select()` function in **POSIX *(Ubuntu)*** systems with the `sim:tcpblaster` configuration, everything seems to work as expected!
Evidence below:

[image]

3. Unfortunately, I don’t currently have another board with the necessary features for testing, but I can arrange one if needed.

---

Example code:

```c
#include <stdio.h>
#include <string.h>
#include <stdbool.h>
#include <stdlib.h>
#include <unistd.h>
#include <fcntl.h>
#include <errno.h>
#include <sys/socket.h>
#include <arpa/inet.h>
#include <pthread.h>
#include <netinet/in.h>
#include <sys/select.h>

#define SERVER_PORT 80
#define MAX_BUFFER 1024

static int server_sock = -1;
static volatile bool server_running = true;

static void handle_client(int client_sock)
{
    char buffer[MAX_BUFFER];
    ssize_t nbytes = recv(client_sock, buffer, MAX_BUFFER - 1, 0);
    if (nbytes <= 0)
    {
        close(client_sock);
        return;
    }
    buffer[nbytes] = '\0';

    if (strstr(buffer, "GET /state") != NULL)
    {
        const char *response_json = "{\"status\":\"ok\"}\r\n";
        char response[MAX_BUFFER];
        snprintf(response, sizeof(response),
                 "HTTP/1.1 200 OK\r\n"
                 "Content-Type: application/json\r\n"
                 "Content-Length: %zu\r\n"
                 "Connection: close\r\n\r\n"
                 "%s",
                 strlen(response_json), response_json);
        send(client_sock, response, strlen(response), 0);
    }
    else
    {
        const char *not_found = "HTTP/1.1 404 Not Found\r\nContent-Length: 0\r\nConnection: close\r\n\r\n";
        send(client_sock, not_found, strlen(not_found), 0);
    }

    close(client_sock);
}

static void *webserver_thread(void *arg)
{
    struct sockaddr_in server_addr, client_addr;
    socklen_t client_len = sizeof(client_addr);

    server_sock = socket(AF_INET, SOCK_STREAM, 0);
    if (server_sock < 0)
    {
        perror("Error creating server socket");
        return NULL;
    }

    int opt = 1;
    setsockopt(server_sock, SOL_SOCKET, SO_REUSEADDR, &opt, sizeof(opt));

    server_addr.sin_family = AF_INET;
    server_addr.sin_port = htons(SERVER_PORT);
    server_addr.sin_addr.s_addr = htonl(INADDR_ANY);

    if (bind(server_sock, (struct sockaddr *)&server_addr, sizeof(server_addr)) < 0)
    {
        perror("Error binding socket");
        close(server_sock);
        return NULL;
    }

    if (listen(server_sock, 5) < 0)
    {
        perror("Error starting listen");
        close(server_sock);
        return NULL;
    }

    printf("HTTP Server started on port %d\n", SERVER_PORT);

    while (server_running)
    {
        fd_set read_fds;
        struct timeval timeout;

        FD_ZERO(&read_fds);
        FD_SET(server_sock, &read_fds);

        timeout.tv_sec = 1;
        timeout.tv_usec = 0;

        int ret = select(server_sock + 1, &read_fds, NULL, NULL, &timeout);
        printf("value returned by select -> %d\n", ret);
        if (ret < 0)
        {
            perror("select failed");
            break;
        }
        else if (ret == 0)
        {
            continue; // Timeout
        }

        if (FD_ISSET(server_sock, &read_fds))
        {
            int client_sock = accept(server_sock, (struct sockaddr *)&client_addr, &client_len);
            if (client_sock >= 0)
            {
                handle_client(client_sock);
            }
        }
    }

    close(server_sock);
    return NULL;
}

int main(int argc, char *argv[])
{
    pthread_t tid;
    pthread_create(&tid, NULL, webserver_thread, NULL);
    pthread_detach(tid);

    while (1)
    {
        sleep(60);
    }

    return 0;
}

```

fdcavalcanti, 2025-03-26:
@engdavidiogo please enable `CONFIG_NET_TCP_WRITE_BUFFERS` and try again.

closed 2025-03-27 by engdavidiogo

fdcavalcanti, 2025-03-27:
> [@fdcavalcanti](https://github.com/fdcavalcanti) should we modify CONFIG_NET_TCPBACKLOG to be enabled by default instead of disabled by default? I think it is better have a more complete kernel image that works than a minimal kernel with network that doesn't work. Someone that doesn't need this feature and want to reduce kernel size will be able to disable it manually later

I support this. I would put this in the category of things that should "just work".

acassis, 2025-03-27:
Exactly! We want NuttX users to have a great first experience, not to enable Network and TCP/IP and important things are not working.

xiaoxiang781216, 2025-03-30:
> What do you think [@xiaoxiang781216](https://github.com/xiaoxiang781216) [@raiden00pl](https://github.com/raiden00pl) ?

Yes, or we shouldn't generate the implementation of accept into binary, so the user could get link error and fix the config instead debugging the runtime problem.