apache/nuttx #16093: [HELP] Proper multi-directory custom app integration

stbenn, 2025-03-28:
### Description

I am having issues with make recognizing file changes in my custom app. As a result, I need to `make clean` before every build which is not ideal. 

I believe the root cause is my app having `src` and `include` directories instead of being contained in one directory like the custom app example in the NuttX documentation. 

## My Project Structure
I made a sample git repo that demonstrates how I have the project set up (https://github.com/stbenn/NuttxCustomAppQuestion.git) to reference in addition to the following explanation.

Directory structure:
```
ProjectDir/
  nuttx/
  apps/
    external/
      Make.defs
      Makefile
      custom_app/ (symlink)
  custom_app/
    /include
      header_files.h
    /src
      custom_app.c
      other_sources.c
    KConfig
    Make.defs
    Makefile
```

Command used to link custom app directory: in `ProjectDir`, `ln -sr custom_app apps/external/custom_app`

### `apps/external/Makefile`
```
MENUDESC = "External Apps"

include $(APPDIR)/Directory.mk
```

### `apps/external/Make.defs`
```
include $(wildcard $(APPDIR)/external/*/Make.defs)
```

### `custom_app/Make.defs`
```
ifneq ($(CONFIG_CUSTOM_APP),)
CONFIGURED_APPS += $(APPDIR)/external/custom_app

endif
```

### `custom_app/Makefile`
```
include $(APPDIR)/Make.defs

ifeq ($(CONFIG_CUSTOM_APP),y)

PROGNAME = $(CONFIG_CUSTOM_APP_PROGNAME)
PRIORITY = $(CONFIG_CUSTOM_APP_PRIORITY)
STACKSIZE = $(CONFIG_CUSTOM_APP_STACKSIZE)

MAINSRC = src/custom_app.c
CSRCS = src/other_sources.c

# This is how to add the include directory to the build system.
# the path to the directory should match AFTER the symlink. So here it's apps/external/custom_app/include
CFLAGS += ${shell $(INCDIR) "$(CC)" "$(APPDIR)$(DELIM)external$(DELIM)custom_app$(DELIM)include"}

endif

include $(APPDIR)/Application.mk

.PHONY: clean
clean::
	rm -f src/*.su
```
Note: I have also experimented with using `CFLAGS += ${INCDIR_PREFIX}${APPDIR}/external/custom_app/include` in `custom_app/Make.defs`

## Desired Behavior
The NuttX build system detects when changes are made to files in the subdirectories of `custom_app` without requiring a `make clean` before `make`.

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-03-31:
@stbenn is this happing only with Makefile build? Or do it happen with CMake too?

@xiaoxiang781216 any idea?

stbenn, 2025-03-31:
I have only tried the makefile build, so I can't comment on if the behavior occurs in CMake build.

TimJTi, 2025-04-01:
I had issues with this and ended up creating a traditional makefile with the paths declared etc. I still had some issues - can't remember what to be honest - so ended up relocating my custom apps directory as a "myapps" subdirectory of the main nuttx-apps tree and it has worked well since.

Makefiles are not my strong point - by any stretch of the imagination - but I suspect that an out-of-tree custom apps folder needs to be specifically included in default NuttX makefiles/make.defs/whatever for it to play ball.

Or there's some magic incantation needed in the custom apps make.defs that makes it all work - I just don't know.

But, instead, perhaps there's a case for a new Kconfig entry in the Build Setup menus for `CONFIG_CUSTOM_APPSDIR` that is then included by default in the NuttX build? I can play with that and see if it works - but won't be for a couple of weeks as I'm going away for a short break.

Not related to this, but I am also looking at how to include a custom file outside of the nuttx-apps tree but used by an in-tree app. That might be easiest with a `CONFIG_CUSTOM_APPINCDIR` which ties in with the above and why I am happy to try this out if it's a viable solution?

TimJTi, 2025-04-01:
Unless the "clue" is in the help text of the Kconfig setting for `CONFIG_APPS_DIR` ? Seems to be clues there that those who know makefiles better than me (pretty much anyone lol) can make use of?

stbenn, 2025-04-01:
@TimJTi based on your Nuttx development environment journey explanation on your profile, you might find this interesting: https://github.com/stbenn/nuttx_docker_env

It is how I make dev environments consistent and documented in project source code. I use WSL and haven't tried a JLink yet, so your millage may vary. I anticipate it being even better when run on a native Linux machine, due to Docker preferring Linux