apache/nuttx #16097: [BUG] NUCLEO-STM32F767zi can not initialize /dev/buttons

vrmay23, 2025-03-30:
### Description / Steps to reproduce the issue

1. Configure the target device as: ./tools/configure.sh nucleo-f767zi:evalos or ./tools/configure.sh nucleo-f767zi:nsh
2. run: make menuconfig
3. active the flag: board_selection / button interrupt support [x]
4. active the flag: device_drive / input device support [x]
5. active the flag: device_drive / input device support / Button inputs [x]
6. active the flag; application_configuration / examples / button driver example [x] 
7. Make sure Button device path has the following path: /dev/buttons
8. save
9. exit menu config
10. make
11. flash via openOCD: openocd -f interface/stlink.cfg -f target/stm32f7x.cfg -c "init" -c "reset halt" -c "flash write_image erase nuttx.bin 0x08000000" -c "reset run"

**[bad_behaviour]**

Error while using the examples 'buttons'. It seems the driver can not be invoked even when all needed configuration was set as expected via menuconfig. Basically when you type "buttons" at nsh, the following message appears:

"buttons_main: Starting the button_daemon"
"buttons_main: button_daemon started"
"button_daemon: Running"
"button_daemon: Opening /dev/buttons"
**"button_daemon: ERROR: Failed to open /dev/buttons: 2"**
"button_daemon: Terminating"

**[expected_behaviour]**
When all steps are performed properly, as written above, once 'buttons' application is called on Nuttx Shell, this app must start to run.

**[workaround_solution]**

To solve this issue I proceed like below:
A. Navigate up to ~/nuttxspace/nuttx/boards/arm/stm32f7/nucleo-f767zi/src
B.  Open the file 'stm32_bringup.c'
C. Add the following code to this file, just below the line 90:

```
#ifdef CONFIG_INPUT_BUTTONS
  ret = btn_lower_initialize("/dev/buttons");
  if (ret<0)
    {
      syslog(LOG_ERR, "ERROR: btn_lower_initialize() failed: %d\n", ret);
    }
#endif
```

Then, after it, just ran make, then flashed the .bin onto the microcotroller. When I've called the application 'buttons', this time it worked as expected and it was possible to use the buttons app!

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.10

### NuttX Version

NuttX 12.8.0 c4f142e45c-dirty Mar 30 2025 17:30:50 arm nucleo-f767zi  

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Board support]

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-03-31:
@vrmay23 please create a PR with the fix.

@raiden00pl PTAL

casaroli, 2026-11-08:
Fixed by #16307.

closed 2026-11-08 by casaroli

vrmay23, 2026-11-08:
Sorry. My apologies!
I forgot about this issue.
I have if fixed. @casaroli have you fix it?
Thanks!