apache/nuttx #16099: [HELP] romfs on esp32

Laczen, 2025-03-31:
### Description

I am trying to mount a romfs filesystem at a specific location in flash. I am using the romfs sample.

I have placed the testdir.img at a specific location in flash (0x200000) and modified desc.image:

`desc.image = (FAR uint8_t *)0x200000;`

However when mounting the romfs filesystem this causes a User Exception, so I guess I cannot just do it like this.

What would be the correct way to mount a romfs filesystem at a specific location in flash ? 

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-03-31:
@lupyuen did you try something like this in the past?

lupyuen, 2025-03-31:
Hi @Laczen: For my RISC-V and Arm64 Ports of NuttX, I mount the ROM FS like this...
https://lupyuen.org/articles/avaota.html#mount-the-romfs
```c
  struct boardioc_romdisk_s desc;
  desc.minor    = RAMDISK_DEVICE_MINOR;
  desc.nsectors = NSECTORS((ssize_t)__ramdisk_size);
  desc.sectsize = SECTORSIZE;
  desc.image    = __ramdisk_start;
  int ret = boardctl(BOARDIOC_ROMDISK, (uintptr_t)&desc);
```

__ramdisk points to a Memory Region defined in the Linker Map:
https://lupyuen.org/articles/avaota.html#linker-script
```c
/* Linker Script: We added the RAM Disk (16 MB) */
MEMORY {
  dram (rwx)    : ORIGIN = 0x40800000, LENGTH = 2M
  pgram (rwx)   : ORIGIN = 0x40A00000, LENGTH = 4M    /* w/ cache */
  ramdisk (rwx) : ORIGIN = 0x40E00000, LENGTH = 16M   /* w/ cache */
}

/* We'll reference these in our code */
__ramdisk_start = ORIGIN(ramdisk);
__ramdisk_size  = LENGTH(ramdisk);
__ramdisk_end   = ORIGIN(ramdisk) + LENGTH(ramdisk);
```

And I copied the ROM FS, from the NuttX Image into the above Memory Region:
https://lupyuen.org/articles/avaota.html#copy-the-romfs

Would this work for ESP32?

Laczen, 2025-04-01:
@lupyuen thanks for the reply.

For the esp32 the linker scripts are a mystery for me.

Maybe @tmedici or @eren-terzioglu know how to reserve a flash region for the romdisk in the linker script and how to access it. It would be great to allow updates (e.g. python mounted in /usr/local/lib) while not modifying the os.

closed 2025-04-01 by Laczen

Laczen, 2025-04-14:
@tmedicci (and others if interested) it is also possible to mount a romfs system directly on a mtd device.

I created a mtd device /dev/mtdblock0 starting at 0x200000 and wrote a generated romfs file directly to 0x200000. This can be mounted using:
```
mount -t romfs -o MS_RDONLY /dev/mtdblock0 /mnt/romfs
```

It is then possible to use /mnt/romfs to access the files in the romfs filesystem. This is a great solution for allowing updates to a romfs without updating the kernel.

acassis, 2025-04-14:
@Laczen maybe it could become a Guide here: https://nuttx.apache.org/docs/latest/guides/index.html

How are you updating the romfs without updating the kernel? Do you just flash a new version at that 0x200000 position?

acassis, 2025-04-14:
That is really cool approach! Maybe using it with progmem you can have create OTA apps update

acassis, 2025-04-15:
@Laczen cool! If you get some inspiration please create a Guide showing how to do an Application update using this ROMFS approach with ELF support

acassis, 2025-06-20:
@Laczen very nice! About wear levelling for NOR Flash, why don't you use SmartFS or LittleFS? It has support to it.

Yes, the things you discovered about how to use ROMFS with MTD should be very useful. If can create a video for it, do you have a good suggestion for a use case, something that show to work: "NuttX is so cool :-)" TM!

Laczen, 2025-06-24:
@tmedicci @acassis the use of romfs with MTD is already covered in the elf example at the end of the sample.

Laczen, 2025-06-25:
> I tried to search for it here: https://nuttx.apache.org/docs/latest/search.html?q=romfs+mtd&check_keywords=yes&area=default but didn't find it. What is the name of your elf sample?

It is the elf sample.

acassis, 2025-06-25:
Understood, then it is another "hidden feature" of NuttX. There are many feature on NuttX that we only discover that exist accidentally reading the code. There are many examples, the last one that I remember I documented here:
https://acassis.wordpress.com/2024/07/02/how-a-supposed-malfunction-revealed-another-hidden-feature-of-nuttx/

I will try to use your sample and write a documentation.