apache/nuttx #16124: [HELP] <ReleaseNotes> Shall we have a summary/highlight new features in new nuttx version ReleaseNotes

jinliangli, 2025-04-03:
### Description

There are too many pull request list in our ReleaseNotes,  and new features and bug fixes are mixed together. 

Shall we have a summary/highlight new features in new nuttx version ReleaseNotes ?
For example,  https://kernelnewbies.org/Linux_6.13

### Verification

- [x] I have verified before submitting the report.

raiden00pl, 2025-04-03:
I think this would be a great improvement that could help with marketing purposes. For potential new users, a list with new features will be much more enticing than a PR list.
But currently the release process is done by one person - @jerpelea. It's not ok to pass on extra work to Alin, so we would need more people wiling to help and improve the process.

jerpelea, 2025-04-03:
@simbit18 https://github.com/apache/nuttx/blob/master/tools/doreleasenotes.py is the tool for release notes

jerpelea, 2025-04-03:
compared with previews releases the commit/pr quality has improved but there is till place for improvement