apache/nuttx #16133: [BUG] RP2350 ostest stuck when SMP enabled

avgoor, 2025-04-04:
### Description / Steps to reproduce the issue

A RaspberryPi's PICO 2W board with a RP2350 MCU is experiencing weird hangs in `ostest` during the nested signals testing when SMP is enabled. It loops forever inside the `spin_lock_notrace` function.
The traces I managed to collect:

```
info threads
Index Tid  Pid  Cpu  Thread                Info                                                                             Frame
 0    0    0    0 '\000' Thread 0x20003d08     (Name: CPU0 IDLE, State: Assigned, Priority: 0, Stack: 1008) 0x100105b2  up_idle() at chip/rp23xx_idle.c:94
 1    1    0    1 '\001' Thread 0x20003dd8     (Name: CPU1 IDLE, State: Assigned, Priority: 0, Stack: 1008) 0x100105b2  up_idle() at chip/rp23xx_idle.c:94
 2    2    2    0 '\000' Thread 0x20006698     (Name: nsh_main, State: Waiting,Semaphore, Priority: 100, Stack: 2008) 0x10005086        nxsem_wait_slow() at semaphore/sem_wait.c:207
 12   12   12   0 '\000' Thread 0x20007880     (Name: ostest, State: Waiting,Semaphore, Priority: 100, Stack: 2016) 0x10005086  nxsem_wait_slow() at semaphore/sem_wait.c:207
 13   13   13   1 '\001' Thread 0x200084e8     (Name: ostest, State: Assigned, Priority: 100, Stack: 8120)  No symbol with pc
*14   62   13   1 '\001' Thread 0x2000a9d8     (Name: ostest, State: Running, Priority: 101, Stack: 8176)   0x1000290c  enter_critical_section_wo_note() at include/nuttx/spinlock.h:199
*15   63   13   0 '\000' Thread 0x2000aab8     (Name: ostest, State: Running, Priority: 102, Stack: 8176)   0x1000290c  enter_critical_section_wo_note() at include/nuttx/spinlock.h:199

bt
#0  0x1000290c in spin_lock_notrace (lock=0x200040c8 <g_cpu_irqlock> "\001") at include/nuttx/spinlock.h:199
#1  enter_critical_section_wo_note () at irq/irq_csection.c:183
#2  0x1000c754 in uart_xmitchars (dev=0x2000121c <g_uart0port>) at serial/serial_io.c:62
#3  0x10000e54 in up_interrupt (irq=49, context=0x0, arg=0x2000121c <g_uart0port>) at chip/rp23xx_serial.c:617
#4  0x10002836 in irq_dispatch (irq=49, context=0x0) at irq/irq_dispatch.c:144
#5  0x10001b64 in exception_direct () at armv8-m/arm_doirq.c:62
#6  <signal handler called>
#7  spin_lock_notrace (lock=0x200040c8 <g_cpu_irqlock> "\001") at include/nuttx/spinlock.h:199
#8  enter_critical_section_wo_note () at irq/irq_csection.c:234
#9  0x10005376 in nxsig_deliver (stcb=0x2000aab8) at signal/sig_deliver.c:178
#10 0x10001e9e in arm_sigdeliver () at armv8-m/arm_sigdeliver.c:107
#11 0x10005fb8 in nxsched_remove_self (tcb=0x40) at sched/sched_removereadytorun.c:280
#12 0x00000000 in ?? ()

list
194     {
195     #ifdef CONFIG_TICKET_SPINLOCK
196       int ticket = atomic_fetch_add(&lock->next, 1);
197       while (atomic_read(&lock->owner) != ticket)
198     #else /* CONFIG_TICKET_SPINLOCK */
199       while (up_testset(lock) == SP_LOCKED)
200     #endif
201         {
202           UP_DSB();
203           UP_WFE();

info args
lock = 0x200040c8 <g_cpu_irqlock> "\001"
```
Additional facts:
- console to the board is connected via UART0
- the issue reproduces 100% of times when running the `ostest` utility
- the `smp` utility runs without problems, no issues found
- the issue does not reproduce on the older RP2040 MCU (different ARM cores)
- the issue does not reproduce when `CONFIG_SMP_NCPUS=1` but SMP is enabled
- the issue reproduces even when `RP23XX_TESTSET_SPINLOCK` is changed from `0` to `31` (see the RP2350-E2 erratum)
- the issue reproduces with today's `master`

The output of the `ostest` utility often times is partially cut off:
```
...
user_main: nested signal handler test
signest_test: Starting signal waiter task at priority 101
signest_test: Started waiter_main pid=62
waiter_main: Waiter started
signest_test: Starting interfering task at priority 102
waiter_main: Setting signal mask
interfere_main: Waiting on semaphore
waiter_main: Registering signal handler
signest_test: Started interfere_main pid=63
waiter_main: Waiting on semaphore
signest_test: Simple case:
  Total signalled
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

ArchLinux, Debian

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Kernel]

### Host information

I use two build environments, in both the issue is reproducing 100% of times.
1: x86_64 PC with ArchLinux and the `arm-none-eabi-*` embedded toolchain
2: aarch64 VM with Debian and the `arm-none-eabi-*` embedded toolchain

### Verification

- [x] I have verified before submitting the report.

keever50, 2025-04-05:
Hey, is this problem in any way or form related to https://github.com/apache/nuttx/issues/16139 ?
Maybe there are details in common. I've had these strange hangs for a while and i feel as if it got worse. But all on the RP2040.
I see that you have no problems with the RP2040, but maybe you have a specific situation where it does not trigger the hang.

avgoor, 2025-04-06:
Hey @keever50. I've rebuilt my debug SMP build for RP2040 with USB console and found no issue. `ostest` runs till the end and exits without any error. Maybe you need to try add SMP to your configurations?

shtirlic, 2025-04-20:
@keever50 @avgoor I got ostest  crash in the same signal test, could you try the my commit https://github.com/apache/nuttx/commit/45478e5d5e71f647cb6ce0ee3f2b920e637410a7 it fixes for me the ostest, rp2350 SMP mode disabled.

keever50, 2025-04-20:
> [@keever50](https://github.com/keever50) [@avgoor](https://github.com/avgoor) I got ostest crash in the same signal test, could you try the my commit [45478e5](https://github.com/apache/nuttx/commit/45478e5d5e71f647cb6ce0ee3f2b920e637410a7) it fixes for me the ostest, rp2350 SMP mode disabled.

That is really nice, however, I never used SMP (atleast, not that i am aware of in the usbnsh) and i am using the RP2040. Is it possible that this also fixes RP2040 non SMP?
I will test this out later.

keever50, 2025-04-21:
> Thanks for the patch! I've tested it on my setup and unfortunately it
> didn't help. I've forgot to say that I build nuttx without optimizations
> (-O0) from the beginning, specifically to exclude any compiler related

Thanks, however, for me the same result. It did not help.
I did similar fixes like this that appeared to have helped, but really just
"shifts" around the issue elsewhere.
Such as the hangs and crashes happening at a different time.
I am afraid there is memory corruption happening somehow.

shtirlic, 2025-04-22:
@keever50 @avgoor  yes, looks like it's just shifting the code, I already tried with rp2350 errata fixes and sched lock/unlock updates, it did not help with ostest, but helped at least boot the SMP (2cores) config on my pico2. I think it's time to attach SWD debugger and see what's really going on.

tmedicci, 2025-04-22:
This seems to be related to https://github.com/apache/nuttx/issues/16193

Can you roll back to https://github.com/apache/nuttx/pull/16030 and check if the SMP + ostest's signest failed after applying it?

avgoor, 2025-04-22:
Hi @tmedicci, unfortunately rewinding back to pre-#16030 commit doesn't help. 
In many runs `ostest` hangs even sooner, the last output is:
```
user_main: pthread_rwlock test
pthread_rwlock: Initializing rwlock
pthread_exit_thread 30: Exiting
```

Once it went further, to the signest test and even produced there some dump:

```
user_main: nested signal handler test
signest_test: Starting signal waiter task at priority 101
signest_test: Started waiter_main pid=53
waiter_main: Waiter started
signest_test: Starting interfering task at priority 102
waiter_main: Setting signal mask
interfere_main: Waiting on semaphore
waiter_main: Registering signal handler
signest_test: Started interfere_main pid=54
waiter_main: Waiting on semaphore
[CPU0] dump_assert_info: Current Version: NuttX  12.9.0 504f838577-dirty Apr 22 2025 17:01:39 arm
[CPU0] dump_assert_info: Assertion failed : at file: :0 task(CPU0): ostest process: ostest 0x10024ccd
[CPU0] up_dump_register: R0: 20001510 R1: 00000000 R2: 20001510  R3: 20001510
[CPU0] up_dump_register: R4: 20003790 R5: 20003770 R6: 00000000  FP: 00000000
[CPU0] up_dump_register: R8: 00000000 SB: 00000000 SL: 00000000 R11: 00000000
[CPU0] up_dump_register: IP: 00000000 SP: 200090c8 LR: 10003141  PC: 10003141
[CPU0] up_dump_register: xPSR: 40000000 BASEPRI: 00000040 CONTROL: 00000002
[CPU0] up_dump_register: EXC_RETURN: 00000000
[CPU0] dump_stackinfo: User Stack:
[CPU0] dump_stackinfo:   base: 0x200072c0
[CPU0] dump_stackinfo:   size: 00008120
[CPU0] dump_stackinfo:     sp: 0x200090c8
[CPU0] stack_dump: 0x200090a8: 40006e10 00000000 40006e10 20006e10 20004ed8 20003790 20003770 100031bd
[CPU0] stack_dump: 0x200090c8: 20001510 2000444b 20001510 00000000 00000000 00000000 4000004d 20006e10
[CPU0] stack_dump: 0x200090e8: 20001510 00000000 00000000 00000000 00000000 20006e10 00006e10 00000040
[CPU0] stack_dump: 0x20009108: 00003790 00000000 20006e10 00000000 00000000 20004436 00000000 00000040
[CPU0] stack_dump: 0x20009128: 00007278 00000000 20004436 20004436 00000000 0000c3bd 20006e10 00007278
[CPU0] stack_dump: 0x20009148: 00000040 0002f697 00000000 00000000 20006e10 00013ca0 24f47300 1000c35f
[CPU0] stack_dump: 0x20009168: 00000013 00000000 00000000 00000000 00000000 100278c9 00000013 00000035
[CPU0] stack_dump: 0x20009188: 00000013 0000004d 24f47300 00000002 10027868 10027ab9 20006fe0 00000036
[CPU0] stack_dump: 0x200091a8: 00000035 00010066 00000000 00000000 00002000 00000066 00000027 10035528
[CPU0] stack_dump: 0x200091c8: 00006b54 00000000 00000013 00000006 00000188 00000188 20003770 10024f17
[CPU0] stack_dump: 0x200091e8: 0007d7ac 00000001 00000023 00078d80 00004a2c 00078d80 00004b98 deadbeef
[CPU0] stack_dump: 0x20009208: 20007288 00000005 deadbeef 00000005 deadbeef 00000000 00000000 1000c84f
[CPU0] stack_dump: 0x20009228: 20009278 20007288 00000005 10024ccd 00000000 100072db 00000000 00000000
[CPU0] stack_dump: 0x20009248: 20006e10 00000000 00000040 00fffffd 00000000 20007288 00000000 20006e10
[CPU0] stack_dump: 0x20009268: 00000005 00000001 1000722c 00000000 00000000 00000000 00000000 00000000
ostest_main: Exiting with status 256
stdio_test: Standard I/O Check: fprintf to stderr
```

Can it be connected with the IRQ handling differences between M0+ and M33 ARM cores? I've seen that the NuttX port to RP2350 is somewhat a copy-paste of the RP2040 port, but the ARM cores are very different between those two.

avgoor, 2025-04-26:
Hi @xiaoxiang781216 and @pussuw, I've seen you closed this issue as resolved by #16262. I've rebuilt everything from scratch with the fresh master branch including this specific commit and unfortunately it doesn't help. The issue persists.
Since it's not fixed please re-open this report.

xiaoxiang781216, 2025-04-26:
Done.

shtirlic, 2025-04-26:
@avgoor I have at least one successful full run of `ostest` after this commit c12aa5663d3046138eb975a8bc872d8348c73e21 . I’m still very interested to hear how you’re able to get rp2350 with SMP(2) running at all from master, because on my side it’s not working fresh from master without the `sched_unlock` patch I posted on the mailing list.

PS My defconfig like this https://github.com/shtirlic/picocalc-nx/blob/main/configs/nsh/defconfig

keever50, 2025-04-26:
ostest also still fails for me on the RP2350 - usbnsh.
It does not get any further than this:
[image]

keever50, 2025-04-26:
I still believe that the issue isn't even centered around SMP specifically. These issues are completely random. Every patch, every PR, the issue just gets moved around in memory.
The RP2040 also still has this problem.

It is not just the RP2350, RP2040 and the ESP(forgot specific). I also notice that my STM32H743 is behaving very strange now. I am stuck in an older nuttx 12.3 version where everything seems to be more stable.
To be more specific about the STM, the USB stack is completely broken. Sometimes, hard to replicate, the ostest also fails at random locations there.

This is a wide spread issue. I hope that developers/maintainers wont keep closing these issues too fast without deep testing.
Maybe its time for a testing board farm.

avgoor, 2025-04-26:
@shtirlic I honestly just used `pimoroni-pico-2-plus:smp` configuration and added debugging options. That's it, nothing more. I connect via UART0, so maybe this is why it works. But `ostest` still gets stuck randomly before `signest` and 100% it gets stuck on `signest`.
@keever50 I'm not sure it's SMP either, however I don't see why it works OK when I set `CONFIG_SMP_NCPUS=1` to limit the CPU count. When only one CPU core is running - everything works fine, doesn't get stuck anywhere. So at least in my brain it looks like a race condition between CPU cores when they lock SPINLOCKs and IRQ. Unfortunately I'm new to NuttX and don't know how it was intended to behave in multi-CPU environments.

keever50, 2025-04-27:
`as if the UART I/O procedure itself interferes with other lock operations.`
I think that is likely. As RP2040 suffers from glitchy and bad UART.
Both on USB and HW UART.
This issue also appeared on the STM32H7, where turning on SYSLOG debug causes crashes.
Suspicious...
It might be worth looking into that.

Perhaps we can run some serial stress tests to confirm?

avgoor, 2025-04-27:
@keever50 It is "running", meaning that the active thread is actively reading the SPINLOCK register for acquiring the spinlock (which never happens for some reason). If you unlock the lock (g_cpu_irqlock in this case) by setting it to 0x0 via debugger the OS is getting unstuck and proceeds running the thread further. But then it gets stuck once again in just couple functions call further down the program. In all my testing I was able to escape from this "deadlock" by manually setting the lock to 0x0 or writing to a corresponding SPINLOCK register of RP2350. So at least in my case it's not like a hardfault when CPU is halted. On the other hand I would not expect any other background thread to run in these moment, I doubt that spinning the lock can be preempted by the scheduler (not sure here).

keever50, 2025-04-29:
@avgoor There have been recent stack overflow issues (not the website). Probably because NuttX is growing.
If you have issues, please enable KASAN (MM_KASAN) and stack canaries in Kconfig.

If KASAN throws panics right away, there is a stack overflow, somewhere. It also shows where and how much.

run "ps" to see the stack usage levels per thread.

[image]

As you can see, it is extremely high. Way too high.
Try to increase this and see if you get the same issues as before.

keever50, 2025-04-29:
I did increase it but we still have hangs if you run ps multiple times.
This happened after the second time i flashed the pico.
What if this is flash related?

[image]

keever50, 2025-04-29:
After enabling KASAN,
everything appears to be "fixed" but i am sure it is not.
Strange.

[image]

avgoor, 2025-04-30:
@keever50 I've enabled KASAN, bumped the stack sizes everywhere but the issue is still there. Once I got a partial debug assert shown:
```
[CPU0] dump_assert_info: Current Version: NuttX  12.9.0 7bbc76f115 Apr 30 2025 09:23:31 arm
[CPU0] dump_assert_info: Assertion failed rtcb != ((void*)0) && rtcb->irqcount > 0: at file: irq/irq_csection.c:379 task(CPU0): ostest process: ostest 0x1002a649
[CPU0] up_dump_register: R0: 20001510 R1: 0000017b R2: 20001510  R3: 20001510
[CPU0] up_dump_regis
```
and that's it.
Not sure what to do now. For me it still looks like a broken logic of locks vs IRQs.

avgoor, 2026-08-21:
Hi @shtirlic , sorry for late reply, somehow missed the notification.

I've tested several configurations built from `0cbde9ebe` and I'd say it's now harder to reproduce.
On RISC-V in SMP config with USB NSH or Serial NSH using both `-O0` and `-O2` the issue is not reproducible after ~30 retries.
On ARM in SMP config with USB NSH or Serial NSH it's not reproduced at `-O2`, but with `-O0` on some occasions it can be triggered. The trace is a bit different now:
```
>>> bt
#0  pause_cpu_handler (arg=0x0) at misc/assert.c:632
#1  0x10004b4e in nxsched_smp_call_handler (irq=41, context=0x0, arg=0x0) at sched/sched_smp.c:131
#2  0x1000163e in rp23xx_smp_call_handler (irq=41, c=0x0, arg=0x0) at chip/rp23xx_smpcall.c:102
#3  0x10002a34 in irq_dispatch (irq=41, context=0x0) at irq/irq_dispatch.c:143
#4  0x10001bce in exception_direct () at armv8-m/arm_doirq.c:63
#5  <signal handler called>
#6  0x10001eda in setbasepri (basepri=0) at nuttx/include/arch/armv8-m/irq.h:344
#7  0x10001f04 in up_irq_enable () at  nuttx/include/arch/armv8-m/irq.h:374
#8  0x10001fce in arm_sigdeliver () at armv8-m/arm_sigdeliver.c:105
#9  0x10022968 in up_interrupt_context () at  nuttx/include/arch/armv8-m/irq.h:500
#10 0x00000080 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
>>> info threads
Index Tid  Pid  Cpu  Thread                Info                                                                             Frame                         
 0    0    0    0 '\000' Thread 0x200014d0     (Name: CPU0 IDLE, State: Assigned, Priority: 0, Stack: 1000) 0x10010862  up_idle() at chip/rp23xx_idle.c:94
 1    1    0    1 '\001' Thread 0x200015a8     (Name: CPU1 IDLE, State: Assigned, Priority: 0, Stack: 1000) 0x10010862  up_idle() at chip/rp23xx_idle.c:94
 3    3    3    0 '\000' Thread 0x20005270     (Name: nsh_main, State: Waiting,Semaphore, Priority: 100, Stack: 2000) 0x10022c34        nxsem_wait_slow() at semaphore/sem_wait.c:290
 10   10   10   0 '\000' Thread 0x200048b8     (Name: ostest, State: Waiting,Semaphore, Priority: 100, Stack: 2008) 0x10022c34  nxsem_wait_slow() at semaphore/sem_wait.c:290
 11   11   11   0 '\000' Thread 0x20004cc8     (Name: ostest, State: Waiting,Semaphore, Priority: 100, Stack: 8112) 0x10022c34  nxsem_wait_slow() at semaphore/sem_wait.c:290
*14   14   11   1 '\001' Thread 0x20008b48     (Name: ostest, State: Running, Priority: 101, Stack: 8168)   0x1000359c  pause_cpu_handler() at misc/assert.c:632
*15   15   11   0 '\000' Thread 0x2000ac38     (Name: ostest, State: Running, Priority: 102, Stack: 8168)   0x1000359c  pause_cpu_handler() at misc/assert.c:632
```
Don't really know what a conclusion can be made here. For me it seems there is still some window when something can get deadlocked during handling of nested signals and serial I/O the same time, and `-O0` makes this window bigger.

avgoor, 2026-10-18:
Something has changed recently and now it gets stuck even on `smp` tests:
```
(gdb) where
#0  0x10002b06 in spin_lock_notrace (lock=0x20002bc4 <g_cpu_irqlock>)
    at nuttx/include/nuttx/spinlock.h:146
#1  0x10002bc2 in enter_critical_section_notrace () at irq/irq_csection.c:183
#2  0x10025400 in wd_expiration (ticks=11802) at wdog/wd_start.c:115
#3  0x1002559c in wd_timer (ticks=11802) at wdog/wd_start.c:362
#4  0x10022b10 in nxsched_process_timer () at sched/sched_processtick.c:192
#5  0x10010a18 in rp23xx_timerisr (irq=15, regs=0x0, arg=0x0) at chip/rp23xx_timerisr.c:90
#6  0x10002a00 in irq_dispatch (irq=15, context=0x0) at irq/irq_dispatch.c:143
#7  0x10001b86 in exception_direct () at armv8-m/arm_doirq.c:63
#8  <signal handler called>
#9  __aeabi_ldivmod () at /build/arm-none-eabi-gcc/src/gcc-16.2.0/libgcc/config/arm/bpabi.S:189
#10 0x100061e4 in clock_systime_timespec (ts=0x20006ac8) at clock/clock_systime_timespec.c:84
#11 0x10021eea in nxclock_gettime (clock_id=1, tp=0x20006ac8) at clock/clock_gettime.c:107
#12 0x10021f90 in clock_gettime (clock_id=1, tp=0x20006ac8) at clock/clock_gettime.c:215
#13 0x1002ac60 in hog_milliseconds (milliseconds=100) at smp_main.c:99
#14 0x1002ad9a in hog_time (caller=0x100500a4 "Thread", threadno=1) at smp_main.c:132
#15 0x1002adde in barrier_thread (parameter=0x1) at smp_main.c:156
#16 0x100301a0 in pthread_startup (entry=0x1002adbd <barrier_thread>, arg=0x1)
    at pthread/pthread_create.c:65
#17 0x10031e9c in pthread_start () at pthread/pthread_create.c:146
#18 0x00000000 in ?? ()
(gdb) inf thr
  Id   Target Id                                                                  Frame 
* 1    Thread 1 "rp2350.dap.core0" (Name: rp2350.dap.core0, state: debug-request) 0x10002b06 in spin_lock_notrace (lock=0x20002bc4 <g_cpu_irqlock>)
    at nuttx/include/nuttx/spinlock.h:146
  2    Thread 2 "rp2350.dap.core1" (Name: rp2350.dap.core1, state: debug-request) 0x10002b06 in spin_lock_notrace (lock=0x20002bc4 <g_cpu_irqlock>)
    at nuttx/include/nuttx/spinlock.h:146
(gdb) 
```

But RISC-V implementation seems to be OK, this is something ARM implementation specific.

avgoor, 2026-10-28:
Hi @shtirlic, can you please take a look at my attempt to fix the issue in #21092?

closed 2026-10-29 by xiaoxiang781216