apache/nuttx #16159: [HELP] SD Card imxrt_dmapreflight: dcache unaligned buffer:0x202458b8 end:0x20245ab7

gustavosousa2208, 2025-04-08:
### Description

I'm trying to use an SD Card with NXP IMXRT1176-EVKB board, i checked the pins for the USDHC and the card is communicating, so no electrical or signal integrity problems. 

mmcsd_decode_csd: CSD:
mmcsd_decode_csd:   CSD_STRUCTURE: 1 SPEC_VERS: 0 (MMC)
mmcsd_decode_csd:   TAAC {TIME_UNIT: 6 TIME_VALUE: 1} NSAC: 0
mmcsd_decode_csd:   TRAN_SPEED {TRANSFER_RATE_UNIT: 2 TIME_VALUE: 6}
mmcsd_decode_csd:   CCC: 1461
mmcsd_decode_csd:   READ_BL_LEN: 9 READ_BL_PARTIAL: 0
mmcsd_decode_csd:   WRITE_BLK_MISALIGN: 0 READ_BLK_MISALIGN: 0
mmcsd_decode_csd:   DSR_IMP: 0
mmcsd_decode_csd:   SD Block Addressing:
mmcsd_decode_csd:     C_SIZE: 30006 SD_ER_BLK_EN: 1
mmcsd_decode_csd:     SD_SECTOR_SIZE: 127 SD_WP_GRP_SIZE: 0
mmcsd_decode_csd:   WP_GRP_EN: 0 MMC DFLT_ECC: 0 (MMC) R2W_FACTOR: 2
mmcsd_decode_csd:   WRITE_BL_LEN: 9 WRITE_BL_PARTIAL: 0
mmcsd_decode_csd:   FILE_FORMAT_GROUP: 0 COPY: 1
mmcsd_decode_csd:   PERM_WRITE_PROTECT: 0 TMP_WRITE_PROTECT: 0
mmcsd_decode_csd:   FILE_FORMAT: 0 ECC: 0 (MMC) CRC: 0
mmcsd_decode_csd: Capacity: 15363584Kb, Block size: 512b, nblocks: 0 wrprotect: 30727168

the problem is 

imxrt_dmapreflight: dcache unaligned buffer:0x202458b8 end:0x20245ab7

i didnt edited the code DMA part, also i have this output when trying to format
nsh> mkfatfs -F 32 /dev/mmcsd0
bchdev_register: blkdev="/dev/mmcsd0" chardev="/dev/tmpc000001" readonly=F
find_blockdriver: pathname="/dev/mmcsd0"
mmcsd_open: Entry
mmcsd_geometry: Entry
imxrt_status: cdstatus=01
mmcsd_geometry: available: true mediachanged: true writeenabled: true
mmcsd_geometry: nsectors: 30727168 sectorsize: 512
mmcsd_geometry: Entry
imxrt_status: cdstatus=01
mmcsd_geometry: available: true mediachanged: false writeenabled: true
mmcsd_geometry: nsectors: 30727168 sectorsize: 512
mkfatfs_clustersearch: Configuring with 32 sectors/cluster...
mkfatfs_tryfat32: nfatsects=7499 nclusters=959753 (max=959872)
mkfatfs_selectfat: Selected FAT32
mkfatfs_configfatfs: WARNING:  Invalid backup boot sector: 0
mkfatfs_configfatfs: Sector size:          512 bytes
mkfatfs_configfatfs: Number of sectors:    30727168 sectors
mkfatfs_configfatfs: FAT size:             32 bits
mkfatfs_configfatfs: Number FATs:          2
mkfatfs_configfatfs: Sectors per cluster:  32 sectors
mkfatfs_configfatfs: FS size:              7499 sectors
mkfatfs_configfatfs:                       959753 clusters
mkfatfs_configfatfs: Volume ID:            00000000
mkfatfs_configfatfs: Volume Label:         "           "
mmcsd_write: startsector: 0 nsectors: 1 sectorsize: 512
mmcsd_writesingle: startblock=0
imxrt_status: cdstatus=01
imxrt_dmapreflight: dcache unaligned buffer:0x202458b8 end:0x20245ab7
bchlib_write: ERROR: Write failed: -14
mkfatfs_devwrite: ERROR:  write failed: size=512 pos=0 error=-14
mmcsd_close: Entry
nsh: mkfatfs: mkfatfs failed: 14

is there a setting i need to tweak in menuconfig?

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-04-10:
@gustavosousa2208 this error was already reported here:

https://github.com/apache/nuttx/issues/12218

@JorgeGzm I remember you fixed this issue, but the issue still opened there and there is no explanation what was the fix. Do you remember how you fixed it?

JorgeGzm, 2025-04-10:
Hi @acassis, 

The error in my case was using the micro stm32h7 and i disable the dma, I don't know if it could be the error he is having.

In the driver of stm32h7/stm32_sdmmc.c there is the following message:

```
 * HW Issues when using IDMA
 *
 *    The DMA buffer must be located in a zone accessible via IDMA.
 * For SDMMC1, IDMA cannot access SRAM123 or SRAM4. Refer to ST AN5200.
 * Buffer validity is checked when CONFIG_ARCH_HAVE_SDIO_PREFLIGHT is set.
 *
 * MDMA is only available on for SDMMC1 and Not supported at this time.
```

acassis, 2025-04-18:
@gustavosousa2208 did you fix the issue?

@davids5 do you remember about this dmapreflight issue? I think you already posted about it in the mailing list some time ago.

keever50, 2025-04-25:
I have the same issue.
The "fix" for me is to disable IDMA transfers STM32H7_SDMMC_IDMA=n
However, I get warned by the system that overruns are possible and i can image SD transfers are not very fast without this feature.
It would be nice if we can use IDMA.

keever50, 2025-04-29:
Little update that is probably very important.

Disabling IDMA will give you overrun warnings. It does not explain what will happen if you ignore these.
If you ignore the warnings and clock the SDMMC too fast, it can cause crashes/panics.

If you have kernel panics, try to increase the clock divider under board.h.
For example in the STM32H7:

[image]
I increased the clock divider to 8. It was lower previously, which caused panics.

snikeguo, 2025-05-17:
@keever50 
solution：

CONFIG_FAT_DMAMEMORY=y
CONFIG_GRAN=y
CONFIG_STM32H7_SDMMC_IDMA=y
CONFIG_ARMV7M_DCACHE_WRITETHROUGH=n   see:armv7m mpu document

in stm32_dmarecvsetup:
#if defined(CONFIG_ARCH_HAVE_SDIO_PREFLIGHT)
  // DEBUGASSERT(stm32_dmapreflight(dev, buffer, buflen) == 0);
#endif

[image]

keever50, 2025-05-18:
> [@keever50](https://github.com/keever50) solution：
>
> CONFIG_FAT_DMAMEMORY=y CONFIG_GRAN=y CONFIG_STM32H7_SDMMC_IDMA=y CONFIG_ARMV7M_DCACHE_WRITETHROUGH=n see:armv7m mpu document
>
> in [...]

Thanks!!

I will try this out soon. Likely end of next week. I hope this will improve the SD performance

keever50, 2025-06-04:
@snikeguo 
This solution worked. Thanks!

I'd like to add that if your board does not contain stm32_dma_alloc.c, you must add this. These can be copied from other existing H7 boards.

13022591351, 2025-06-06:
I am using the STM32H743 SDMMC1 interface and encountered the same problem as you. I am now using CONFIG_MM_REGIONS=1 to avoid using SRAM1234. I can use IDMA normally, but I don't know if it will affect other DMAs.