apache/nuttx #16193: [BUG] esp32-devkitc:smp fails on ostest after #16030

tmedicci, 2025-04-11:
### Description / Steps to reproduce the issue

`esp32-devkitc:smp` eventually fails to run `ostest`'s `signest` test after #16030.

## How to Reproduce

To make it run faster, apply the attached patch to `nuttx-apps` repository to disable other ostest's tests:
```
cd apps
git apply ostest.diff.txt
```

Back to the OS repository, checkout to https://github.com/apache/nuttx/commit/19a8e2403f4d49c3655a9db43d1abc589ae6204e and build the firmware (adjust the `ESPTOOL_PORT` accordingly):
```
make -j distclean && ./tools/configure.sh esp32-devkitc:smp && make flash ESPTOOL_PORT=/dev/ttyUSB1 ESPTOOL_BINDIR=./ -s -j$(nproc) && picocom -b 115200 /dev/ttyUSB1
```

Then, run `ostest` on NSH:
```
nsh> ostest
```

### Multiple Error Outputs

Please note that some of the following errors may occur after one or more attempts (usually, something fails during the first or the second run):

#### `ostest` halts exiting `user_main`

```
nsh> ostest
stdio_test: write fd=1
stdio_test: Standard I/O Check: printf
stdio_test: write fd=2
stdio_test: Standard I/O Check: fprintf to stderr
ostest_main: putenv(Variable1=BadValue3)
ostest_main: setenv(Variable1, GoodValue1, TRUE)
ostest_main: setenv(Variable2, BadValue1, FALSE)
ostest_main: setenv(Variable2, GoodValue2, TRUE)
ostest_main: setenv(Variable3, GoodValue3, FALSE)
ostest_main: setenv(Variable3, BadValue2, FALSE)
show_variable: Variable=Variable1 has value=GoodValue1
show_variable: Variable=Variable2 has value=GoodValue2
show_variable: Variable=Variable3 has value=GoodValue3
ostest_main: Started user_main at PID=4

user_main: Begin argument test
user_main: Started with argc=5
user_main: argv[0]="ostest"
user_main: argv[1]="Arg1"
user_main: argv[2]="Arg2"
user_main: argv[3]="Arg3"
user_main: argv[4]="Arg4"

user_main: nested signal handler test
signest_test: Starting signal waiter task at priority 101
signest_test: Started waiter_main pid=5
waiter_main: Waiter started
signest_test: Starting interfering task at priority 102
waiter_main: Setting signal mask
interfere_main: Waiting on semaphore
waiter_main: Registering signal handler
signest_test: Started interfere_main pid=6
waiter_main: Waiting on semaphore
signest_test: Simple case:
  Total signalled 1240  Odd=620 Even=620
  Total handled   1240  Odd=620 Even=620
  Total nested    0    Odd=0   Even=0  
signest_test: With task locking
  Total signalled 2480  Odd=1240 Even=1240
  Total handled   2480  Odd=1240 Even=1240
[... 608 lines cut ...]
  4.8.1

HAL version:
  sync/release_v5.1.c-nuttx-20230814-1582-g5d4868f08b

CHIP ID:
  ESP32 has no Chip ID. Reading MAC instead.

Flash ID:
  Manufacturer: 20
  Device: 4016

Security information:
  Not found

Flash status:
  0x0000

MAC address:
  70:b8:f6:12:c4:04
```

### Verification

- [x] I have verified before submitting the report.

tmedicci, 2025-04-11:
@jlaitine , can you take a look?

@xiaoxiang781216 , @anchao and @pussuw , you participated on the original discussion for #16030. Do you have any clues about it?

jlaitine, 2025-04-12:
> @jlaitine , can you take a look?
> 
> @xiaoxiang781216 , @anchao and @pussuw , you participated on the original discussion for #16030. Do you have any clues about it?

I remember seeing one older bug, which maybe could have triggered with SMP and CONFIG_PRIORITY_INHERITANCE=n, after the mutex fast paths were introduced first time before this PR. I'll search for it and we can try if fixing that simply fixes the issue. Otherwise I can give a better look on monday.

jlaitine, 2025-04-12:
Please check out 

https://github.com/tiiuae/nuttx/commit/10db59296e7307e16c8dbab3b0c22b6dc1d13f6b 

That is something that hit my eye earlier, and it just looked wrong (it just didn't look like an smp acquire-release). I might be wrong about that, and the fix above is completely untested - except for a single run in qemu. I'll come back to this on Monday, unless someone wants to do testing already during the weekend.

jlaitine, 2025-04-13:
@tmedicci see PR #16197

jlaitine, 2025-04-14:
Tried to re-produce it on multiple HW now, but I don't have anything close to esp32.

Tried the "esp32-devkitc:smp" also on qemu with following changes:

- used your config file, but added "# CONFIG_ARCH_LEDS is not set" to make it compile for qemu
- added "#define ESP32_IGNORE_CHIP_REVISION_CHECK" in arch/xtensa/src/esp32/esp32_start.c to make it run on qemu

qemu-system-xtensa -nographic -machine esp32 -drive file=nuttx.merged.bin,if=mtd,format=raw -smp 2

This is the closest I can get to esp32, I don't have any esp32 boards....

Still can't re-produce the issue.

I wonder what does the patches actually even change in this case, when CONFIG_PRIORITY_INHERITANCE is not defined; functionally the patch should be no-op. The fastpath functions are just moved to another file, the changed IF clause is not even compiled in.

I'll also compare the  assembly before and after....

jlaitine, 2025-04-14:
The signest test is also not even using mutexes.... Could you @tmedicci verify that you are not overflowing ostest stack causing some random behavior? Maybe just  try to double it for a quick test to see if that makes a difference?

tmedicci, 2025-04-14:
> The signest test is also not even using mutexes.... Could you [@tmedicci](https://github.com/tmedicci) verify that you are not overflowing ostest stack causing some random behavior? Maybe just try to double it for a quick test to see if that makes a difference?

Thanks for your attempts at testing it. I increased `ostest`'s stack size (to 16384) and the default stack size to 4096. Nothing changed: the device still halts in the same place.

Just some additional information:

> Tried to re-produce it on multiple HW now, but I don't have anything close to esp32.
>
> Tried the "esp32-devkitc:smp" also on qemu with following changes:
>
> * used your config file, but added "# [...]

I couldn't reproduce it on QEMU too, but it happens independently of the board (I tried with several different ESP32 boards). Just reverting the commit, it gets fixed.

Can I provide more information or anything else that could help?

jlaitine, 2025-04-14:
Is the address where it is crashing some sane address? Could you post your nuttx elf file, or assembly objdump of the crashing address?

I managed to make a configurarion for mpfs, which crashes in the same test occasionally. But I don't know if that is the same issue; my issue seems to be entirely timing related and the semaphore patch doesn't affect to it. But now I suspect that there is something wrong with signal delivery in SMP. I'll try to find the cause for that, and then we can check if it was the same root cause. But I didn't find the cause yet, it seems to be some race in sig queuing/delivering.

closed 2025-04-26 by xiaoxiang781216

jlaitine, 2025-04-26:
@tmedicci could you check if the PR#16262 affects your issue? It was the cause for our crashes in signest test, and can also cause it to get stuck (test forever sleeping in semaphore).

tmedicci, 2025-04-28:
> [@tmedicci](https://github.com/tmedicci) could you check if the PR#16262 affects your issue? It was the cause for our crashes in signest test, and can also cause it to get stuck (test forever sleeping in semaphore).

Testing...

tmedicci, 2025-04-28:
> [@tmedicci](https://github.com/tmedicci) could you check if the PR#16262 affects your issue? It was the cause for our crashes in signest test, and can also cause it to get stuck (test forever sleeping in semaphore).

Thanks for reporting... Unfortunately, it doesn't affect this bug. The device is still stuck on the `signest` test.

jlaitine, 2025-05-02:
Hi, trying to get around the race conditions between signal dispatching and delivery in smp in https://github.com/apache/nuttx/pull/16301

Could you check, if the PR works on your esp32-devkit system, and whether it affects this particular bug?

tmedicci, 2025-05-05:
> Hi, trying to get around the race conditions between signal dispatching and delivery in smp in [#16301](https://github.com/apache/nuttx/pull/16301)
> 
> Could you check, if the PR works on your esp32-devkit system, and whether it affects this particular bug?
> 
> I also made another patch for the ostest app itself, which appears to be broken for SMP, in [apache/nuttx-apps#3067](https://github.com/apache/nuttx-apps/pull/3067)

Hi @jlaitine !

Thanks for tagging me here... I couldn't make further testing if this specific change fixed it, but our internal CI is now passing the `ostest` for ESP32 with SMP. So, this change probably fixed the issue (along with other possible related changes). Thanks for your attention!