apache/nuttx #16201: [BUG] esp32-devkitc:elf broken after #16154

tmedicci, 2025-04-14:
### Description / Steps to reproduce the issue

The `esp32-devkitc:elf ` defconfig (with `CONFIG_EXAMPLES_ELF=y`) is broken after https://github.com/apache/nuttx/pull/16154 (after 422c43949a95845712b31ea8f1260e2f964aa5cf commit, specifically).

## How to Reproduce

Build the firmware with:
```
make -j distclean && ./tools/configure.sh esp32-devkitc:elf && make flash ESPTOOL_PORT=/dev/ttyUSB1 ESPTOOL_BINDIR=./ -s -j$(nproc)
```

and, on NSH, run `elf`:

## Current Status

During build, the following logs are shown:
```
CC: errno.c
CC:  semaphore/sem_trywait.c LD: errno.o
CC:  inode/fs_files.c xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC:  mempool/mempool_procfs.c CC: hello.c
CC:  fixedmath/lib_b16atan2.c LD: hello.o
CC:  readline.c xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC:  nsh_consolemain.c CC: struct_main.c
CC:  readline_fd.c CC: struct_dummy.c
CC:  nsh_ddcmd.c LD: struct_main.o
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC:  grp/lib_getgrbuf.c CC: signal.c
CC: hello++1.cxx
LD: signal.o
CC:  binfmt_unregister.c xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC:  chip/esp32_extraheaps.c CC: hello++2.cxx
CC: hello++3.cxx
CC:  libgen/lib_basename.c CC: mutex.c
LD: hello++1.o
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
LD: hello++2.o
CC:  inode/fs_inoderelease.c xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC:  nsh_mmcmds.c LD: hello++3.o
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC:  binfmt_copyactions.c LD: mutex.o
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC:  binfmt_exit.c CC: pthread.c
CC: task.c
CC:  inode/fs_inodesearch.c LD: pthread.o
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC:  vfs/fs_chstat.c LD: task.o
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
```

And, while running `elf`, it crashes:
```
*** Booting NuttX ***
dram: lma 0x00001020 vma 0x3ffb2170 len 0x740    (1856)
iram: lma 0x00001768 vma 0x40080000 len 0x56fc   (22268)
padd: lma 0x00006e78 vma 0x00000000 len 0x9180   (37248)
imap: lma 0x00010000 vma 0x40100000 len 0x2868c  (165516)
padd: lma 0x00038694 vma 0x00000000 len 0x7964   (31076)
dmap: lma 0x00040000 vma 0x3f410000 len 0x2a658  (173656)
total segments stored 6

NuttShell (NSH) NuttX-10.4.0
nsh> elf
Initial memory usage: 13604
Registering romdisk at /dev/ram0

Memory Usage after romdisk_register:
  Before:    13604 After:    13668 Change:       64
Mounting ROMFS filesystem at target=/mnt/elf/romfs with source=/dev/ram0

[... 485 lines cut ...]
Esptool version:
  4.8.1

HAL version:
  sync/release_v5.1.c-nuttx-20230814-1583-ga461ca0750

CHIP ID:
  ESP32 has no Chip ID. Reading MAC instead.

Flash ID:
  Manufacturer: 20
  Device: 4016

Security information:
  Not found

Flash status:
  0x0000

MAC address:
  70:b8:f6:12:c4:04

### Verification

- [x] I have verified before submitting the report.

tmedicci, 2025-04-14:
@anjiahao1 , can you please take a look? Any ideas?

anjiahao1, 2025-04-15:
https://github.com/apache/nuttx/pull/16208 @tmedicci hello can try this pr, i build it in local no warnning now

Laczen, 2025-04-15:
@anjiahao1 I have tested the PR and can confirm that it solved the problem.

closed 2025-04-15 by xiaoxiang781216