apache/nuttx #16221: [BUG] Missing Standard C++ Headers When Building CxxTest Using CMake

trns1997, 2025-04-15:
### Description / Steps to reproduce the issue

## Description

When building NuttX using CMake (as documented in the [CMake Quickstart](https://nuttx.apache.org/docs/latest/quickstart/compiling_cmake.html)), the build fails during the compilation of `cxxtest_main.cxx` with the error:
```
fatal error: cstdio: No such file or directory
```

This issue does not occur when building using the classic Makefile configuration.

## Steps to Reproduce

1. **Configure the Board**
Run the following command in your NuttX source directory:
```
cmake -B build -DBOARD_CONFIG=xmc4800-relax:nsh -GNinja
```

2. **Build NuttX with CxxTest**
Build the project with:
```
cmake --build build
```

3. Observed Error
The build fails with the following output:
```
[2/8] Building CXX object apps/testing/cxx/cxxtest/CMakeFiles/apps_cxxtest.dir/cxxtest_main.cxx.obj
FAILED: apps/testing/cxx/cxxtest/CMakeFiles/apps_cxxtest.dir/cxxtest_main.cxx.obj 
/home/thomas/arm-gnu-toolchain-13.2.Rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-g++ -D__NuttX__ -Dmain=cxxtest_main -I/home/thomas/nuttxspace/apps/include -I/home/thomas/nuttxspace/nuttx/build/apps/include -isystem /home/thomas/nuttxspace/nuttx/include -isystem /home/thomas/nuttxspace/nuttx/build/include -mtune=cortex-m4 -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard --param=min-pagesize=0 -fno-common -Wall -Wshadow -Wundef -nostdlib -fomit-frame-pointer -mthumb -Wa,-mthumb -Wa,-mimplicit-it=always -ffunction-sections -fdata-sections -Wno-attributes -Wno-unknown-pragmas -nostdinc++ -Wno-psabi -std=c++17 -fdiagnostics-color=always -DNDEBUG -fmacro-prefix-map=/home/thomas/nuttxspace/nuttx= -fmacro-prefix-map=/home/thomas/nuttxspace/apps= -fmacro-prefix-map=/home/thomas/nuttxspace/nuttx/boards/arm/xmc4/xmc4800-relax= -fmacro-prefix-map=/home/thomas/nuttxspace/nuttx/arch/arm/src/xmc4= @apps/testing/cxx/cxxtest/CMakeFiles/apps_cxxtest.dir/cxxtest_main.cxx.obj.rsp -MD -MT apps/testing/cxx/cxxtest/CMakeFiles/apps_cxxtest.dir/cxxtest_main.cxx.obj -MF apps/testing/cxx/cxxtest/CMakeFiles/apps_cxxtest.dir/cxxtest_main.cxx.obj.d -o apps/testing/cxx/cxxtest/CMakeFiles/apps_cxxtest.dir/cxxtest_main.cxx.obj -c /home/thomas/nuttxspace/apps/testing/cxx/cxxtest/cxxtest_main.cxx
/home/thomas/nuttxspace/apps/testing/cxx/cxxtest/cxxtest_main.cxx:29:10: fatal error: cstdio: No such file or directory
   29 | #include <cstdio>
      |          ^~~~~~~~
compilation terminated.
[3/8] Linking C static library libs/libc/libc.a
ninja: build stopped: subcommand failed.
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

24.04

### NuttX Version

master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Build System]

### Host information

file sysinfo.h not exists
NuttX CFLAGS:
  -Wstrict-prototypes
  -Wno-attributes
  -Wno-unknown-pragmas
  -Wno-psabi
  -fomit-frame-pointer
  --param=min-pagesize=0
  -fno-common
[... 319 lines cut ...]
  CONFIG_TESTING_RAMTEST_PROGNAME=\ramtest\
  CONFIG_TESTING_RAMTEST_PRIORITY=90
  CONFIG_TESTING_RAMTEST_STACKSIZE=2048

Host system OS:
  Ubuntu 24.04.2 LTS Linux dank-engine 6.8.0-56-generic #58-Ubuntu SMP PREEMPT_DYNAMIC Fri Feb 14 15:33:28 UTC 2025 x86_64 x86_64

Host system PATH:
  /home/thomas/miniconda3/condabin
  /home/thomas/arm-gnu-toolchain-13.2.Rel1-x86_64-arm-none-eabi/bin
  /home/thomas/.local/bin
  /usr/local/sbin
  /usr/local/bin
  /usr/sbin
  /usr/bin
  /sbin
  /bin
  /usr/games
  /usr/local/games
  /snap/bin
  /snap/bin

### Verification

- [x] I have verified before submitting the report.

acassis, 2025-04-15:
@trns1997 does the same issue happen when you compile with Makefile?

@xiaoxiang781216 Have you or your team seen this issue before?

xiaoxiang781216, 2025-04-16:
@xuxin930 please take a look.

trns1997, 2025-04-16:
> [@trns1997](https://github.com/trns1997) does the same issue happen when you compile with Makefile?
>
No the issue does not occur when compiling using the Makefile.

xuxin930, 2025-04-17:
> [@xuxin930](https://github.com/xuxin930)请看一下。

sure @xiaoxiang781216 @acassis @trns1997 
I have located the issue, because when neither libcxx nor uclibc++ is selected, cmake does not have a default cxx standard library header file search path.
I consider adding it in a suitable place

Barto22, 2025-07-16:
Is there any update about this? I faced the same issue or maybe is there any temporary solution?

xuxin930, 2025-07-20:
let try this patch https://github.com/apache/nuttx/pull/16751

trns1997, 2025-07-20:
@xuxin930 I'll give it a try this evening if the patch is ready :)

closed 2025-07-21 by xiaoxiang781216