apache/nuttx #16245: [BUG] esp32: cannot create fatfs on /dev/mtdblock0

Laczen, 2025-04-21:
### Description / Steps to reproduce the issue

Esp32 allows the creation of a `/dev/mtdblock0` on spiflash. The `/dev/mtdblock0` however cannot be used for a fat filesystem.

To reproduce:
```console
./tools/configure.sh -l esp32-devkitc:nsh
```
use menuconfig to enable fatfs and spiflash, enable debug logging for file:
```console
nsh> mkfatfs -F 32 /dev/mtdblock0
bchdev_register: blkdev="/dev/mtdblock0" chardev="/dev/tmpc000001" readonly=F
find_blockdriver: pathname="/dev/mtdblock0"
ftl_geometry: Entry
ftl_geometry: available: true mediachanged: false writeenabled: true
ftl_geometry: nsectors: 4096 sectorsize: 256
ftl_geometry: Entry
ftl_geometry: available: true mediachanged: false writeenabled: true
ftl_geometry: nsectors: 4096 sectorsize: 256
mkfatfs_getgeometry: ERROR: Unsupported sector size: 256
nsh: mkfatfs: mkfatfs failed: 1
```

I think it is caused by a incompatibility between the definition of the `/dev/mtdblock0` (the sectorsize) and the sectorsize required by fatfs. However I have not found a way to configure the sectorsize of `/dev/mtdblock0`.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.04

### NuttX Version

master

### Issue Architecture

[Arch: xtensa]

### Issue Area

[Area: Drivers]

### Verification

- [x] I have verified before submitting the report.

Laczen, 2025-04-22:
I was able to circumvent this problem by enabling `CONFIG_MTD_SECT512` and adding the following lines to `esp32_board_spiflash.c`:

insert at line 432:
```
# if defined(CONFIG_MTD_SECT512)
  mtd = s512_initialize(mtd);
# endif
```

I am unsure if something like this should be added.

TimJTi, 2025-04-22:
Yes, FAT needs 512 byte sectors whereas most flash device (drivers) use 256 byte sectors.

Some flash device drivers have an "emulate 512 erase sector" feature which can be enabled instead of using the `CONFIG_MTD_SECT512`. I don't think _either_ are especially well documented - at least not when I went through this a few years ago!

abisop, 2025-05-02:
@Laczen FAT FS is not supposed to be used with mtd flash. you should go with SmartFS, SPIFFS,...

Laczen, 2025-05-02:
@abisop thanks for the reply.

While I understand that it is not ideal to use fatfs as a read/write FS on MTD flash there are situations where it is advantageous to use it. It is very easy to create a fatfs bin file on Linux and to populate it, this is afaik not supported for SmartFS, SPIFFS,...

Also, when the ftl layer would support flash wear levelling the main problem with fatfs on flash would disappear.

xiaoxiang781216, 2025-05-03:
If you really want to use fatfs on mtd, you need enable https://github.com/apache/nuttx/blob/master/drivers/mtd/dhara.c.

closed 2025-05-06 by Laczen