apache/nuttx #16246: [BUG] NXStyle is not getting braces declaration violation

acassis, 2025-04-22:
### Description / Steps to reproduce the issue

[image]

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

all

### NuttX Version

mainline

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Other]

### Host information

nxstyle needs to be fixed to catch this issue

### Verification

- [x] I have verified before submitting the report.

TimJTi, 2025-04-22:
What does a git bisect find?

acassis, 2025-04-22:
@TimJTi it is something that never was suppose to work

acassis, 2025-04-23:
@TimJTi maybe we can do a test submitting a PR with same code to double check. Maybe nxstyle in our machine catch it, but in the CI ignoring it.

raiden00pl, 2025-04-23:
it works for global variables but it doesn't work for function local variables: 

```
/****************************************************************************
 * Private Data
 ****************************************************************************/

const uint8_t g_xxx[] = {    ///////////// detected correctly
  0x00
};

/****************************************************************************
 * Public Functions
 ****************************************************************************/

void foo(void)
{
  int i;

  const uint8_t xxx[] = {   //////////////////// not detected
    0x00, 0x00, 0x00, 0x00
  };

  return;
}

```

closed 2026-10-15 by xiaoxiang781216