apache/nuttx #16249: [BUG] NXStyle is not reporting up case hexadecimals

acassis, 2025-04-22:
### Description / Steps to reproduce the issue

[image]

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

all

### NuttX Version

mainline

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Other]

### Host information

nxstyle needs to detect hexadecimal with upper case and report a warning

### Verification

- [x] I have verified before submitting the report.

TimJTi, 2025-04-22:
What does a git bisect find?

acassis, 2025-04-23:
@TimJTi strange, for some reason it didn't detect the issue in the CI.

@lupyuen do you know why CI's nxstyle didn't catch this issue?

lupyuen, 2025-04-23:
Sorry @simbit18 any idea why? Are we running a different nxstyle for CI? Thanks!

simbit18, 2025-04-23:
Hi @lupyuen as far as I can see nxstyle is the same.

checkpatch.sh
https://github.com/apache/nuttx/blob/216c4b00e2358fc0bb09db9c4eeb3cc0b240679a/tools/checkpatch.sh#L197

check.yml
https://github.com/apache/nuttx/blob/216c4b00e2358fc0bb09db9c4eeb3cc0b240679a/.github/workflows/check.yml#L49

I did a test locally with nxstyle and it does not detect up case hexadecimals

acassis, 2025-04-23:
Thanks for confirming @simbit18 !

raiden00pl, 2026-08-25:
I have a fix for nxstyle to detect this, but this rule isn't in the coding standard. If we want this to be implemented in the official tool, we should add this rule to the coding standard.