apache/nuttx #16256: [BUG] STM32H7 43zi unstable, USB master, ps command and syslog crashes/hangs

keever50, 2025-04-24:
### Description / Steps to reproduce the issue

I am using the STMH743ZI nucleo dev board and nuttx with default usb otg fs configuration.

**Reproduce USB crash at startup:**
I enabled syslog and debug logging for USB and GPIO. Check all the logging levels.
When the board starts, it hangs after USB is trying to log.

**Reproduce 'ps' command crash**
When you run the ps command,
the serial output is incomplete or garbage, then hangs.

**ostest hangs, hard to reproduce**
Sometimes, depending on configuration can sometimes hang.
This is not 100% of the time.

**CPU slow down when USB mass storage device is connected**
When an USB mass storage device like a flashdrive is connected, the NSH is noticeably slowing down.
This makes every operation extremely slow.

**USB crash after connection when syslog is on**
When you only enable SYSLOG for USB, USB stack does not crash right away. But when you connect something, it hangs with ``nsh> stm32_wait: RHport C``

I'd like to paste more logs and dumps, but NuttX is not giving me any on these problems. They are hangs.
I tried to debug this issue, but the debugger is stuck in the dumping process.

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Arch

### NuttX Version

Master

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Kernel]

### Host information

file sysinfo.h not exists
NuttX CFLAGS:
  -Wstrict-prototypes
  -Wno-attributes
  -Wno-unknown-pragmas
  -Wno-psabi
  -fomit-frame-pointer
  --param=min-pagesize=0
  -fno-common
  -Wall
  -Wshadow
  -Wundef
  -ffunction-sections
  -fdata-sections \-g\
  -mlittle-endian
  -march=armv7e-m
  -mtune=cortex-m7
  -mfpu=fpv5-d16
  -mfloat-abi=hard
  -mthumb
  -Wa-mthumb
  -Wa-mimplicit-it=always
[... 378 lines cut ...]
  CONFIG_TESTING_OSTEST_LOOPS=1
  CONFIG_TESTING_OSTEST_STACKSIZE=8192
  CONFIG_TESTING_OSTEST_NBARRIER_THREADS=8
  CONFIG_TESTING_OSTEST_RR_RANGE=10000
  CONFIG_TESTING_OSTEST_RR_RUNS=10
  CONFIG_TESTING_OSTEST_FPULOOPS=16
  CONFIG_TESTING_OSTEST_FPUMSDELAY=750
  CONFIG_TESTING_OSTEST_FPUPRIORITY=100
  CONFIG_TESTING_OSTEST_FPUSTACKSIZE=2048
  CONFIG_TESTING_OSTEST_WAITRESULT=y

Host system OS:
  Arch Linux Linux martini 6.14.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu 10 Apr 2025 18:43:59 +0000 x86_64 

Host system PATH:
  /usr/local/bin
  /usr/bin
  /usr/local/sbin
  /usr/bin/site_perl
  /usr/bin/vendor_perl
  /usr/bin/core_perl

### Verification

- [x] I have verified before submitting the report.

keever50, 2025-04-24:
[image]

It also randomly just hangs here. Its completely random. Happens sometimes after a reset.
You can reproduce this by resetting a few times.

raiden00pl, 2025-04-25:
the first thing you should do when something doesn't work: increase stack size! 1024 for IDLE may be not enough for complex configurations. 2048 as the default value for stack is probably also too low.

raiden00pl, 2025-04-25:
I confirm that there is something wrong when USB debug is enabled: the system crash when usbdev is connected

raiden00pl, 2025-04-25:
release/12.3 - works fine
release/12.4 - crash when usb stick plugged in and `ps` command is executed. USB debug works fine. When USB debug is enabled, `ps` command also works.
release/12.5 - same like 12.4, but additionally USB debug is broken
release/12.6 - totally broken when usb stick is plugged in
release/12.7 - totally broken when usb stick is plugged in
release/12.8 - totally broken when usb stick is plugged in

keever50, 2025-04-25:
@raiden00pl Thank you for testing! This is great info.

And wow, ofcourse the stack causes issues too. Its always stack. I have overlooked that.
However, I cannot find the option to increase minimum stack size or increase IDLE stack size, where can i find this?

I will also try the older versions. Thanks

keever50, 2025-04-25:
oh i actually found the stack size settings!

keever50, 2025-04-25:
@raiden00pl ps appears to be still broken with all stack defaults to 4kb

[image]

However, USB _seems_ to work now

keever50, 2025-04-25:
Also listing files using 'ls' in a USB drive also crashes. If the directory has many files atleast.

[image]

keever50, 2025-04-25:
If you enable kasan, it crashes at startup

[image]

keever50, 2025-04-25:
Confirmed that release/12.3 works fine. Been working on that for a little bit with different configs. It is stable. The default stack size really has to be increased in these configs.

raiden00pl, 2025-04-25:
I think `release/12.3` is the last release when I heavy tested usb stack for stm32h7 (both usbdev and usbhost). I used then `boards/arm/stm32h7/openh743i` board and everything works fine. Unfortunately, after all these years it is completely broken and I don't really feel like fixing it now.

raiden00pl, 2025-04-28:
what ostest problems are you talking about here? the ostest with UART has been working for me for more than hour and I don't see any problems.

keever50, 2025-04-28:
Unsure. I am unable to replicate. Might have been a fluke.
It was hanging once but I cannot get it to do that again. I'll report back
if it starts happening again.

keever50, 2025-04-28:
@raiden00pl Maybe you already discovered this but the usbhost thread uses more than 1kB stack after a connection. This is more than the default 1kB stacksize.
So i increased it to 4kB and that was more than enough.
I can not easily test this on a newer nuttx version at this moment.

[image]

raiden00pl, 2025-04-28:
good find. I didn't look at this problem more. Does increasing this stack value help? I didn't see `CONFIG_NUCLEOH743ZI_USBHOST_STACKSIZE` in configuration, because it's not present in board Kconfig:

https://github.com/apache/nuttx/blob/c4506929e94b14f36669689872ed1d0e1a2270de/boards/arm/stm32h7/nucleo-h743zi/src/stm32_usb.c#L61-L67

Lines like this are common in NuttX and they are dangerous! If the configuration is not in any Kconfig, the user is not even aware that something like this exists and can cause problems.

Soon I should add a patch that moves the usbhost initialization logic to a common place for all architectures. This way we will be able to get rid of files like ` boards/arm/stm32h7/nucleo-h743zi/src/stm32_usb.c` completely and avoid similar issues in the future.

keever50, 2025-04-28:
Ahh I thought it was not showing up because i am using a custom out of tree board. I spent a lot of time looking for it, but decided to look into the source code and indeed, i increased that value directly. That made it much more stable, as it does not corrupt memory anymore. But I am still at 12.3 and cannot easily switch because custom configurations, custom apps and things for this custom board.

Perhaps things are breaking in later versions because the stack is slowly growing everywhere.

raiden00pl, 2025-04-28:
@keever50 I tested usbhost with increased CONFIG_NUCLEOH743ZI_USBHOST_STACKSIZE to 4092 and it doesn't crash even with debug messages on. So there is a good chance that this is a valid fix. Congrats :)

keever50, 2025-04-28:
Nice!! I will attempt upgrading to master soon.
Thanks

keever50, 2025-04-29:
Solved! Thanks @raiden00pl 

Looking forward to your usb.c on one place solution

closed 2025-04-29 by keever50

raiden00pl, 2025-05-20:
@keever50 Hi, here is an example of how to use a common usb host waiter that is already upstream: https://github.com/apache/nuttx/pull/16414

One day we should migrate all usb host drivers to use this, but it will take some time.